mletenay/home-assistant-goodwe-inverter #1: UDP port is open but no response?

R0yk3, 2020-09-15:
When i ping using nmap i get this response: 
`
roy@Mosquito:~$ sudo nmap -sU -p 8899 192.168.1.166
Starting Nmap 7.70 ( https://nmap.org ) at 2020-09-15 16:37 UTC
Nmap scan report for 192.168.1.166
Host is up (0.016s latency).
PORT     STATE         SERVICE
8899/udp open|filtered ospf-lite
MAC Address: F0:FE:6B:D6:25:D2 (Shanghai High-Flying Electronics Technology)
Nmap done: 1 IP address (1 host up) scanned in 0.61 seconds
`
So the port is open.
but using the test script i get this:

`
bash-5.0# python test.py
2020-09-15 18:45:22,643 __init__(59) - DEBUG: Using selector: EpollSelector
2020-09-15 18:45:22,644 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-15 18:45:24,646 _timeout_heartbeat(248) - DEBUG: Timeout #1
2020-09-15 18:45:24,647 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-15 18:45:26,649 _timeout_heartbeat(248) - DEBUG: Timeout #2
2020-09-15 18:45:26,649 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-15 18:45:28,651 _timeout_heartbeat(248) - DEBUG: Timeout #3
2020-09-15 18:45:28,651 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-15 18:45:30,653 _timeout_heartbeat(252) - DEBUG: Timeout #4
^CTraceback (most recent call last):
  File "test.py", line 17, in <module>
    inverter = asyncio.run(inverter.discover(IP_ADDRESS, 8899))
  File "/usr/local/lib/python3.8/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 603, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 570, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1823, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/local/lib/python3.8/selectors.py", line 468, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
`

The logs from HA are

`2020-09-15 18:21:43 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for goodwe which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.
2020-09-15 18:21:53 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform goodwe is taking over 10 seconds.
2020-09-15 18:22:43 ERROR (MainThread) [homeassistant.components.sensor] Setup of platform goodwe is taking longer than 60 seconds. Startup will proceed without waiting any longer
`
I gues that is because the inverter in not responding the right way? 

I have a GW6000-DT

wishie, 2020-09-16:
I too have confirmed port 8899 is open|filtered but I do not get any data from either of my GoodWe GW5048-EM inverters. I am using inverter_test.py 

One inverter has a battery, and that is reporting
`2020-09-17 00:47:06,089 __init__(59) - DEBUG: Using selector: EpollSelector
2020-09-17 00:47:06,089 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:06,258 datagram_received(230) - DEBUG: Received: b'\xaaU\xf7\x83\x02 \xc3'
2020-09-17 00:47:06,258 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-17 00:47:06,259 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:06,502 datagram_received(230) - DEBUG: Received: b'\xaaU\xf7\x83\x02 \xc3'
2020-09-17 00:47:06,502 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-17 00:47:06,503 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:06,830 datagram_received(230) - DEBUG: Received: b'\xaaU\xf7\x83\x02 \xc3'
2020-09-17 00:47:06,830 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-17 00:47:06,830 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:07,073 datagram_received(230) - DEBUG: Received: b'\xaaU\xf7\x83\x02 \xc3'
2020-09-17 00:47:07,074 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-17 00:47:07,074 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:07,401 datagram_received(230) - DEBUG: Received: b'\xaaU\xf7\x83\x02 \xc3'
2020-09-17 00:47:07,401 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-17 00:47:07,401 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:07,638 datagram_received(230) - DEBUG: Received: b'\xaaU\xf7\x83\x02 \xc3'
2020-09-17 00:47:07,638 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-17 00:47:07,638 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:07,972 datagram_received(230) - DEBUG: Received: b'\xaaU\xf7\x83\x02 \xc3'
2020-09-17 00:47:07,972 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-17 00:47:07,972 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:08,090 _timeout_heartbeat(252) - DEBUG: Timeout #8
2020-09-17 00:47:08,260 _timeout_heartbeat(252) - DEBUG: Timeout #8
2020-09-17 00:47:08,504 _timeout_heartbeat(252) - DEBUG: Timeout #8
2020-09-17 00:47:08,831 _timeout_heartbeat(252) - DEBUG: Timeout #8
2020-09-17 00:47:09,075 _timeout_heartbeat(252) - DEBUG: Timeout #8
2020-09-17 00:47:09,402 _timeout_heartbeat(252) - DEBUG: Timeout #8
2020-09-17 00:47:09,640 _timeout_heartbeat(252) - DEBUG: Timeout #8
2020-09-17 00:47:09,974 _timeout_heartbeat(252) - DEBUG: Timeout #8`

Whilst the inverter without a battery (could possibly be sleeping, as its 1AM) outputs

`2020-09-17 00:47:20,015 __init__(59) - DEBUG: Using selector: EpollSelector
2020-09-17 00:47:20,015 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:22,017 _timeout_heartbeat(248) - DEBUG: Timeout #1
2020-09-17 00:47:22,018 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:24,020 _timeout_heartbeat(248) - DEBUG: Timeout #2
2020-09-17 00:47:24,020 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:26,022 _timeout_heartbeat(248) - DEBUG: Timeout #3
2020-09-17 00:47:26,022 connection_made(225) - DEBUG: Send: b'\xf7\x03\x88\xb8\x00!:\xc1'
2020-09-17 00:47:28,024 _timeout_heartbeat(252) - DEBUG: Timeout #4`

mletenay, 2020-09-16:
Hi @R0yk3 @wishie
can you try to connect to your inverter via official GoodWe's mobile application "PV Master" ?
It is intended for newer hybrid inverters, but maybe it can work with DT or EM as well.
If it works, than I may be able to figure out the UDP protocol specifics and extends the existing ET-only code.

wishie, 2020-09-17:
The GW5048-EM is a hybrid inverter. I have 2 of them. 1 with a battery, the other without. PV Master app works, but only when I am on the inverters WiFi AP I believe.

wishie, 2020-09-17:
I have also noticed that UDP port 48899 is listening, too.

R0yk3, 2020-09-17:
Friday i'm @home again i wil try it then.

R0yk3, 2020-09-18:
Hi, the app connects but there are no readings? evenrything is N/A

wishie, 2020-09-18:
PV Master only shows my inverter if I am connected to the Inverters WiFi.. but I can certainly see data from it.

[image]

R0yk3, 2020-09-18:
If i connect tot the inverters wifi the app does more fond the inverter.???

wishie, 2020-09-18:
I am having the issue that if I connect to the inverters WiFi, PV Master then sees the inverters. But it might be due to an issue on my side. I will do some more testing.

wishie, 2020-09-18:
Very randomly I will get some sort of response from the inverter..

```
C:\Users\Travis Place\Desktop\home-assistant-goodwe-inverter-master>python inverter_test.py
2020-09-18 21:52:10,557 __init__(623) - DEBUG: Using proactor: IocpProactor
2020-09-18 21:52:10,558 discover(346) - DEBUG: Probing ET inverter at 192.168.100.6:8899
2020-09-18 21:52:10,559 connection_made(225) - DEBUG: Send: 'f70388b800213ac1'
2020-09-18 21:52:10,835 datagram_received(230) - DEBUG: Received: 'aa55f7830220c3'
2020-09-18 21:52:10,835 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-18 21:52:10,838 connection_made(225) - DEBUG: Send: 'f70388b800213ac1'
2020-09-18 21:52:11,078 datagram_received(230) - DEBUG: Received: 'aa55f7830220c3'
2020-09-18 21:52:11,079 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-18 21:52:11,081 connection_made(225) - DEBUG: Send: 'f70388b800213ac1'
2020-09-18 21:52:11,411 datagram_received(230) - DEBUG: Received: 'aa55f7830220c3'
2020-09-18 21:52:11,412 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-18 21:52:11,414 connection_made(225) - DEBUG: Send: 'f70388b800213ac1'
2020-09-18 21:52:11,644 datagram_received(230) - DEBUG: Received: 'aa55f7830220c3'
2020-09-18 21:52:11,652 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-18 21:52:11,653 connection_made(225) - DEBUG: Send: 'f70388b800213ac1'
2020-09-18 21:52:11,982 datagram_received(230) - DEBUG: Received: 'aa55f7830220c3'
2020-09-18 21:52:11,982 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 0
2020-09-18 21:52:11,986 connection_made(225) - DEBUG: Send: 'f70388b800213ac1'
2020-09-18 21:52:12,514 datagram_received(230) - DEBUG: Received: 'aa557fc001868c0000000000000000000001fe00000052010e00e20064006400000e000e00020000010977002b000113800109770021016f13800102013000000000000024320000051600ae01460000441f04300100300002b4000100000009000000400000140912153406000000000000000000000000000000000000000000000000000011f8000010ea020000000000000f43'
2020-09-18 21:52:12,522 datagram_received(236) - DEBUG: Unexpected response length: expected 66, received: 142
2020-09-18 21:52:12,525 connection_made(225) - DEBUG: Send: 'f70388b800213ac1'
2020-09-18 21:52:12,579 _timeout_heartbeat(252) - DEBUG: Timeout #7
2020-09-18 21:52:12,844 _timeout_heartbeat(252) - DEBUG: Timeout #7
2020-09-18 21:52:13,101 _timeout_heartbeat(252) - DEBUG: Timeout #7
2020-09-18 21:52:13,420 _timeout_heartbeat(252) - DEBUG: Timeout #7
2020-09-18 21:52:13,680 _timeout_heartbeat(252) - DEBUG: Timeout #7
2020-09-18 21:52:14,007 _timeout_heartbeat(252) - DEBUG: Timeout #7
2020-09-18 21:52:14,539 _timeout_heartbeat(252) - DEBUG: Timeout #7
```

mletenay, 2020-09-18:
It seems the EM/ES inverters do talk different protocol than ET.
I've found some reference here https://powerforum.co.za/topic/6931-goodwe-5048es-protocol-reverse-engineering/.
I'll try to incorporate that into my code, just the testing will be a pain since I have no access to such inverters ...

R0yk3, 2020-09-19:
I'll test for you. Maybe i can make a tunnel so you have direct access.

wishie, 2020-09-19:
Im happy to help test things also, just incase the EM and DT inverters are different again..

mletenay, 2020-09-19:
I've just pushed commit with basic support for EM/ES inverters, not sure about DT yet.
Still work in progress, but may already provide some data.

wishie, 2020-09-19:
So far, I get this:
```2020-09-20 06:40:30,714 __init__(59) - DEBUG: Using selector: EpollSelector
2020-09-20 06:40:30,714 discover(421) - DEBUG: Probing ET inverter at 192.168.100.6:8899
2020-09-20 06:40:30,714 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 06:40:31,150 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 06:40:31,150 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 06:40:31,150 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 06:40:31,376 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 06:40:31,377 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 06:40:31,377 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 06:40:31,740 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 06:40:31,740 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 06:40:31,740 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 06:40:32,074 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 06:40:32,074 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 06:40:32,074 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 06:40:32,299 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 06:40:32,299 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 06:40:32,299 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 06:40:32,589 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 06:40:32,590 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 06:40:32,590 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 06:40:32,715 _timeout_heartbeat(327) - DEBUG: Timeout #7, closing socket
2020-09-20 06:40:32,715 discover(421) - DEBUG: Probing ES inverter at 192.168.100.6:8899
2020-09-20 06:40:32,716 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:40:32,867 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 06:40:32,867 datagram_received(308) - DEBUG: Unexpected response length: expected 85, received: 7
2020-09-20 06:40:32,867 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:40:33,225 datagram_received(303) - DEBUG: Received: 'aa557fc001824d31303130424757353034382d454d20233130202020202020202020202020203735303438454d553139365730333230333630303431302d30343030302d3130203431302d30323033342d31311011b6'
2020-09-20 06:40:33,225 datagram_received(308) - DEBUG: Unexpected response length: expected 85, received: 86
2020-09-20 06:40:33,225 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:40:33,489 datagram_received(303) - DEBUG: Received: 'aa557fc001824d31303130424757353034382d454d20233130202020202020202020202020203735303438454d553139365730333230333630303431302d30343030302d3130203431302d30323033342d31311011b6'
2020-09-20 06:40:33,489 datagram_received(308) - DEBUG: Unexpected response length: expected 85, received: 86
2020-09-20 06:40:33,489 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:40:33,830 datagram_received(303) - DEBUG: Received: 'aa557fc001824d31303130424757353034382d454d20233130202020202020202020202020203735303438454d553139365730333230333630303431302d30343030302d3130203431302d30323033342d31311011b6'
2020-09-20 06:40:33,830 datagram_received(308) - DEBUG: Unexpected response length: expected 85, received: 86
2020-09-20 06:40:33,830 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:40:34,133 datagram_received(303) - DEBUG: Received: 'aa557fc001824d31303130424757353034382d454d20233130202020202020202020202020203735303438454d553139365730333230333630303431302d30343030302d3130203431302d30323033342d31311011b6'
2020-09-20 06:40:34,133 datagram_received(308) - DEBUG: Unexpected response length: expected 85, received: 86
2020-09-20 06:40:34,133 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:40:34,525 datagram_received(303) - DEBUG: Received: 'aa557fc001824d31303130424757353034382d454d20233130202020202020202020202020203735303438454d553139365730333230333630303431302d30343030302d3130203431302d30323033342d31311011b6'
2020-09-20 06:40:34,525 datagram_received(308) - DEBUG: Unexpected response length: expected 85, received: 86
2020-09-20 06:40:34,525 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:40:34,716 _timeout_heartbeat(327) - DEBUG: Timeout #7, closing socket```

I am just walking out the door to go to a job but ill be home again in a few hours and will continue testing. Thanks

wishie, 2020-09-19:
The inverter without the battery connected shows this:
```
2020-09-20 06:43:32,833 __init__(59) - DEBUG: Using selector: EpollSelector
2020-09-20 06:43:32,833 discover(421) - DEBUG: Probing ET inverter at 192.168.100.22:8899
2020-09-20 06:43:32,834 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 06:43:34,836 _timeout_heartbeat(323) - DEBUG: Timeout #1
2020-09-20 06:43:34,836 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 06:43:36,838 _timeout_heartbeat(323) - DEBUG: Timeout #2
2020-09-20 06:43:36,838 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 06:43:38,840 _timeout_heartbeat(323) - DEBUG: Timeout #3
2020-09-20 06:43:38,840 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 06:43:40,843 _timeout_heartbeat(327) - DEBUG: Timeout #4, closing socket
2020-09-20 06:43:40,843 discover(421) - DEBUG: Probing ES inverter at 192.168.100.22:8899
2020-09-20 06:43:40,843 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:43:41,441 datagram_received(303) - DEBUG: Received: 'aa557fc001824d30393039384757353034382d454d20233130202020202020202020202020203935303438454d553139415730313232333630303431302d30343030302d3039203431302d30323033342d30380911d0'
2020-09-20 06:43:41,441 datagram_received(308) - DEBUG: Unexpected response length: expected 85, received: 86
2020-09-20 06:43:41,441 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:43:41,794 datagram_received(303) - DEBUG: Received: 'aa557fc001824d30393039384757353034382d454d20233130202020202020202020202020203935303438454d553139415730313232333630303431302d30343030302d3039203431302d30323033342d30380911d0'
2020-09-20 06:43:41,794 datagram_received(308) - DEBUG: Unexpected response length: expected 85, received: 86
2020-09-20 06:43:41,794 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:43:42,235 datagram_received(303) - DEBUG: Received: 'aa557fc001824d30393039384757353034382d454d20233130202020202020202020202020203935303438454d553139415730313232333630303431302d30343030302d3039203431302d30323033342d30380911d0'
2020-09-20 06:43:42,235 datagram_received(308) - DEBUG: Unexpected response length: expected 85, received: 86
2020-09-20 06:43:42,235 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:43:42,640 datagram_received(303) - DEBUG: Received: 'aa557fc001824d30393039384757353034382d454d20233130202020202020202020202020203935303438454d553139415730313232333630303431302d30343030302d3039203431302d30323033342d30380911d0'
2020-09-20 06:43:42,640 datagram_received(308) - DEBUG: Unexpected response length: expected 85, received: 86
2020-09-20 06:43:42,640 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 06:43:42,844 _timeout_heartbeat(327) - DEBUG: Timeout #5, closing socket
```

mletenay, 2020-09-19:
ES and ET seem to have different lengths of the 'aa55c07f0102000241'. Fixed from 85 to 86.

wishie, 2020-09-20:
On my GW5048-EM with a battery connected:
```
/config # python3 inverter_test.py 
2020-09-20 15:42:21,319 __init__(59) - DEBUG: Using selector: EpollSelector
2020-09-20 15:42:21,319 discover(421) - DEBUG: Probing ET inverter at 192.168.100.6:8899
2020-09-20 15:42:21,319 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:42:23,321 _timeout_heartbeat(323) - DEBUG: Timeout #1
2020-09-20 15:42:23,321 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:42:23,546 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 15:42:23,546 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 15:42:23,546 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:42:23,880 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 15:42:23,880 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 15:42:23,880 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:42:24,108 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 15:42:24,108 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 15:42:24,108 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:42:24,455 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 15:42:24,455 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 15:42:24,455 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:42:24,698 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 15:42:24,698 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 15:42:24,698 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:42:25,039 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 15:42:25,039 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 15:42:25,039 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:42:25,275 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 15:42:25,275 datagram_received(308) - DEBUG: Unexpected response length: expected 73, received: 7
2020-09-20 15:42:25,275 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:42:25,322 _timeout_heartbeat(327) - DEBUG: Timeout #9, closing socket
2020-09-20 15:42:25,322 discover(421) - DEBUG: Probing ES inverter at 192.168.100.6:8899
2020-09-20 15:42:25,323 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 15:42:25,613 datagram_received(303) - DEBUG: Received: 'aa55f7830220c3'
2020-09-20 15:42:25,613 datagram_received(308) - DEBUG: Unexpected response length: expected 86, received: 7
2020-09-20 15:42:25,613 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 15:42:25,835 datagram_received(303) - DEBUG: Received: 'aa557fc001824d31303130424757353034382d454d20233130202020202020202020202020203735303438454d553139365730333230333630303431302d30343030302d3130203431302d30323033342d31311011b6'
2020-09-20 15:42:25,835 discover(423) - DEBUG: Detected ES inverter GW5048-EM , S/N:75048EMU196W0320
2020-09-20 15:42:25,836 __init__(59) - DEBUG: Using selector: EpollSelector
2020-09-20 15:42:25,836 connection_made(292) - DEBUG: Send: 'aa55c07f0106000245'
2020-09-20 15:42:26,126 datagram_received(303) - DEBUG: Received: 'aa557fc001824d31303130424757353034382d454d20233130202020202020202020202020203735303438454d553139365730333230333630303431302d30343030302d3130203431302d30323033342d31311011b6'
2020-09-20 15:42:26,126 datagram_received(308) - DEBUG: Unexpected response length: expected 149, received: 86
2020-09-20 15:42:26,126 connection_made(292) - DEBUG: Send: 'aa55c07f0106000245'
2020-09-20 15:42:26,667 datagram_received(303) - DEBUG: Received: 'aa557fc001868c051c0003000d51000302021100000052012200b90063006300006100610002000001096f002b0002137b01096f002400d1137b0102015c000000000000256a0000053f005800b000004675040801003000034f00010000fffd00000040
00001409140f2b16000000000000000000000000000000000000000000000000000012cb0000115b020000000000001161'
```

I see it detected the model number and the serial number.

wishie, 2020-09-20:
On my GW5048-EM inverter **without** a battery connected
```
/config # python3 inverter_test.py 
2020-09-20 15:44:16,412 __init__(59) - DEBUG: Using selector: EpollSelector
2020-09-20 15:44:16,412 discover(421) - DEBUG: Probing ET inverter at 192.168.100.22:8899
2020-09-20 15:44:16,412 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:44:18,414 _timeout_heartbeat(323) - DEBUG: Timeout #1
2020-09-20 15:44:18,414 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:44:20,417 _timeout_heartbeat(323) - DEBUG: Timeout #2
2020-09-20 15:44:20,417 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:44:22,419 _timeout_heartbeat(323) - DEBUG: Timeout #3
2020-09-20 15:44:22,419 connection_made(292) - DEBUG: Send: 'f70388b800213ac1'
2020-09-20 15:44:24,421 _timeout_heartbeat(327) - DEBUG: Timeout #4, closing socket
2020-09-20 15:44:24,422 discover(421) - DEBUG: Probing ES inverter at 192.168.100.22:8899
2020-09-20 15:44:24,422 connection_made(292) - DEBUG: Send: 'aa55c07f0102000241'
2020-09-20 15:44:24,758 datagram_received(303) - DEBUG: Received: 'aa557fc001824d30393039384757353034382d454d20233130202020202020202020202020203935303438454d553139415730313232333630303431302d30343030302d3039203431302d30323033342d30380911d0'
2020-09-20 15:44:24,759 discover(423) - DEBUG: Detected ES inverter GW5048-EM , S/N:95048EMU19AW0122
2020-09-20 15:44:24,759 __init__(59) - DEBUG: Using selector: EpollSelector
2020-09-20 15:44:24,759 connection_made(292) - DEBUG: Send: 'aa55c07f0106000245'
2020-09-20 15:44:25,306 datagram_received(303) - DEBUG: Received: 'aa557fc00186850ce7000402084e0005020000000000000000000000000000010000000000000000010966000a0000137f010000000000d700000002012c00000000000016f50000012b00e900cd0000158000d501002000000000010000fff700004845
00001409140f2d1000000000000000000000000000000000000000000000000000000000000000000e2d'
2020-09-20 15:44:25,306 datagram_received(308) - DEBUG: Unexpected response length: expected 149, received: 142
2020-09-20 15:44:25,306 connection_made(292) - DEBUG: Send: 'aa55c07f0106000245'
2020-09-20 15:44:25,807 datagram_received(303) - DEBUG: Received: 'aa557fc00186850cf90004020855000502000000000000000000000000000001000000000000000001096a000a0009137f0100000000010200000002012c00000000000016f50000012b00e900cd0000158000d501002000000000010000ffcd00004845
00001409140f2d1000000000000000000000000000000000000000000000000000000000000000000d55'
2020-09-20 15:44:25,807 datagram_received(308) - DEBUG: Unexpected response length: expected 149, received: 142
2020-09-20 15:44:25,807 connection_made(292) - DEBUG: Send: 'aa55c07f0106000245'
2020-09-20 15:44:26,272 datagram_received(303) - DEBUG: Received: 'aa557fc00186850cfa0004020855000502000000000000000000000000000001000000000000000001096a000a0000137f010000000000d700000002012c00000000000016f50000012b00e900cd0000158000d501002000000000010000fff800004845
00001409140f2d1000000000000000000000000000000000000000000000000000000000000000000e4c'
2020-09-20 15:44:26,272 datagram_received(308) - DEBUG: Unexpected response length: expected 149, received: 142
2020-09-20 15:44:26,272 connection_made(292) - DEBUG: Send: 'aa55c07f0106000245'
2020-09-20 15:44:26,760 datagram_received(303) - DEBUG: Received: 'aa557fc00186850d0f000402085c0005020000000000000000000000000000010000000000000000010968000a0007137f010000000000f900000002012b00000000000016f50000012b00e900cd0000158000d401002000000000010000ffdd00004845
00001409140f2d1300000000000000000000000000000000000000000000000000000000000000000d76'
2020-09-20 15:44:26,760 datagram_received(308) - DEBUG: Unexpected response length: expected 149, received: 142
2020-09-20 15:44:26,760 connection_made(292) - DEBUG: Send: 'aa55c07f0106000245'
2020-09-20 15:44:26,760 _timeout_heartbeat(327) - DEBUG: Timeout #5, closing socket
```

I see it detected the model number and the serial number of this inverter, also.

mletenay, 2020-09-20:
The with/without battery inverters return responses with different lengths, I've adjusted the code accordingly.
It parses the values now, but at least the power values seem to be interpreted wrong way, please verify.

wishie, 2020-09-20:
I will clarify that both of my inverters are "GW5048-EM". The one with the battery is actually sold as an 'Alpha-ESS eSTORE 10.3' system (10.3kWh battery) and contains a different firmware version that the other inverter, which is a plain old GW5048-EM directly from GoodWe.

GoodWe have stated that the firmware differences are only related to battery functions, and I can confirm that both inverters behave the same in relation to PVMaster and SEMSPortal.

I will check your updated code now. Thanks!

wishie, 2020-09-20:
Here is the output from the inverter without a battery:

```
PV1 Voltage = 0.0 V
PV1 Current = 0.0 A
PV1 Mode = Disconnect the inverter and PV panels 
PV2 Voltage = 0.0 V
PV2 Current = 0.0 A
PV2 Mode = Disconnect the inverter and PV panels 
Battery Voltage = 0.0 V
Battery Voltage 2 = 0.0 V
Battery Voltage 3 = 0.0 V
Battery Voltage 4 = 0.0 V
Battery Current = 0.0 A
Battery Charge Limit = 0 A
Battery Discharge Limit = 0 A
Battery State of Charge = 0 %
Battery State of Charge 2 = 0 %
Battery State of Charge 3 = 0 %
Battery State of Health = 0 %
Battery Mode = No battery or battery disconnected 
On-grid Voltage = 241.2 V
On-grid Current = 0.0 A
On-grid Power = 594831 W
On-grid Frequency = 50.07 Hz
Work Mode = None 
Back-up Voltage = 0.0 V
Back-up Current = 0.0 A
Back-up Power = 589824 W
Back-up Frequency = 0.0 Hz
Load Mode = The inverter is connected to a load 
Energy Mode = Wait Mode 
Inverter Temperature = 22.2 C
Error Codes = 0 
Total Energy = 587.9 kW
Today's Energy = 1533563.9 kW
Today's Load = 14024704 kW
Total Load = 551.4 kW
Total Power = 0 kW
On-grid Mode = Inverter neither send power to grid,nor get power from grid 
Diag Status = 540740 
```

I think "Today's Energy" seems a little inaccurate haha.

wishie, 2020-09-20:
And here is the output from the inverter WITH the battery:

```
PV1 Voltage = 0.0 V
PV1 Current = 0.0 A
PV1 Mode = Disconnect the inverter and PV panels 
PV2 Voltage = 0.0 V
PV2 Current = 0.0 A
PV2 Mode = Disconnect the inverter and PV panels 
Battery Voltage = 50.5 V
Battery Voltage 2 = 0.0 V
Battery Voltage 3 = 8.2 V
Battery Voltage 4 = 30.0 V
Battery Current = 28.6 A
Battery Charge Limit = 99 A
Battery Discharge Limit = 99 A
Battery State of Charge = 12 %
Battery State of Charge 2 = 0 %
Battery State of Charge 3 = 12 %
Battery State of Health = 0 %
Battery Mode = Discharge 
On-grid Voltage = 242.2 V
On-grid Current = 5.4 A
On-grid Power = 4998 W
On-grid Frequency = 49.98 Hz
Work Mode = None 
Back-up Voltage = 242.2 V
Back-up Current = 4.3 A
Back-up Power = 21369736 W
Back-up Frequency = 50.0 Hz
Load Mode = Inverter and the load is disconnected 
Energy Mode = Normal(On-Grid) 
Inverter Temperature = 38.3 C
Error Codes = 0 
Total Energy = 966.1 kW
Today's Energy = 1114138.6 kW
Today's Load = 17367040 kW
Total Load = 1812.7 kW
Total Power = 1330 kW
On-grid Mode = Inverter neither send power to grid,nor get power from grid 
Diag Status = 64 
```

R0yk3, 2020-09-20:
That looks prommising.
Is it possible in stead of converting it to Hass. Make te output to mqtt.
So anybody van use it.
Ik am still out doe the weekend nut wil try your code towmorrow.

mletenay, 2020-09-20:
I've fixed the power values calculation, it's correct now, although the inverter reports weird values ...

wishie, 2020-09-20:
I can confirm the power values now seem sane.

So in my case with 2 inverters, does this component support me adding 2, or will modifications need to be made?

mletenay, 2020-09-20:
The inverter's serialNr is part of the entity unique ID, so it should be possible to have more inverters at the same time.
There's also the optional "sensor_name_prefix" parameter, so you can control the external sensor names to have meaningful/distinctive entity names.
I mean something like this:
```
sensor:
  - platform: goodwe
    ip_address: 192.168.1.100
    sensor_name_prefix: "gw1"

sensor:
  - platform: goodwe
    ip_address: 192.168.1.101
    sensor_name_prefix: "gw2"
```
As mentioned in the docs, the sensor_name_prefix is used by home-assistant during initial (first-time) creation of sensor entities.
They are internally identified by some uuids, but externally by the entity names.
So you may e.g. define some value for the first load of the component, that name will get into the entity names and later change it into something more human readable, the entity names will already established and won't change.

wishie, 2020-09-20:
Ok, excellent. I will add both inverters and do some testing in the morning when there is Sun again.

Thank you very much for your efforts so far. The one thing that annoyed me about the GoodWe (as compared to my old ABB inverter) was that the GoodWe required the SEMSPortal to get any information. Not anymore! I used to pull data from my ABB inverter via an RS485 to USB adapter).

wishie, 2020-09-20:
I get some odd behaviour when adding 2 inverters. My config is as follows:
```
sensor:
  - platform: goodwe
    ip_address: 192.168.100.6
    scan_interval: 30
    sensor_name_prefix: Inverter1
  - platform: goodwe
    ip_address: 192.168.100.22
    scan_interval: 30
    sensor_name_prefix: Inverter2
```

But I don't get all the sensors.. only 1 from 'Inverter1'
[image]

R0yk3, 2020-09-21:
Installed it in HASS again, but no result.
you dit not changed the test?

mletenay, 2020-09-21:
@wishie That's odd. I'll try to somehow emulate 2 inverters to see what's wrong.
But maybe it will be the best if you file a separate issue for the "2 inverters" problem, so I can track it better, this one is becoming quite long.

wishie, 2020-09-21:
@mletenay done. I was about to suggest splitting off a new issue myself.

mletenay, 2020-09-21:
@R0yk3 It seems the DT inverter does not response neither to ET nor EM/ES protocol.
Did you say the "PV Master" application was not working ?
One thing which may help is to try to install the "SolarGo" application instead and if that one works, use some packet sniffer to catch the traffic between the app and the inverter.
Maybe I could decode that one too, but I need at least some data ...

R0yk3, 2020-09-21:
Solar gives no response. At least it says the inverter in not compatible with the app.
I wil try if i get some packets between the PV master and inverter. It does connect but all data stays 0

mletenay, 2020-09-21:
@R0yk3 If you'll succeed catching some packets, I suggest to continue the DT inverter support within the issue #3 and I will close this one as EM/ES specific one.

gbyleveldt, 2020-09-21:
Ah, I just read through this ticket now, after opening another ticket. I'm the one that started the protocol decode on the Power Forum website you referenced earlier. I guess we can continue this on the other ticket that's been opened. Thanks again!!

Ascariota, 2020-09-24:
I have the same "Error" with GW3000D-NS.

custom_components.goodwe.goodwe_inverter.DiscoveryError: Unable to connect to the inverter at host=10.0.4.10 port=8899, or your inverter is not supported yet.
Failures=[CancelledError(), CancelledError()]

mletenay, 2020-09-24:
@Ascariota Can you connect to your inverter by either PV Master or Solar Go mobile app ? Do you see any data there ?
If yes, can you sniff the packets sent between the application and inverter so we can analyze them ?
If no, is there another way you can watch the inverter data (except the SEMS portal) ?

mletenay, 2020-09-24:
@Ascariota , @R0yk3 If none of the GoodWe specific apps work, can you try solution from https://github.com/XtheOne/Inverter-Data-Logger ? If that works, maybe there's still hope for you :-)

Ascariota, 2020-09-27:
@mletenay None apps OK for me I try tomorrow the other intégration I keep you inform

mletenay, 2020-10-11:
Hi @R0yk3 
can you try to execute commands mentioned in 
https://github.com/mletenay/home-assistant-goodwe-inverter/issues/3#issuecomment-706777452
on your DT inverter ?
It's a long shot, but maybe they could work and get some data.

closed 2020-12-13 by mletenay