mletenay/home-assistant-goodwe-inverter #3: Support for NS Invertors?

rutgerputter, 2020-09-21:
Hi, this looks like a perfect way to measure all output from my GW2500-NS!

I used nmap and I see that ports 8899 and 48899 are open on the IP address of the invertor, but the invertor-test.py doesn't get any response.

Is there a way to make this work on NS type invertors?

mletenay, 2020-09-21:
The grid-tie NS inverter is probably the same case as the DT inverter mentioned in 
https://github.com/mletenay/home-assistant-goodwe-inverter/issues/1#issuecomment-695979882 
They seem to talk different protocol than ET/ES, so we need to sniff and decode it first.

mletenay, 2020-09-24:
@rutgerputter Can you check if you can get some data from your inverter via https://github.com/XtheOne/Inverter-Data-Logger ?

rutgerputter, 2020-09-25:
I've checked, but it tells me this:
`2020-09-25 09:26:50,333 DEBUG Importing output plugin ConsoleOutput
2020-09-25 09:26:50,334 INFO Connecting to logger with IP: 192.168.9.20 and SN 18BW****
2020-09-25 09:26:50,334 INFO connecting to 192.168.9.20 port 8899
2020-09-25 09:26:55,337 ERROR Could not open socket
2020-09-25 09:26:55,338 ERROR timed out`

And that's strange because the port is open:
`nc -vzu 192.168.9.20 8899
192.168.9.20 (192.168.9.20:8899) open`

I've tried both 8899 and 48899 UDP.

mletenay, 2020-09-25:
The UDP of the inverters is rather unreliable and timeouts are frequent, you may have to try several times.
Did you checked also the ScanLoggers.py ? What is its output ?

rutgerputter, 2020-09-25:
I tried, several time over the span of 15 minutes.

ScanLoggers.py doesnt see the inverter as well, and it is on the same subnet. Also ran this multiple times.

TimSoethout, 2020-09-25:
Same result here for Goodwe GW3000-NS.

```
$python LiveStats.py
2020-09-25 10:21:22,538 DEBUG Importing output plugin ConsoleOutput
2020-09-25 10:21:22,538 INFO Connecting to logger with IP: 192.168.1.125 and SN 93000SSNXXXX
2020-09-25 10:21:22,539 INFO connecting to 192.168.1.125 port 8899
2020-09-25 10:21:22,661 ERROR Could not open socket
2020-09-25 10:21:22,661 ERROR [Errno 61] Connection refused
2020-09-25 10:21:22,661 ERROR Error connecting to logger with IP: 192.168.1.125 and SN 93000SSNXXXX, trying next logger.
```

`python ScanLoggers.py` results in `socket.error: [Errno 49] Can't assign requested address` on OSX.

mletenay, 2020-09-25:
@rutgerputter That's odd and smells like some network related issue, since as long as the inverter listens on the port 48899 it should at least answer to broadcast message (and potentially fail later). Could you play e.g. with the "time-to-live" param in the ScanLoggers.py ?

mletenay, 2020-10-11:
Hi @TimSoethout or @rutgerputter 
It's just shoot from the hip, but can you try to execute the inverter_test.py with following changes ?
Remove everything after the IP_ADDRESS line and add there following commands:
```
print(
    asyncio.run(
        inverter.ProtocolCommand(
            bytes.fromhex("7F03753100280409"), lambda r: True
        ).execute(IP_ADDRESS, 8899)
    ).hex()
)
print(
    asyncio.run(
        inverter.ProtocolCommand(
            bytes.fromhex("7F0375940049D5C2"), lambda r: True
        ).execute(IP_ADDRESS, 8899)
    ).hex()
) 
```
Do you get any response ?

rutgerputter, 2020-10-13:
I tried your approach for the inverter_test.py and sadly no dice. I tried on both ports 8899 and 48899.

This is the output:
/config # python ./inverter_test.py 
2020-10-13 10:58:58,981 __init__(59) - DEBUG: Using selector: EpollSelector
2020-10-13 10:58:58,982 connection_made(319) - DEBUG: Send: '7f03753100280409'
2020-10-13 10:59:00,984 _timeout_heartbeat(349) - DEBUG: Re-try #1
2020-10-13 10:59:00,984 connection_made(319) - DEBUG: Send: '7f03753100280409'
2020-10-13 10:59:02,986 _timeout_heartbeat(349) - DEBUG: Re-try #2
2020-10-13 10:59:02,986 connection_made(319) - DEBUG: Send: '7f03753100280409'
2020-10-13 10:59:04,988 _timeout_heartbeat(349) - DEBUG: Re-try #3
2020-10-13 10:59:04,988 connection_made(319) - DEBUG: Send: '7f03753100280409'
2020-10-13 10:59:06,990 _timeout_heartbeat(353) - DEBUG: Re-try #4, closing socket
Traceback (most recent call last):
  File "./inverter_test.py", line 18, in <module>
    asyncio.run(
  File "/usr/local/lib/python3.8/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/config/custom_components/goodwe/goodwe_inverter.py", line 386, in execute
    raise InverterError(
custom_components.goodwe.goodwe_inverter.InverterError: No valid response received to '7f03753100280409' request

peternijssen, 2021-01-10:
I am trying to see if I can aid in this. 

I have a GW2000-SN and I am wondering if I should fill in the Serial Number of the device within the other library; https://github.com/XtheOne/Inverter-Data-Logger. (Due to having a wifi stick).

Anyway, I was unable to get the ScanLoggers.py working in OSX due to error `socket.error: [Errno 49] Can't assign requested address`. 
Problem was within this line: https://github.com/XtheOne/Inverter-Data-Logger/blob/master/ScanLoggers.py#L46
After I replaced `InverterLib.getNetworkIp()` with the IP of my laptop, I was able to run the ScanLoggers.py.

Afterwards I ran the ScanLoggers.py with the following result;
```
This script will look for iGEN WiFi Kit loggers from SolarMAN PV
List their IPs and S/Ns and connected inverters S/N
These loggers are found in Omnik, Hosola, Ginlong, Kstar, Seasun, SolaX, Samil, Sofar, Trannergy
and other Solar inverters
closing socket, scanning done!
```

Unfortunately, no results. I noticed that there was a second branch in the repository and after trying that, I got this result;
```
This script will look for iGEN WiFi Kit loggers from SolarMAN PV
List their IPs and S/Ns and connected inverters S/N
These loggers are found in Omnik, Hosola, Ginlong, Kstar, Seasun, SolaX, Samil, Sofar, Trannergy
and other Solar inverters
WiFi kit logger found, IP = b'192.168.2.124' and S/N = b'Solar-rak\x00\xff\xff\xff\xff\xff\xff'
closing socket, scanning done!
Traceback (most recent call last):
  File "ScanLoggers.py", line 70, in <module>
    data = InverterLib.createV5RequestFrame(int(logger_sn))
ValueError: invalid literal for int() with base 10: b'Solar-rak\x00\xff\xff\xff\xff\xff\xff'
```

I am wondering now if I can fix the final error. Will take a closer look later today hopefully.

mletenay, 2021-01-14:
Any luck with the NS inverter ?
I assume you have tried both PvMaster and SolarGo apps with no success, right ?

rutgerputter, 2021-01-15:
I've just tried the same as @peternijssen and got the same results:

`This script will look for iGEN WiFi Kit loggers from SolarMAN PV
List their IPs and S/Ns and connected inverters S/N
These loggers are found in Omnik, Hosola, Ginlong, Kstar, Seasun, SolaX, Samil, Sofar, Trannergy
and other Solar inverters
WiFi kit logger found, IP = b'192.168.9.20' and S/N = b'Solar-WiFi18BW0053'
closing socket, scanning done!
Traceback (most recent call last):
  File "./ScanLoggers.py", line 69, in <module>
    data = InverterLib.createV4RequestFrame(int(logger_sn))
ValueError: invalid literal for int() with base 10: b'Solar-WiFi18BW0053'
`

But my python skills aren't good enough to solve this. The InverterLib expects a int value and out SN is clearly more of a string type.

peternijssen, 2021-01-17:
I unfortunately was not able to get any further. PVMaster and SolarGo aren't really working, so I expect I am unable to set up this connection.

@rutgerputter; your serial number seems to make more sense. You could replace `int(logger_sn)` on line 69 i believe with `int.from_bytes(logger_sn, "big")`. Not sure of it should be `big` or `little`.

You could also test the lib with using serial no `18BW0053` or `Solar-WiFi18BW0053`. Purely a guess from my side, but at least your output seems to make more sense.

mletenay, 2021-01-19:
@rutgerputter @peternijssen I have enhanced the inverter_scan.py from this repo to discover the inverter and try to initiate all known protocols.
Could you please execute that on your machines and publish the output ?

TimSoethout, 2021-01-20:
```
$python3 inverter_scan.py
2021-01-20 09:14:04,400 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-20 09:14:04,400 search_inverters(603) - DEBUG: Searching inverters by broadcast to port 48899
2021-01-20 09:14:04,401 connection_made(316) - DEBUG: Send: '574946494b49542d3231343032382d52454144'
2021-01-20 09:14:05,202 datagram_received(327) - DEBUG: Received: '3139322e3136382e312e3132322c3938443836333041463938342c536f6c61722d576946693138385730333036'
Located inverter at IP: 192.168.1.122, mac: 98D8XXXXX, name: Solar-WiFi188WXXXX
Trying command: AA55C07F0102000241
2021-01-20 09:14:05,204 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-20 09:14:05,204 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-01-20 09:14:07,205 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-01-20 09:14:07,205 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-01-20 09:14:09,207 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-01-20 09:14:09,208 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-01-20 09:14:11,210 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-01-20 09:14:11,210 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-01-20 09:14:13,214 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to AA55C07F0102000241 command
Trying command: 7F03753100280409
2021-01-20 09:14:13,216 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-20 09:14:13,216 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-01-20 09:14:15,217 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-01-20 09:14:15,217 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-01-20 09:14:17,221 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-01-20 09:14:17,222 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-01-20 09:14:19,227 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-01-20 09:14:19,227 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-01-20 09:14:21,230 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to 7F03753100280409 command
Trying command: 197d0001000dff045e50303036564657f6e60d
2021-01-20 09:14:21,231 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-20 09:14:21,232 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-01-20 09:14:23,237 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-01-20 09:14:23,238 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-01-20 09:14:25,240 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-01-20 09:14:25,240 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-01-20 09:14:27,243 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-01-20 09:14:27,244 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-01-20 09:14:29,247 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to 197d0001000dff045e50303036564657f6e60d command
Trying command: 680241b131383857303330360100b616
2021-01-20 09:14:29,249 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-20 09:14:29,249 connection_made(316) - DEBUG: Send: '680241b131383857303330360100b616'
2021-01-20 09:14:31,250 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-01-20 09:14:31,250 connection_made(316) - DEBUG: Send: '680241b131383857303330360100b616'
2021-01-20 09:14:33,254 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-01-20 09:14:33,254 connection_made(316) - DEBUG: Send: '680241b131383857303330360100b616'
2021-01-20 09:14:35,259 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-01-20 09:14:35,259 connection_made(316) - DEBUG: Send: '680241b131383857303330360100b616'
2021-01-20 09:14:37,264 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to 680241b131383857303330360100b616 command
Trying command: 680241b136303330573838310100b616
2021-01-20 09:14:37,266 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-20 09:14:37,267 connection_made(316) - DEBUG: Send: '680241b136303330573838310100b616'
2021-01-20 09:14:39,271 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-01-20 09:14:39,271 connection_made(316) - DEBUG: Send: '680241b136303330573838310100b616'
2021-01-20 09:14:41,275 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-01-20 09:14:41,276 connection_made(316) - DEBUG: Send: '680241b136303330573838310100b616'
2021-01-20 09:14:43,280 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-01-20 09:14:43,281 connection_made(316) - DEBUG: Send: '680241b136303330573838310100b616'
2021-01-20 09:14:45,283 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to 680241b136303330573838310100b616 command
Identifying inverter at IP: 192.168.1.122
2021-01-20 09:14:45,285 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-20 09:14:45,285 discover(636) - DEBUG: Probing inverter at 192.168.1.122:8899
2021-01-20 09:14:45,286 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-01-20 09:14:47,291 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-01-20 09:14:47,291 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-01-20 09:14:49,297 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-01-20 09:14:49,297 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
[... 10 lines cut ...]
2021-01-20 09:14:59,314 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-01-20 09:15:01,319 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
2021-01-20 09:15:01,319 discover(656) - DEBUG: Probing ET inverter at 192.168.1.122:8899
[... cut ...]
  File "/Users/tim/Downloads/home-assistant-goodwe-inverter/custom_components/goodwe/goodwe_inverter.py", line 667, in discover
    raise InverterError(
custom_components.goodwe.goodwe_inverter.InverterError: Unable to connect to the inverter at host=192.168.1.122 port=8899, or your inverter is not supported yet.
Failures=[InverterError("No valid response received to 'aa55c07f0102000241' request"), InverterError("No valid response received to 'aa55c07f0102000241' request"), InverterError("No valid response received to 'f70388b800213ac1' request")]
```

Not working yet? Goodwe GW3000-NS

mletenay, 2021-01-20:
Hmmm, that's bad.
None of the known protocols passed, not even the attempts to "omnik".
The real omnik protocol expects serial number to be plain numeric, but goodwee has these alphanumeric.
I made 2 guesses there, but either the NS does don speek omnik protocol at all, or the serial is passed in request in different way ...

peternijssen, 2021-01-21:
I got a complete different error on my GW2000-NS

```
2021-01-21 10:18:02,690 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-21 10:18:02,690 search_inverters(603) - DEBUG: Searching inverters by broadcast to port 48899
2021-01-21 10:18:02,691 connection_made(316) - DEBUG: Send: '574946494b49542d3231343032382d52454144'
2021-01-21 10:18:02,755 datagram_received(327) - DEBUG: Received: '3139322e3136382e322e3132342c3630433541383641363034422c536f6c61722d72616b00ffffffffffff'
Traceback (most recent call last):
  File "inverter_scan.py", line 42, in <module>
    result = asyncio.run(search_inverters()).decode("utf-8").split(",")
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 37: invalid start byte
```

I've changed the code slightly: `result = asyncio.run(search_inverters()).split(b",")`
Hope that this is correct.

Now I get the following output;
```
2021-01-21 10:22:26,359 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-21 10:22:26,359 search_inverters(603) - DEBUG: Searching inverters by broadcast to port 48899
2021-01-21 10:22:26,360 connection_made(316) - DEBUG: Send: '574946494b49542d3231343032382d52454144'
2021-01-21 10:22:26,460 datagram_received(327) - DEBUG: Received: '3139322e3136382e322e3132342c3630433541383641363034422c536f6c61722d72616b00ffffffffffff'
Located inverter at IP: b'192.168.2.124', mac: b'60C5A86A604B', name: b'Solar-rak\x00\xff\xff\xff\xff\xff\xff'
Trying command: AA55C07F0102000241
2021-01-21 10:22:26,461 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-21 10:22:26,465 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-01-21 10:22:28,470 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-01-21 10:22:28,470 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-01-21 10:22:30,472 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-01-21 10:22:30,472 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-01-21 10:22:32,472 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-01-21 10:22:32,473 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-01-21 10:22:34,473 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to AA55C07F0102000241 command
Trying command: 7F03753100280409
2021-01-21 10:22:34,475 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-21 10:22:34,476 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-01-21 10:22:36,476 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-01-21 10:22:36,477 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-01-21 10:22:38,482 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-01-21 10:22:38,482 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-01-21 10:22:40,488 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-01-21 10:22:40,488 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-01-21 10:22:42,493 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to 7F03753100280409 command
Trying command: 197d0001000dff045e50303036564657f6e60d
2021-01-21 10:22:42,495 __init__(59) - DEBUG: Using selector: KqueueSelector
2021-01-21 10:22:42,496 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-01-21 10:22:44,501 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-01-21 10:22:44,501 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-01-21 10:22:46,507 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-01-21 10:22:46,507 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-01-21 10:22:48,512 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-01-21 10:22:48,512 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-01-21 10:22:50,514 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to 197d0001000dff045e50303036564657f6e60d command
Traceback (most recent call last):
  File "inverter_scan.py", line 53, in <module>
    sn = bytes(result[2][10:], 'utf-8').hex()
TypeError: encoding without a string argument
```

rutgerputter, 2021-01-21:
Running in the same issue:

`2021-01-21 12:26:34,530 __init__(59) - DEBUG: Using selector: EpollSelector
2021-01-21 12:26:34,531 search_inverters(603) - DEBUG: Searching inverters by broadcast to port 48899
2021-01-21 12:26:34,532 connection_made(316) - DEBUG: Send: '574946494b49542d3231343032382d52454144'
2021-01-21 12:26:36,535 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-01-21 12:26:36,535 connection_made(316) - DEBUG: Send: '574946494b49542d3231343032382d52454144'
2021-01-21 12:26:38,536 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-01-21 12:26:38,536 connection_made(316) - DEBUG: Send: '574946494b49542d3231343032382d52454144'
2021-01-21 12:26:40,539 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-01-21 12:26:40,539 connection_made(316) - DEBUG: Send: '574946494b49542d3231343032382d52454144'
2021-01-21 12:26:42,542 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
Traceback (most recent call last):
  File "./inverter_scan.py", line 42, in <module>
    result = asyncio.run(search_inverters()).decode("utf-8").split(",")
  File "/usr/lib/python3.8/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/rputter/Downloads/home-assistant-goodwe-inverter-master/custom_components/goodwe/goodwe_inverter.py", line 623, in search_inverters
    raise InverterError("No valid response received to broadcast request") from None
custom_components.goodwe.goodwe_inverter.InverterError: No valid response received to broadcast request
`

Maybe the sytax is wrong?  Also tried the result = asyncio.run(search_inverters()).split(b",") line but no dice.

mletenay, 2021-01-21:
Well, I'm running out of ideas. As said before there are 2 possibilities here:
1) the NS inverters do not support any protocol for retrieval inverter data at all and the wifi is just one-way inverter->sems portal.
2) the NS inverters do support some protocol, we just do not know which. 
The protocols of ET or ES/EM inverter families do not work, that's proven now.
That leaves the only other known protocol on the scene - "omnik".
Standard (non-goodwe) implementation assumes (plain numeric inverter) serialnr being present in udp request payload.
@TimSoethout reports serial in form 188WXXXX, which is not a plain decimal number.
@peternijssen reports even more odd value - its actually 0x00ffffffffffff.
@rutgerputter - your inverter did not even responds to udp broadcast and was not found at all.

So if we want to believe possibility 2 is the correct one, we need to figure out how to construct the initial udp request and what to put to those 8 bytes meant for inverter address (serialnr).

Maybe if you can sniff (e.g. wireshark) the communication between inverter and the SEMS portal, we can figure out something ?

peternijssen, 2021-01-24:
I am willing to dive into Wireshark, just need to figure how it exactly works 😅  I do doubt though if we are able to find something. The strange different behaviors between all those devices is not really giving me any trust in a solution :(

ThinkPadNL, 2021-02-04:
For the NS-series just use [GoodWeLogger](https://github.com/jantenhove/GoodWeLogger) which connects to the RS485 port of the inverter. Works fine for over 3 years here now. Downside is that you cannot use the original wifi-module anymore and thus will lose logging to Goodwe SEMS Portal. But who needs that when you have PVOutput 😌

mlakerveld, 2021-08-10:
My GW3600D-NS is now supported, thanks to @MathyV's pull request #27.
Any other people that can verify?

peternijssen, 2021-08-11:
Still not working for me unfortunately. My firmware version:
```
Current version: 1.17.16.40
```

mlakerveld, 2021-08-11:
I'm guessing that is because you have the older NS model, not the D-NS model.
If it doesnt work in the SolarGo App I don't think older NS support is going to happen.

I think this means NS support doesn't seem likely, but D-NS support is now here, if you are on the correct firmware.

mletenay, 2021-08-22:
I'm closing this issue, as it seems the NS inverters do not support any applicable communication protocol and there's nothing we can do about.
But as mention above the D-NS inverters (at least with some firmwares) are now supported.

closed 2021-08-22 by mletenay

sergutpal, 2021-10-29:
> > Hi, I can confirm that home-assistant-goodwe-inverter component works correctly in a GW3000D-NS but only after updating the firmware to version 1.51.18. With the original firmware version it did not work. Thank you very much!
> 
> Sorry to be difficult, but is there an easy way to get the firmware? Or do I need to do contact my installer?

Hi zeddD1abl0, I asked my installer to upgrade the firmware of my GW3000D-NS and he do it by OTA easily. I'm sorry, but I don't know how to upgrade the firmware myself

zeddD1abl0, 2021-11-13:
So, just an update on this. Called GoodWe Australia and was told "We don't update inverters unless required to for grid reasons" by the grumpiest support agent I've ever dealt with. So, in Australia, you're probably out of luck unfortunately.

draxxy9, 2022-09-19:
> So, just an update on this. Called GoodWe Australia and was told "We don't update inverters unless required to for grid reasons" by the grumpiest support agent I've ever dealt with. So, in Australia, you're probably out of luck unfortunately.

@zeddD1abl0 I know it's been a while since you posted this, but I'm also in Australia and was able to get an ARM firmware update for my inverter.
The GoodWe customer support person said that she was unable to complete the update for me, but provided me the number for GoodWe technical support, and they were able to complete it once I provided the S/N of the inverter (once it was connected to SEMS).

-> Call (03) 9918 3905 
-> Select option 2 
-> Select option 1
-> Advise them that you are trying to connect to your inverter using the SolarGo app, but it is telling you that you need at least version 14, but you are currently on version x (you can see which ARM version you have from the LCD panel on the inverter, it's the last segment of digits in the version listed there). 

If they ask why you need SolarGo, tell them you are trying to set up DRED (Demand response enabling device) with a CT sensor, but the manual says you need to enable that through the SolarGo app (technically true: https://www.goodwe.com.au/Ftp/EN/Downloads/User%20Manual/GW_DNS_User%20Manual-EN.pdf). Otherwise they may suggest you just use the SEMS app.

Hope that helps someone else.

ifm90, 2023-01-26:
> Called GoodWe Australia and was told "We don't update inverters unless required to for grid reasons" by the grumpiest support agent I've ever dealt with

found [this document](https://www.austraenergy.com.au/wp-content/uploads/2021/12/Goodwe-Onsite-Upgrading-via-SolarGo-app-V1.0.pdf) which mentions "After 18th of Dec. 2021, all inverters installed in Australia must be complied with AS/NZS4777.2:2020 standard"

could you ask the support agent that you wish to comply with the standard as yours was installed before this date?

they should be able to do an OTA update as under the section **solution for inverters not with latest standard** it says "Goodwe still recommends to connect inverter to wifi so Goodwe can upgrade the firmware remotely"

registro27, 2023-04-26:
Hi

I have a GW1500-NS and my inverter responds to the udp broadcast, but it doesn't respond to any udp command to retrieve data.
I have managed to capture the GW1500-NS traffic with tcp.goodwe-power.com using wireshark. It sends every minute a tcp packet with:
- First 6 bytes: the text POSTGW
- Next 8 bytes: Unkwnown but usually fixed (HEX): 00 00 00 f9 01 04 00 00. Sometimes f9 is d9 and 04 is 45
- Next 16 bytes: My inverter identifier in plain text
- 1 hex byte: Last two digits of Year
- 1 hex byte: Month
- 1 hex byte: Day
- 1 hex byte: Hour
- 1 hex byte: Minute
- 1 hex byte: Second (but it is not exactly the one shown at the full report at sems portal)
- 10 bytes 00
- 6 bytes with (usually) the same date using the same format. 
- 210 bytes of unknown data. It seems that it is somehow encrypted because two consequent packets has nothing to do each other. Sometimes a packet contains 32 more data bytes (242 bytes instead of 210).

tcp.goodwe-power.com responds with a packet:
- 2 bytes: the text GW
- 5 fixed bytes (HEX): 00 00 00 31 01
- 1 byte. Several times is 04 (HEX), but sometimes is 45 (HEX)
- 16 bytes with my inverter identifier in plain text
- 6 bytes with the date using the previously described format
- 10 bytes 00
- 18 bytes of unknown data. It seems that is also somehow encrypted as it is totally diferent each time

I can't figure out how is the data encrypted so there is nothing more I can do.
One thing is for sure: The inverter sends the data without any external request: Every minute since it starts up until looses power. So I agree: NS inverters do not support any applicable communication protocol and there's nothing we can do about.

I hope this helps someone

Regards

Abdull, 2024-02-15:
https://smlx.dev/posts/goodwe-sems-protocol-teardown/

TimSoethout, 2024-07-12:
I recently asked Goodwe to update my inverter's (GW3000-NS installed in 2018) ARM and DSP firmware via service.nl@goodwe.com . This was done quickly and suddenly the SolarGo could actually connect.

Also this component suddenly started working for me. :D Including the on/off buttons, which come in handy with negative energy rates.