mletenay/home-assistant-goodwe-inverter #4: Issue with more than 1 inverter.

wishie, 2020-09-21:
I have 2 x GoodWe GW5048-EM inverters. When setting them up in Home Assistant with the following config:

```
sensor:                                                  
  - platform: goodwe                                     
    ip_address: 192.168.100.6                            
    scan_interval: 30                                    
    sensor_name_prefix: Inverter1                  
  - platform: goodwe                                     
    ip_address: 192.168.100.22                     
    scan_interval: 30                                    
    sensor_name_prefix: Inverter2
```

I do not get 2 sets of sensors as expected.

Below is list of sensors found in HA for the above config:
[image]

wishie, 2020-09-21:
@mletenay perhaps you can just add your single inverter twice, but using a different sensor_name_prefix for each.. that would simulate 2 inverters well enough for this testing, no?

wishie, 2020-09-22:
Oddly enough.. I updated the code this morning with your latest version and 2 inverters seem to be working.

mletenay, 2020-09-23:
I believe I have found a problem with the multiple inverter setup.
The required change was however substantial and a lot of internal code was redesigned, leading to a breaking change.
The way how individual sensors get their unique_id and entity_id values was changed.
Now they are properly "multi instance" capable and more stable (the default entity names no longer play a role there and can be easily changed).
However, the unfortunate side-effect means the sensors will now get different uuids, so if you already created custom automations and/or UI tweaks, you have to re-apply them again, now with proper/stable IDs.

closed 2020-09-23 by mletenay