mletenay/home-assistant-goodwe-inverter #5: Wonderful work! Just a few parsing errors

gbyleveldt, 2020-09-21:
Hi,

Thank you very much for all the work you've done so far, it saved me a heap of trouble getting my **_ES_** inverter integrated into Home Assistant. I've done a lot of work on my side to decode the protocol, but it would appear that you have at least some documentation (I had none and had to sit for days to reverse engineer the protocol). It's a relief to see that I wasn't too far off 👍 

I do have a running system here (not sure if you do?) so I was able to run the inverter_test.py script to verify your output against my system. I have picked up a few things though, hopefully it can be of assistance. I attach a data dump off my running inverter for reference:

aa557fc001868c0bcc003a02000000000002150001005000f70000000a00640000640064640100000108c6004c011e13830108c6003003ca1383010201b10000000000001fd900000457007600e100003204069b0100300203dc00010000fd840008002000001409150e260f0000000000000000000000000000000000000000000000000000090b0000081f020000000000001334

1. At byte offset 16 in your code, you state Battery Voltage 4. This is actually the battery temperature as reported by the BMS.
2. You make reference to up to 3 batteries on the inverter; this inverter can only manage one BMS? Which leads me to byte offset 28 - Battery SOC 3. There is only 1 BMS on my system but you parse it as showing 100%?
3. Offset 33 you suspect is Meter Status. I can confirm that it comes up as 0x01 with an EzMeter connected. I need to verify this changes to 0x00 when the meter is disconnected. I'm not sure which other states there may be.
4. For the work mode, you need to change your offset from byte 41 to byte 42 in order to parse the working mode of the inverter correctly.
5. Load mode, offset 51. Can you please confirm you parsing this correctly? You state that 0x00 is working and 0x01 is load disconnected. On my live system that return 0x01, and load is connected and working
6. From offset 59 onwards, you are parsing the plant's daily and total production/consumption. Unfortunately this is not parsing correctly, so I suspect there may be some misinterpretation of whatever docs you have? I will gladly supply some screenshots and data dumps in order to help with this, just not sure what you need in order to get this resolved?

Thanks again for all the work!

mletenay, 2020-09-21:
Hi,
I've started with similar idea like yours, first sniffing packets my ET inverter is sending to SEMS portal.
But they turned out to be encrypted, so I then focused on the UDP protocol used by the mobile application.
Kudos to your work, I was not that successful with plain data decoding.
So I decompiled the app and was able to read/understand reasonable amount of code/data (that's for my "documentation" ;-).
But to your points:
1. You're right, they seem to save that value into variable named vbattery4, but on UI it is presented as temperature. Fixed.
2. vbattery2 and vbattery3 seem to be read but I could not find any reference to them in UI or anywhere, so probably just dummies.
3. 33 is meter status indeed.
4. You're right, it was my typo, 42 it is. Fixed.
5. I've re-checked the coding of 51 and it seems to be right 0 - connected, 1 not-connected
6. I'm still not sure about interpretation of the e_total, h_total, e_day, e_load_day ... values, they still need bit more analysis. What I just did is that the variables values are now directly put into sensor definition tuples.

If you can compare and match the reported values to the app or sems data, that would help a lot, since I have the ET inverter which used different UDP protocol flavor ...

wishie, 2020-09-22:
I will chime in with my EM inverter(s) which I believe use the same protocol as the ES. I will fetch whatever data I can to assist.

wishie, 2020-09-22:
I've started to add sensors to HA. I can't seen to find a 'daily export' (e_day perhaps) and 'daily import (i_day perhaps) sensors..

I am just trying to match values shown in HA with values shown in SEMS.

gbyleveldt, 2020-09-23:
Ok, almost there. See notes on screenshot
[image]

These are the changes I made to the ES side code. I've kept the incorrect descriptions and shown what they should be in the screenshot. I didn't want to change it yet as I don't want to break HA with Friendly name changes yet.

# Sensors ("id, offset, getter, unit, name, icon")
    __sensors = (
        (0, _read_voltage, "V", "PV1 Voltage", _ICON_PV),
        (2, _read_current, "A", "PV1 Current", _ICON_PV),
        (
            -2,
            lambda data, x: round(_read_voltage(data, 0) * _read_current(data, 2)),
            "W", "PV1 Power", _ICON_PV,
        ),
        (4, _read_pv_mode1, "", "PV1 Mode", _ICON_PV),
        (5, _read_voltage, "V", "PV2 Voltage", _ICON_PV),
        (7, _read_current, "A", "PV2 Current", _ICON_PV),
        (
            -7,
            lambda data, x: round(_read_voltage(data, 5) * _read_current(data, 7)),
            "W", "PV2 Power", _ICON_PV,
        ),
        (9, _read_pv_mode1, "", "PV2 Mode", _ICON_PV),
        (
            -9,
            lambda data, x: round(_read_voltage(data, 0) * _read_current(data, 2)) + round(_read_voltage(data, 5) * _read_current(data, 7)),
            "W", "Total PV Power", _ICON_PV,
        ),
        (10, _read_voltage, "V", "Battery Voltage", _ICON_BATT),
        #(12, _read_voltage, "V", "Battery Voltage 2", _ICON_BATT),
        #(14, _read_voltage, "V", "Battery Voltage 3", _ICON_BATT),
        (16, _read_temp, "C", "Battery Temperature", None),
        (18, _read_current, "A", "Battery Current", _ICON_BATT),
        (20, _read_bytes2, "A", "Battery Charge Limit", _ICON_BATT),
        (22, _read_bytes2, "A", "Battery Discharge Limit", _ICON_BATT),
        # BMS status 24-25
        (26, _read_byte, "%", "Battery State of Charge", _ICON_BATT),
        #(27, _read_byte, "%", "Battery State of Charge 2", _ICON_BATT),
        #(28, _read_byte, "%", "Battery State of Charge 3", _ICON_BATT),
        (29, _read_byte, "%", "Battery State of Health", _ICON_BATT),
        (30, _read_battery_mode1, "", "Battery Mode", _ICON_BATT),
        # BMS warning 31-32
        # Meter status 33
        (34, _read_voltage, "V", "On-grid Voltage", _ICON_AC),
        (36, _read_current, "A", "On-grid Current", _ICON_AC),
        (38, _read_power2, "W", "On-grid Power", _ICON_AC),
        (40, _read_freq, "Hz", "On-grid Frequency", _ICON_AC),
        (42, _read_work_mode1, "", "Work Mode", None),
        (43, _read_voltage, "V", "Back-up Voltage", _ICON_AC_BACK),
        (45, _read_current, "A", "Back-up Current", _ICON_AC_BACK),
        (47, _read_power2, "W", "Back-up Power", _ICON_AC_BACK),
        (49, _read_freq, "Hz", "Back-up Frequency", _ICON_AC_BACK),
        (51, _read_load_mode1, "", "Load Mode", None),
        (52, _read_energy_mode1, "", "Energy Mode", None),
        (53, _read_temp, "C", "Inverter Temperature", None),
        (55, _read_bytes4, "", "Error Codes", None),
        (59, _read_power_k, "kW", "Total Energy", None),
        # htotal 63-66
        (67, _read_power_k2, "kW", "Today's Energy", None),
        (69, _read_power_k2, "kW", "Today's Load", None),
        (71, _read_power_k, "kW", "Total Load", None),
        (75, _read_power2, "W", "Total Power", None),
        # Effective work mode 77
        # Effective relay control 78-79
        (80, _read_grid_mode1, '', 'On-grid Mode', None),
        (81, _read_power2, "W", "Load Power (Back-up)", None),
        (
            -81,
            lambda data, x: round(_read_power2(data, 47) + _read_power2(data, 81)),
            "W", "Plant Power", None,
        ),
        (89, _read_bytes4, "", "Diag Status", None),
    )

wishie, 2020-09-23:
Any indication of if there is a 'imported today' and 'exported today' counter at all? as well as PV selfUse and buy/sell percentages? These are present on SEMS, so must come from somewhere.

gbyleveldt, 2020-09-23:
There are daily and total counters for PV and Load consumption, see attached. The descriptions need some work, but like I said above I don't want to break HA yet with Friendly name changes. For the most part, the various variables are figured out, it's just a matter of verifying it (which seems ok so far)

[image]

What I haven't found yet is usage ratios. There are still some values in the data packet that I've not been able to identify so maybe those are usage ratios. It's also possible that SEMS takes the various consumption figures and calculates the usage ratios itself. Hard to say at the moment to be honest.

wishie, 2020-09-23:
The PC generation is important, but what is more important (to me) is PV selfUse and PV export. Also, not just consumed load, but how much of that load was from the grid.

Do you have some other way I can contact you quickly instead of posting a bunch of stuff in here?

mletenay, 2020-09-23:
I have created a discord server, the link is in the repository info or here - https://discord.gg/TaXyWXT
As for the changes/fixes to sensors, I will probably create a breaking change in the code, to move away from offsets to the new "names" in sensor tuples. It should make the home-assistant integration more robust and enable to add more "calculated" sensors.

mletenay, 2020-09-26:
I believe all the known sensors are being parsed correctly now, closing.

closed 2020-09-26 by mletenay