mletenay/home-assistant-goodwe-inverter #6: Show sensor positv and negativ

kolossboss, 2020-09-21:
Hi,
thanks for the fast reply to my previews request.

I have a few other things that would be nice.

1. The "sensor.goodwe_on_grid_power" show only positiv values. It would be nice. If "gets power grid" is negativ and "sends power grid" is positiv.

2. There is no sensor for the complete house consumption. It would be nice if its possible to see this stats ([screenshot](https://user-images.githubusercontent.com/11015086/93781492-b93f1800-fc29-11ea-834c-420915fced26.PNG))

3. For any reasons. The "todays energy" and "total energy" did not show the same value like the Sems Portal.

4. The "sensor.goodwe_total_power" show value in KW and not in W. And If the value ist 0 the graph goes crazy in HA
[image]

Thanks a lot again for your hard work !!!

gbyleveldt, 2020-09-21:
See the previous ticket that I've opened. The power stats aren't being properly parsed yet. I believe the author based some of his parsing on my (admittedly imperfect) reverse engineering that I published on Power Forum.

To answer your first question though, the On Grid Power value is a reading coming directly from the EzMeter directly. It doesn't have a sign (+-). You have to look at the Grid Mode to see if it's importing or exporting. I do agree it would be better to include that in the sensor value though (importing value can have a - in front of it and exporting can have a + in front of it, like it appears on the SEMS portal). I've already made those changes on my implementation of the code but ideally let's leave it to the author to properly implement it

mletenay, 2020-09-22:
Ad 1.
The ET inverter seem to deliver the +/- power with its "active_power" value (and then deducts "grid_in_out" from the sign).
The ES provides the "grid_in_out" value explicitly, so I've created new "active_power" sensor also for ES inverter (as sign * "pgrid").
Ad 2.
Am I correct the formula for "house consumption" should "ppv + pload - active_power" (i.e. PV Power + Back-up Power - (signed) On-grid Power) ?

kolossboss, 2020-09-22:
1. sounds great

2. Yes thats correct!

gbyleveldt, 2020-09-22:
Ok, I've just quickly messed some more with it:

Offset 38 is the reading directly from the EzMeter.
Offset 47 is actually the grid side load power (W), not the backup side.
Offset 81 you're not currently parsing, that's the backup side load power (W).
Add those two together and you get the plant (house) consumption (less around 50W, which I suspect is the inverter dissipation).

I'll try a few more examples to be sure, but so far the above looks correct. I've slightly modified the code on my side as a test

wishie, 2020-09-22:
The important data points for me are (from SEMS)

```
[energeStatisticsCharts] => Array
        (
            [sum] => 0.5
            [buy] => 8.6
            [buyPercent] => 94.8
            [sell] => 0
            [sellPercent] => 6
            [selfUseOfPv] => 0.5
            [consumptionOfLoad] => 9.09
```
So I can see how much I have bought, sold, and consumed.

I also heavily used the 'power flow' section to get loadStatus, pvStatus and batteryStatus to know which way the power was flowing. These should all be available somehow from the UDP packets.

mletenay, 2020-09-26:
New sensor "house_consumption" was added, rest of items already fixed or moved to other issues I believe.

closed 2020-09-26 by mletenay