mletenay/home-assistant-goodwe-inverter #7: House consumption is not correct calculated

kolossboss, 2020-09-29:
Hi,

I think the `sensor.goodwe_house_consumption` is not correct calculated. 
It should be  `sensor.goodwe_ppv - sensor.goodwe_pbattery1 + sensor.goodwe_pgrid`
[image]

Also,
I'm not completly sure what sensor.goodwe_total_power is, but I think it should be shown in W and not KW

mletenay, 2020-09-29:
The pull request was merged, thanks.
Regarding the house consumption, I'm really sure.
The formula is at least for the ET inverters correct: 
_pbattery_ is signed value - when charging battery it is negative, when discharging, positive.
_active_power_ (the EzMeter value) is signed as well - negative when importing, positive when exporting to grid.
It is possible the ES/EM inverters provide different values res. sign.
Is _pbattery1_ res. _ibattery1_ always positive or are those signed values depending on charging status ?

kolossboss, 2020-09-29:
Ah ok, I see the problem here. pbattery1 and ibattery1 are always positiv. 
Maybe you can change the sign depending on battery_mode (2 discharge and 3 charge).
Or you can change the sign in the calculation depending on battery mode.

mletenay, 2020-09-29:
Seems the ET protocol is a bit smarter than ES/EM.
It would be great if someone can run the inverter_test.py and collect the raw UDP response in various inverter conditions - battery charging/discharging, grid import/export etc.
I will then adjust the code so it provide proper data (and ideally some unit tests).

mletenay, 2020-09-29:
I have added the signs to ibattery1, pbattery1 and pgrid.
Now they make more sense, are compatible with ET inverters and as consequence provide correct house_consumption.

closed 2020-09-29 by mletenay

kolossboss, 2020-09-30:
Thanks a lot. I try to run the inverter_test.py today.

kolossboss, 2020-09-30:
Hi,

the calculation is still not correct. When battery is charging pbattery is positiv and when discharging its negativ.

mletenay, 2020-09-30:
Aaah. You're of course right. I messed the sign in battery values (used same condition as in grid, which is wrong).
Fixed now.