mletenay/home-assistant-goodwe-inverter #9: Incorrect value for sensor.goodwe.pgrid

kolossboss, 2020-10-02:
Hi,

I think the `pgrid sensor` did not show the right value. 

As I understand it, the sensor shows the value of `sensor.goodwe_pload` with the sign depending on `sensor.goodwe_work_mode`

Also the sensor work_mode did not change status anymore.

[image]

mletenay, 2020-10-02:
Naming is a bit confusing, but no, pgrid and pload are two different measurements.
pgrid is the power (flowing in or out the grid) as reported by EzMeter.
pload is the power inverter is producing for its on-grid side.
pgrid may now be negative/positive, pload is always positive.
The energy mode is again different measurement one of (with weird name).
Look at energy_mode_label sensor, there is human readable label of reported value 2.
One of": 
    0: "Check Mode",
    1: "Wait Mode",
    2: "Normal(On-Grid)",
    4: "Normal(Off-Grid)",
    8: "Flash Mode",
    16: "Fault Mode",
    32: "Battery Standby",
    64: "Battery Charging",
    128: "Battery Discharging",

kolossboss, 2020-10-02:
Thanks for clarifying this.

But also, the value of `pgrid` probably not correct.
It should be the same value like the `pmeter` in the SEMS integration. And like this in the SEMS Portal. But its not. Its always low then `pmeter`. 

[image]
[image]

mletenay, 2020-10-02:
I'm confused a bit.
From the SEMS picture I would assume the pgrid should be -121, pload = 360 and pbattery1 -24.
What are the reported values ? And which inverter model do you have ?

kolossboss, 2020-10-02:
Sorry. `PV Leistung von/zu Netz` is the pmeter sensor from the SEMS integration.
My Model is: GW3648-EM

I think pgrid should be -106, but its not.

[image]
[image]

mletenay, 2020-10-02:
Hmm, those numbers do no match at all.
I can't find problem with the data from the UDP packets and the discrepancies are probably caused by big latency of SEMS.
The time between measurement, its transfer from inverter to sems portal and processing there can be in several seconds or tens of seconds.
If the pv production is volatile (cloudy and/or changes in consumption), those two numbers can differ significantly.
I think we should try to compare the results of sems vs udp again in stable conditions (on sunny day).

kolossboss, 2020-10-02:
I executed the inverter_test.py to a completely different time.

But the Screens are at the same time, right after refresh. 
In HA I can see the graph. pgrid and pmeter (semsportal) differs over the  whole day on about 100W

mletenay, 2020-10-02:
I can't explain differences with sems data and there's probably not much we can do. 
But if there is a problem with decoding data from udp packets, the only way to identify them is to compare them with the immediate data from PvMastet app, not sems.

closed 2020-12-13 by mletenay