mletenay/home-assistant-goodwe-inverter #11: cannot connect to inverter via port

usman6062, 2020-10-06:
Hello there,
I am receiving the following error. 

custom_components.goodwe.goodwe_inverter.InverterError: Unable to connect to the inverter at host=x.x.x.x port=8899, or your inverter is not supported yet.

the model is GW3000-NS.  

Appreciate if someone can help me with this?

mletenay, 2020-10-06:
At the moment the component is known to work with ET, ES and EM types of inverters.
As you can see in issues #1 and #3, we do not know yet how to read data from NS or DT types of inverters.
You can check if you can read data via either PvMaster or SolarGo mobile applications, but reports so far indicate neither of those applications can retrieve runtime data from NS.
In theory the NS inverters may response to older protocol as implemented in https://github.com/XtheOne/Inverter-Data-Logger
You could try to test that, although again there are no positive reports there yet.

mletenay, 2021-08-22:
As mentioned in #3, unfortunately NS inverts (contrary to D-NS) do not support any applicable communication protocol, are not supported and there's nothing we can do about it :-(

closed 2021-08-22 by mletenay