mletenay/home-assistant-goodwe-inverter #12: Cannot import name 'DEVICE_CLASS_CURRENT'

nakoukal, 2020-10-08:
After update to last version :

 Platform error sensor.goodwe - cannot import name 'DEVICE_CLASS_CURRENT' from 'homeassistant.const' 

```

Logger: homeassistant.config
Source: custom_components/goodwe/sensor.py:10
First occurred: 11:57:06 (1 occurrences)
Last logged: 11:57:06

Platform error: sensor
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/config.py", line 804, in async_process_component_config
    platform = p_integration.get_platform(domain)
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/loader.py", line 390, in get_platform
    f"{self.pkg_path}.{platform_name}"
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/homeassistant/.homeassistant/custom_components/goodwe/sensor.py", line 10, in <module>
    from homeassistant.const import (
ImportError: cannot import name 'DEVICE_CLASS_CURRENT' from 'homeassistant.const' (/srv/homeassistant/lib/python3.7/site-packages/homeassistant/const.py)
```

My system info 👍 

arch | armv7l -- | --
dev | false
docker | false
hassio | false
installation_type | Home Assistant Core
os_name | Linux
os_version | 5.4.51-v7+
python_version | 3.7.3
timezone | Europe/Prague
version | 0.114.4
virtualenv | true

mletenay, 2020-10-08:
Weird.
The electrical device classes were added to HA here https://github.com/home-assistant/core/pull/36800 on Aug 8.
I would expect they should already be present in your 0.114.4 release.
Can you update to version 0.115 ?
(If not, I'll revert those changes, they are not so important at the moment.)

nakoukal, 2020-10-08:
yes thank you i updated /srv/homeassistant/lib/python3.7/site-packages/homeassistant/config.py and now it works in version 0.114.4

closed 2020-10-08 by nakoukal

mletenay, 2020-10-08:
Great, I'll update the hacs manifest file to properly require at least 0.115.0