mletenay/home-assistant-goodwe-inverter #14: Watt no kiloWatt

smithmarius, 2020-10-21:
Hi,

For the ES inverter line 996 should be
        Sensor("e_load_total", 71, _read_power_k, "W", "Total Load", None),
and not 
        Sensor("e_load_total", 71, _read_power_k, "kW", "Total Load", None),

in file 
home-assistant-goodwe-inverter/custom_components/goodwe/goodwe_inverter.py

mletenay, 2020-10-21:
Fixed.
But it's suspicious as it is 4 bytes value.
Does it provide correct values at all ?
(I don't have ES inverter so I can't verify that)

smithmarius, 2020-10-22:
Hi, Yes the values is correct but the display only of 'kW' is incorrect the value should be only 'W' . If I was using that much power I would have to get a Aurora Nuclear plant iso a solar installation lol . So the indicator should be W only. The 4 Bytes is only to improve accuracy , maybe also to comply with IEC screen shot before
[image]
after change..Thank you
[image]

gbyleveldt, 2020-10-22:
I need to go check to confirm, but that value SHOULD be kW. That’s the total amount of power the load has drawn since the inverter was installed. 2.6MW is only a few months worth of power.

smithmarius, 2020-10-22:
Sorry I was wrong it should be kWh not W . The "load" confused me

gbyleveldt, 2020-10-22:
No worries, I’m pretty sure that the units, as they stand, were correct prior to yesterday’s change. There’s a few of us that’s been using this integration for a few months already.

smithmarius, 2020-10-22:
> I need to go check to confirm, but that value SHOULD be kW. That’s the total amount of power the load has drawn since the inverter was installed. 2.6MW is only a few months worth of power.

Hi please note there is a difference between MW and MWh the one is snapshot(instantaneous) and the other can be cumulative .  The "h" is important. Just for info GoodweYES is my alias on pvoutput.

gbyleveldt, 2020-10-22:
Yes, you are correct of course. It’s a little detail issue that should still be fixed. But like I said above, we’ve all been beating our heads against the wall trying to validate the actual daily usage data from UDP as there’s still a discrepancy with SEMS

smithmarius, 2020-10-22:
I noticed the double count. The SEMS figure excludes the battery being used by inverter while this figured used on this option includes the battery being used as power and show it included.

smithmarius, 2020-10-22:
I am in ZA

gbyleveldt, 2020-10-22:
> I noticed the double count. The SEMS figure excludes the battery being used by inverter while this figured used on this option includes the battery being used as power and show it included.

I suspected as much, as the value shows generation in the morning prior to any PV being generated at all. Thing is, in my case, the value shows less generation by the end of the day when compared to SEMS. So, in my case, it’s not a simple matter of just subtracting the battery generation amount to get the PV generation amount. Like I said, this one is still a brain bashed.

You mind if I send you a PM on power forum? I’m having a serious fight with Goodwe because of random inverter restarts, and I just want to compare notes

smithmarius, 2020-10-22:
No issue see you there. I am 100% sorted on Goodwe no problems.

mletenay, 2020-10-22:
Unit fixed to kWh

closed 2020-10-22 by mletenay

smithmarius, 2020-10-23:
Thanks Martin working perfectly !