mletenay/home-assistant-goodwe-inverter #17: "Today Buy" and/or "Today Load Consumption" are missing for ET series

schamock, 2020-12-05:
Hi,
I have an ET series inverter and came across this project. With your script I can get the values for todays PV generation and the todays sell total (and indirectly todays self use as Gen - Sell).
But unfortunately the values for todays total buy and/or todays load consumption are missing (I can see them in the SEMS App).
Is there a way to get at least one of those values? On would be enough as Load Consumption = Self Use + Buy

Thanks a lot!

mletenay, 2020-12-08:
Well, such data are not directly available.
The inverter only provides "immediate" or snapshot values.
The only 2 exceptions are the daily and total PV production, for those the inverter maintains its cumulative  counters.
(The daily reset on midnight).
What SEMS portal displays is actually some aggregate function of these snapshot values kept in its own database.
However something similar can also be done on HomeAssistant level, I'll provide examples soon.

wishie, 2020-12-09:
You need to add sensors as integration sensors, and then use utility_meter integration to create the daily/monthly sensors..

```
- platform: integration                  
  source: sensor.goodwe_solar_total                                    
  name: energy_pv                        
  unit_prefix: k                         
  round: 3                               
- platform: integration                                                  
  source: sensor.goodwe_house_consumption
  name: energy_house                  
  unit_prefix: k             
  round: 3                                                            
- platform: integration       
  source: sensor.goodwe_pgrid                 
  name: energy_grid
  unit_prefix: k                                                         
  round: 3             

utility_meter:                                                                                                                                          
  pv_daily:                                                                                                                                             
    source: sensor.energy_pv                                                                                                                            
    cycle: daily                                                                                                                                        
  pv_monthly:                                                                                                                                           
    source: sensor.energy_pv                                                                                                                            
    cycle: monthly                                                                                                                                      
  house_daily:                                                                                                                                          
    source: sensor.energy_house                                                                                                                         
    cycle: daily                                                                                                                                        
  house_monthly:                                                                                                                                        
    source: sensor.energy_house                                                                                                                         
    cycle: monthly                                                                                                                                      
  grid_daily:                                                                                                                                           
    source: sensor.energy_grid                                                                                                                          
    cycle: daily                                                                                                                                        
    net_consumption: true                                                                                                                               
  grid_monthly:                                                                                                                                         
    source: sensor.energy_grid                                                                                                                          
    cycle: monthly                                                                                                                                      
    net_consumption: true 
```

wishie, 2020-12-09:
Note: your 'source' will likely be named different to mine. (eg goodwe_solar_total will likely be goodie_ppv for most people)

schamock, 2020-12-09:
> Well, such data are not directly available.
> The inverter only provides "immediate" or snapshot values.
> The only 2 exceptions are the daily and total PV production, for those the inverter maintains [...]

Oh wow, I wasn't aware of this. I thought that the inverter also sums up the total buy and/or load consumption. That really explains why the numbers in the SEMS App don't really match to the values on the smart meter...
Thanks anyway. Your scripts are great for getting the available datapoints from the inverter!

closed 2020-12-09 by schamock

mletenay, 2020-12-13:
I have added example of the buy/sell values to the readme file: https://github.com/mletenay/home-assistant-goodwe-inverter#cumulative-energy-values