mletenay/home-assistant-goodwe-inverter #18: GW10K-ET

perg67, 2020-12-25:
What a fantastic piece of code you written. I have an ET inverter GW10K-ET and managed to communicate and read out data with your code. However, GW10K-ET seems to reply on this message from the PV Master APP: f7038ca00011bbe2. Inverter replies with:  aa55f7032200010048000a01110001f651fa35fa82eb060112fc1cfc26fc44fc2a138b4997d3a37d4c (41 bytes). Do you have any idea what this  data is used for? (I tried to decode it as ES __settings is decoded but the message is shorter and mapping isn't the same).
Any idea why this message is requested by the APP?

mletenay, 2021-01-11:
ET seems to provide runtime data in several API calls - so-called runtimeData1 [..0389..] , runtimeData2 [..038C] and batteryInfo [..0390..].
The data1 and battery are invoked are read in read_runtime_data() method.
The command you observed is the data2, which is not call there (yet?) since it does not seem to provide any useful values.
AFAIK, the payload offset 8 (2bytes) is "meter_status", offset 16 (2bytes) is "total_active_power" and 18 (2bytes) is "power_meter".
When/if we decode any useful data from it, I can add it to the read_runtime_data() invocation.

mletenay, 2021-09-02:
Closing the issue, battery and meter sensors are properly decoded now.