mletenay/home-assistant-goodwe-inverter #19: NaN values

rubenltp, 2021-01-14:
Hi,

First at all, great job! it's amazing!
I've used this integration for months, but now, with the cold weather, I think the wifi doesn't work very well, because I've had reads from the Inverter with NaN values when the response doesn't return at time. The result is that I had a lot of "blank" spaces on te datas.
Before I was using the cloud integration, but I prefer to read it  on my own net, it's faster, it's more reliable; but with this "blanks" datas is harder to know at any time the correct values.
Maybe, if the value only would update when had a valid response could be better, at least for me, I'll prefer to know the last valid value better than None (NaN).
What do you think about it?

Greetings!

mletenay, 2021-01-14:
Well, the NaN values are reported intentionally, it's a change implemented just recently (Dec 8, 2020).
Prior to the fix it was impossible to detect problems with connection to inverter since, the stale values were reported indefinitely.
Network dropouts were also affecting the cumulative counters.

I personally never observed such network dropouts, but at least one user observed network issues (actually packet fragmentation - incomplete udp packets).

What I can do is to try to expose the protocol timeout and re-try counts which are actually hardcoded (2s/3x).
The defaults are probably fine for most users, but if someone like you have unreliable network, it would be possible for him to increase those values in home assistant config to avoid the NaN reads.

rubenltp, 2021-01-14:
Well, maybe it could be a fix. The change of the re-try value to set the
better value for my "special" network, hahaha.
Thanks for all!

mletenay, 2021-01-14:
I've just made that change.
When you'll update the HA plugin, you should be able to set 2 new config params - _network_timeout_ and _network_retries_. See the readme.
I guess it's more probably some packet delivery issue, so I would start increasing the retries first.

rubenltp, 2021-01-15:
Thanks, I'm trying it now.
I'll tell you something about the results.

closed 2021-01-23 by mletenay