mletenay/home-assistant-goodwe-inverter #22: Work Mode set but not applied - ES 4.6kw

GIJOE-SA, 2021-02-06:
Hi, although the HA goodwe.set_work_mode service sets the work mode and you can actually see it as set in PV Master it does not take effect. When you reapply in PV Master it does take effect. Goo

tinuva, 2021-04-30:
I also have this issue.

I can see the code is [here](https://github.com/mletenay/home-assistant-goodwe-inverter/blob/master/custom_components/goodwe/goodwe_inverter.py#L1241) to do it.

What I notice in the last part, the hex code is capitals where the set_ongrid_battery_dod is small letters in the hex and set_ongrid_battery_dod works.

But GIJOE mentions he can see it is set in pv master but not applied...I wonder.

mletenay, 2022-01-09:
I believe the work mode switching on ES/EM inverters was finally addressed, please verify.

closed 2022-01-09 by mletenay

tinuva, 2022-01-10:
Can confirm this now works for me in 0.9.7

After setting the mode I can see the inverter reboot, including the click sounds it make and then after the new mode as was set, is in effect.

GIJOE-SA, 2022-01-10:
Thank you gents, will only be upgrading by the end of next week when I am back but i am sure if it is working for David it will work on my side.

GIJOE-SA, 2022-01-10:
Hi, I decided to to upgrade today remotely and luckily everything worked fine. I can confirm that the mode setting is working thank you very much.  

Unfortunately the new way of doing this without the services broke some of my automations. This were particularly helpful when Eskom load shedding picked that there were will be power outages and I used the service to set the discharge depth to 0 (Without restating the Inverter). This forced it basically to use Eskom power and any access solar power to charge batteries through automations with no human intervention. 

The new way uses number.depth_of_discharge_on_grid value which I can set but it does not take effect, it even shows on the slide bar that it changes but has no effect until you use the slide bar itself. 

Is there another way I can use the new method to achieve this? If not, is it possible to bring the services back?

Thanks for the help

tinuva, 2022-01-11:
@GIJOE-SA 

You can still use it in automations. I manage my DoD multiple times a day and it still works as previously.

Instead of a service, just update the number.

This works for me:
```
      - alias: "Set DoD on inverter"
        service: number.set_value
        data:
          value: >
            {{ dod }}
        target:
          entity_id: number.depth_of_discharge_on_grid
```

My complete script looks (will work for automations too) look like this:
```
script:
  ######################################################################
  # Re-usable script to set the DoD via input
  ######################################################################
  # Example:
  ######################################################################
  # - service: script.goodwe_set_dod
  #   data:
  #     forecast: "50"
  ######################################################################
  goodwe_set_dod:
    sequence:
      - alias: "Set variables"
        variables:
          dod: >
            {% set forecast_int = forecast|int %}
            {% if forecast_int >= 28 %}70
            {% elif forecast_int >= 23 and forecast_int < 28 %}65
            {% elif forecast_int >= 20 and forecast_int < 23 %}60
            {% elif forecast_int >= 18 and forecast_int < 20 %}55
            {% elif forecast_int >= 16 and forecast_int < 18 %}50
            {% elif forecast_int >= 12 and forecast_int < 16 %}40
            {% elif forecast_int >= 10 and forecast_int < 12 %}30
            {% else %}16{% endif -%}
      - alias: "Set DoD on inverter"
        service: number.set_value
        data:
          value: >
            {{ dod }}
        target:
          entity_id: number.depth_of_discharge_on_grid
      - alias: "Log action taken"
        service: script.log
        data:
          icon: ':zap:'
          object: 'GoodWe'
          message: >
            Setting DoD to {{ dod }} based on {{ forecast }}kWh forecast
```

tinuva, 2022-01-11:
I know it works, because it takes affect as long as you are back in general mode.

Jayksopp, 2022-01-11:
> @GIJOE-SA
>
> You can still use it in automations. I manage my DoD multiple times a day and it still works as previously.
>
> Instead of a service, just update the number.
>
> This works for me:
>
> ```
>       - [...]

From this do I understand correctly if you set the DoD above the current DoD the invertor will start charging the battery until it reaches that DoD? 
What do you use to determine the forecast value? Is it based on sol cast? Or do you have a way to include weather forecast etc as well?

tinuva, 2022-01-11:
@Jayksopp 

> From this do I understand correctly if you set the DoD above the current DoD the invertor will start charging the battery until it reaches that DoD?
No. If you set the DoD to lets say 60%, it means, when there is no sunlight and you are in general mode, that the inverter will use 60% of the battery, eg. it will go down to 40% charge left, before it switches the load over to grid.

> What do you use to determine the forecast value? Is it based on sol cast? Or do you have a way to include weather forecast etc as well?
I do use solcast. You can see my solcast config here --> https://github.com/tinuva/home-assistant-config/blob/master/packages/energy/solcast_rest.yaml
My DoD automation are here --> https://github.com/tinuva/home-assistant-config/blob/master/packages/energy/goodwe_dod.yaml

I see I still need to remove the old input I used.

GIJOE-SA, 2022-01-11:
@tinuva 
> I know it works, because it takes affect as long as you are back in general mode.

Thank you very much David. The service: number.set_value is indeed working. I have tried yesterday setting the value using developers tools without using the number.set_value which obviously did not work. 
Very nice script you wrote there.

tinuva, 2022-01-11:
@Jayksopp 

> From this do I understand correctly if you set the DoD above the current DoD the invertor will start charging the battery until it reaches that DoD?

I think I understood the question wrong first time. If you set DoD above the current DoD, inverter will not charge it, but will only use the battery if it goes even higher than the last DoD set which is now higher.

mif11k, 2022-05-02:
I struggled a week with the problem related to setting the DoD value, almost going crazy. After finding this thread, I checked again by setting the value of number.depth_of_discharge_on_grid entity. I found out, that decreasing the value works always, but you can increase the value only up to 89. If you set any higher value (90 -99) it is ignored by the inverter, however it looks just fine in the HA.