mletenay/home-assistant-goodwe-inverter #23: Support for MS Models

marcoevcom, 2021-02-09:
Hi, I have installed a MS Series model, unfortunatly it doesn't work with your script.

Is there a way to find this commands, and help add this model ?

2021-02-09 11:04:26,767 __init__(59) - DEBUG: Using selector: EpollSelector
2021-02-09 11:04:26,769 discover(636) - DEBUG: Probing inverter at 192.168.0.168                                                                   :8899
2021-02-09 11:04:26,770 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-02-09 11:04:28,774 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-02-09 11:04:28,775 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-02-09 11:04:30,778 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-02-09 11:04:30,779 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-02-09 11:04:32,783 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-02-09 11:04:32,784 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-02-09 11:04:34,787 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
2021-02-09 11:04:34,789 discover(656) - DEBUG: Probing ES inverter at 192.168.0.168:8899
2021-02-09 11:04:34,791 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-02-09 11:04:36,795 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-02-09 11:04:36,795 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-02-09 11:04:38,798 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-02-09 11:04:38,799 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-02-09 11:04:40,803 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-02-09 11:04:40,804 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-02-09 11:04:42,807 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
2021-02-09 11:04:42,808 discover(656) - DEBUG: Probing ET inverter at 192.168.0.168:8899
2021-02-09 11:04:42,810 connection_made(316) - DEBUG: Send: 'f70388b800213ac1'
2021-02-09 11:04:44,814 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-02-09 11:04:44,815 connection_made(316) - DEBUG: Send: 'f70388b800213ac1'
2021-02-09 11:04:46,818 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-02-09 11:04:46,819 connection_made(316) - DEBUG: Send: 'f70388b800213ac1'
2021-02-09 11:04:48,823 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-02-09 11:04:48,824 connection_made(316) - DEBUG: Send: 'f70388b800213ac1'
2021-02-09 11:04:50,828 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
problem in goodwe-test - InverterError: Unable to connect to the inverter at host=192.168.0.168 port=8899, or your inverter is not supported yet.
Failures=[InverterError("No valid response received to 'aa55c07f0102000241' request"), InverterError("No valid response received to 'aa55c07f0102000241' request"), InverterError("No valid response received to 'f70388b800213ac1' request")]

mletenay, 2021-02-09:
From the log it seems your model does not talk any of 2 known protocols.
You can try to execute the _inverter_scan.py_ script (outside HA, plain python).
It tries to establish initial communication with few other inverter protocols out there.
If it will get any kind of response we may investigate it further.

boc-the-git, 2021-04-17:
@marcoevcom have you looked into this any further?

marcoevcom, 2021-04-27:
Hi, sorry about taking this long to answer.

➜  /config python3 inverter_scan.py
2021-04-27 15:26:59,726 __init__(59) - DEBUG: Using selector: EpollSelector
2021-04-27 15:26:59,729 search_inverters(603) - DEBUG: Searching inverters by broadcast to port 48899
2021-04-27 15:26:59,731 connection_made(316) - DEBUG: Send: '574946494b49542d3231343032382d52454144'
2021-04-27 15:26:59,901 datagram_received(327) - DEBUG: Received: '3139322e3136382e302e3136382c3334454145373232444432302c536f6c61722d576946693230415730333238'
Located inverter at IP: 192.168.0.168, mac: 34EAE722DD20, name: Solar-WiFi20AW0328
Trying command: AA55C07F0102000241
2021-04-27 15:26:59,908 __init__(59) - DEBUG: Using selector: EpollSelector
2021-04-27 15:26:59,913 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-04-27 15:27:01,918 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-04-27 15:27:01,920 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-04-27 15:27:03,924 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-04-27 15:27:03,925 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-04-27 15:27:05,930 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-04-27 15:27:05,931 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-04-27 15:27:07,934 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to AA55C07F0102000241 command
Trying command: 7F03753100280409
2021-04-27 15:27:07,939 __init__(59) - DEBUG: Using selector: EpollSelector
2021-04-27 15:27:07,943 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-04-27 15:27:09,947 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-04-27 15:27:09,948 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-04-27 15:27:11,952 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-04-27 15:27:11,953 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-04-27 15:27:13,957 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-04-27 15:27:13,958 connection_made(316) - DEBUG: Send: '7f03753100280409'
2021-04-27 15:27:15,960 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to 7F03753100280409 command
Trying command: 197d0001000dff045e50303036564657f6e60d
2021-04-27 15:27:15,963 __init__(59) - DEBUG: Using selector: EpollSelector
2021-04-27 15:27:15,967 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-04-27 15:27:17,970 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-04-27 15:27:17,971 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-04-27 15:27:19,974 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-04-27 15:27:19,975 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-04-27 15:27:21,979 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-04-27 15:27:21,979 connection_made(316) - DEBUG: Send: '197d0001000dff045e50303036564657f6e60d'
2021-04-27 15:27:23,982 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to 197d0001000dff045e50303036564657f6e60d command
Trying command: 680241b132304157303332380100bc16
2021-04-27 15:27:23,988 __init__(59) - DEBUG: Using selector: EpollSelector
2021-04-27 15:27:23,992 connection_made(316) - DEBUG: Send: '680241b132304157303332380100bc16'
2021-04-27 15:27:26,000 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-04-27 15:27:26,001 connection_made(316) - DEBUG: Send: '680241b132304157303332380100bc16'
2021-04-27 15:27:28,005 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-04-27 15:27:28,007 connection_made(316) - DEBUG: Send: '680241b132304157303332380100bc16'
2021-04-27 15:27:30,009 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-04-27 15:27:30,011 connection_made(316) - DEBUG: Send: '680241b132304157303332380100bc16'
2021-04-27 15:27:32,014 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to 680241b132304157303332380100bc16 command
Trying command: 680241b138323330574130320100bc16
2021-04-27 15:27:32,018 __init__(59) - DEBUG: Using selector: EpollSelector
2021-04-27 15:27:32,021 connection_made(316) - DEBUG: Send: '680241b138323330574130320100bc16'
2021-04-27 15:27:34,023 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-04-27 15:27:34,024 connection_made(316) - DEBUG: Send: '680241b138323330574130320100bc16'
2021-04-27 15:27:36,027 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-04-27 15:27:36,028 connection_made(316) - DEBUG: Send: '680241b138323330574130320100bc16'
2021-04-27 15:27:38,031 _timeout_heartbeat(346) - DEBUG: Re-try #3
2021-04-27 15:27:38,033 connection_made(316) - DEBUG: Send: '680241b138323330574130320100bc16'
2021-04-27 15:27:40,036 _timeout_heartbeat(350) - DEBUG: Re-try #4, closing socket
No response to 680241b138323330574130320100bc16 command
Identifying inverter at IP: 192.168.0.168
2021-04-27 15:27:40,040 __init__(59) - DEBUG: Using selector: EpollSelector
2021-04-27 15:27:40,047 discover(636) - DEBUG: Probing inverter at 192.168.0.168:8899
2021-04-27 15:27:40,052 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-04-27 15:27:42,055 _timeout_heartbeat(346) - DEBUG: Re-try #1
2021-04-27 15:27:42,057 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-04-27 15:27:44,060 _timeout_heartbeat(346) - DEBUG: Re-try #2
[... 8 lines cut ...]
2021-04-27 15:27:52,085 _timeout_heartbeat(346) - DEBUG: Re-try #2
2021-04-27 15:27:52,087 connection_made(316) - DEBUG: Send: 'aa55c07f0102000241'
2021-04-27 15:27:54,090 _timeout_heartbeat(346) - DEBUG: Re-try #3
[... cut ...]
  File "/config/custom_components/goodwe/goodwe_inverter.py", line 667, in discover
    raise InverterError(
custom_components.goodwe.goodwe_inverter.InverterError: Unable to connect to the inverter at host=192.168.0.168 port=8899, or your inverter is not supported yet.
Failures=[InverterError("No valid response received to 'aa55c07f0102000241' request"), InverterError("No valid response received to 'aa55c07f0102000241' request"), InverterError("No valid response received to 'f70388b800213ac1' request")]

mletenay, 2021-08-29:
There was some progress adding some new inverter types recently, try the latest code and e.g. try to explicitly set the inverter_type to DT. That type seem to work or D-NS and XS inverters, so MS might work as well ...

darek-margas, 2021-09-06:
> There was some progress adding some new inverter types recently, try the latest code and e.g. try to explicitly set the inverter_type to DT. That type seem to work or D-NS and XS inverters, so MS might work as well ...

I just tried - no luck.

mletenay, 2021-09-06:
Are you able to connect to the inverter via the mobile app ? (SolarGo or PvMaster)

darek-margas, 2021-09-06:
> Are you able to connect to the inverter via the mobile app ? (SolarGo or PvMaster)

I think PvMaster works.

mletenay, 2021-09-06:
Could you update to latest version 0.8.9, remove all optional configuration parameters from the yaml config file (leave there only IP address) and set/adjust the HA logger this way ?:
```
logger:
  default: warning
  logs:
    custom_components.goodwe: debug
```
And then post relevant log entries here.

darek-margas, 2021-09-06:
I'll do once it wakes up tomorrow.

darek-margas, 2021-09-06:
OK, 192.168.16.130 is homekit, 192.168.16.131 is inverter:
2021-09-07 08:23:07 DEBUG (MainThread) [custom_components.goodwe.goodwe] Probing inverter at 192.168.16.131:8899
2021-09-07 08:23:07 DEBUG (MainThread) [custom_components.goodwe.goodwe] Probing inverter at 192.168.16.130:8899
2021-09-07 08:23:09 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.16.131', 8899)
2021-09-07 08:23:09 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.16.130', 8899)
2021-09-07 08:23:11 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-07 08:23:11 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.16.131', 8899)
2021-09-07 08:23:11 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-07 08:23:11 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.16.130', 8899)
2021-09-07 08:23:13 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-07 08:23:13 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.16.131', 8899)
2021-09-07 08:23:13 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-07 08:23:13 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.16.130', 8899)
2021-09-07 08:23:15 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-07 08:23:15 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.16.131', 8899)
2021-09-07 08:23:15 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-07 08:23:15 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.16.130', 8899)
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe] Probing ET inverter at 192.168.16.131:8899
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe] Probing ET inverter at 192.168.16.130:8899
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.16.131', 8899)
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.16.130', 8899)
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Received: aa55f7830220c3
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.modbus] Response returned command failure: 131, expected 3.
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Invalid response: aa55f7830220c3
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.16.130', 8899)
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Received: aa55f7830220c3
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.modbus] Response returned command failure: 131, expected 3.
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Invalid response: aa55f7830220c3
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.16.130', 8899)
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Received: aa55f7830220c3
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.modbus] Response returned command failure: 131, expected 3.
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Invalid response: aa55f7830220c3
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.16.130', 8899)
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Received: aa55f7830220c3
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.modbus] Response returned command failure: 131, expected 3.
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Invalid response: aa55f7830220c3
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.16.130', 8899)
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Received: aa55f7830220c3
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.modbus] Response returned command failure: 131, expected 3.
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Invalid response: aa55f7830220c3
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.16.130', 8899)
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Received: aa55f7830220c3
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.modbus] Response returned command failure: 131, expected 3.
2021-09-07 08:23:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Invalid response: aa55f7830220c3
[... cut ...]
2021-09-07 08:23:37 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-07 08:23:37 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.16.131', 8899)
2021-09-07 08:23:39 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-07 08:23:39 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.16.131', 8899)
2021-09-07 08:23:41 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket

darek-margas, 2021-09-07:
I've checked apps and:
Solargo says both homekit and inverter are incompatible
Pvmaster opens both but shows all parameters with zeros, N/As, etc and diagram with all zeros too.

mletenay, 2021-09-07:
I'm afraid you're out of luck and it seems your MS cannot be monitored via local network :-(

closed 2021-09-07 by mletenay

darek-margas, 2021-09-07:
I've asked my installer who promised to chase Goodwe support to find out what actually works for MS. Previously their support responded when chased that way so I may come with further clue.