mletenay/home-assistant-goodwe-inverter #25: GW5K-ET

TPochobradsky, 2021-03-08:
Hi,

first of all I would like to thank you very much for the great work. I have an GW5K-ET inverter and trying to use that component but I get an error.
What is bad?

Thank for your help.

Log Core:
2021-03-08 23:19:01 WARNING (MainThread) [homeassistant.loader] No 'version' key in the manifest file for custom integration 'goodwe'. This will not be allowed in a future version of Home Assistant. Please report this to the maintainer of 'goodwe'
2021-03-08 23:20:43 WARNING (MainThread) [custom_components.goodwe.sensor] Could not retrieve data from inverter: No valid response received to 'f703891c007d7ae7' request
2021-03-08 23:21:13 WARNING (MainThread) [custom_components.goodwe.sensor] Could not retrieve data from inverter: No valid response received to 'f703891c007d7ae7' request

[image]
[image]

mletenay, 2021-03-20:
Sorry, for late reply, was kinda bussy...
Do you still have problems ?
I have the the 10K-ET, exactly the same firmware, so I suspect some network layer issues.
Anyway, can you try to execute the inverter_scan.py ?

TPochobradsky, 2021-03-20:
Thank you for answer. 
Yes, I Have still the problem. Can it be because I don't have batteries connected yet?
I have tried to execute the inverter_scan.py but there are errors. I have never use scripts in hassio.
Thank you for advice

inverter_scan.py log:
_2021-03-20 22:17:40 WARNING (SyncWorker_2) [homeassistant.components.python_script] Warning loading script inverter_scan.py: Line 15: Prints, but never reads 'printed' variable., Line None: Prints, but never reads 'printed' variable.
2021-03-20 22:17:40 INFO (SyncWorker_2) [homeassistant.components.python_script] Executing inverter_scan.py: {}
2021-03-20 22:17:40 ERROR (SyncWorker_2) [homeassistant.components.python_script.inverter_scan.py] Error executing script: __import__ not found
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/python_script/__init__.py", line 218, in execute
    exec(compiled.code, restricted_globals)
  File "inverter_scan.py", line 2, in <module>
ImportError: __import__ not found_

TPochobradsky, 2021-07-20:
Hi,
after connecting the battery the problem still persists. Could you check debug report to see if you find a issue?
Thank you for your help
Tomas
[Debug.txt](https://github.com/mletenay/home-assistant-goodwe-inverter/files/6849678/Debug.txt)

voodoopt, 2021-08-02:
Hi same problem with 
Device: Goodwe 3648d-ES
Current version:  V1.0.3.8
`2021-08-02 12:38:24 WARNING (MainThread) [custom_components.goodwe.sensor] Could not retrieve data from inverter: No valid response received to 'xxxxxxxxxxxxxxxxxx' request
2021-08-02 12:39:06 WARNING (MainThread) [custom_components.goodwe.sensor] Could not retrieve data from inverter: No valid response received to 'xxxxxxxxxxxxxxxxxx' request`

mletenay, 2021-08-29:
@TPochobradsky Looking at the debug log it seems like problem on network layer.
You get an awfull lot of truncated UDP packets.
There's not much I can do on the code level, you somehow have to resolve your network issue, maybe improve wifi reception resp. prevent udp packets truncation whatever is causing that ...
On the software side, you can increase the retry_timeout parameter in (a bit desperate) attempt to get some valid data more frequently.

TPochobradsky, 2021-08-30:
I use wired connection of GoodWe and network seems to be stable. I have read about problem between GoodWe and Router MikroTik hAP ac2, which I have. Do you have sam indication about it?

mletenay, 2021-08-30:
Well, I don't have any specifics, there's not enough knowledge about these problems, but there's a guy in forum which started with lan adapter and faced same packet truncation problems. And they disappeared after he switched to wifi.

TPochobradsky, 2021-08-31:
Thank you for interest. I forced the supplier to replace the wifi module with a LAN and now is no way back. :-)
I will be happy if someone manages to solve the problem in the future.

mletenay, 2021-08-31:
I don't how to help you at the moment.
From the point of view of the software, the UDP packets received by the python code are most of the time incomplete, the packet payload is missing a lots of bytes.
You need to figure out which layer is responsible for that problem.
Is it TCP/IP stack implementation ? Is it physical layer and some MTU issue ?
What is your HA environment ? RPi ? Maybe try another machine on LAN segment and plain python ?

TPochobradsky, 2021-08-31:
I'm afraid that is to complicated for me.  :-(
My HA is installed on VMM in Synology NAS. Version core-2021.8.8. Inverter communicate with SEMS portal without problems. LAN settings are beyond my capabilities.