mletenay/home-assistant-goodwe-inverter #28: nothing showing in Home Assistant

matda59, 2021-05-09:
perhaps this isn't the best place to put this. i have a GW5000L-DT goodwe inverter:
i'm running home assistant OS, i've installed HACS and GoodWe Inverter Solar Sensor (UDP - no cloud), added the lines to the configuration.yaml, restarting - but i cannot seem to add my goodwe inverter anywhere. how is this done? also how can i run the inverter_test.py.. i cannot find anywhere how to run that and get an output, i've tried SSHing in and running, it adding python into the configuration.yaml but it seems really difficult to run a python script.

matda59, 2021-05-25:
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:250
Integration: Sensor (documentation, issues)
First occurred: 4:49:28 PM (1 occurrences)
Last logged: 4:49:28 PM
Platform goodwe not ready yet: Unable to connect to the inverter at host=192.168.0.162 port=8899, or your inverter is not supported yet. Failures=[InverterError("No valid response received to 'aa55c07f0102000241' request"), InverterError("No valid response received to 'aa55c07f0102000241' request"), InverterError("No valid response received to 'f70388b800213ac1' request"), InverterError("No valid response received to '7f03753100280409' request")]; Retrying in background in 30 seconds

jrial, 2021-08-11:
You should see some entities pop up if you go to Configuration => Entities and search for "goodwe":

[image]

mletenay, 2021-08-22:
Closing this issue, as DT inverters are properly supported now and I assume it should work now for you.

closed 2021-08-22 by mletenay

matda59, 2021-08-23:
Still not working for me.

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:268
Integration: Sensor (documentation, issues)
First occurred: 2:34:34 PM (1 occurrences)
Last logged: 2:34:34 PM
Platform goodwe not ready yet: Unable to connect to the inverter at host=192.168.0.165 port=8899, or your inverter is not supported yet. Failures=[InverterError("No valid response received to 'aa55c07f0102000241' request"), InverterError("No valid response received to 'f70388b800213ac10000' request"), InverterError("No valid response received to 'f70388b800213ac1' request"), InverterError("No valid response received to '7f03753100280409' request"), InverterError("No valid response received to 'aa55c07f0102000241' request")]; Retrying in background in 30 seconds