mletenay/home-assistant-goodwe-inverter #29: GW6KT and SEMS app

piluxh, 2021-05-09:
I have a GW6K-DT and since the update earlier this week it now collects data properly in HA. However as soon as it started collecting data, the SEMS app indicated my device had gone offline and it no longer worked. When I turned off the integration, the SEMS app started working again.

mareklab, 2021-06-03:
I have the same problem, with GW8TK. 

I think this is due to a very slow WiFi-Box from GoodWe that is unable to handle both data streams at the same time. It would be great if this integration could be available under all other HA integrations, so we could temporarily suspend pulling data via automation, just to give a chance to upload data to the regular SEMS portal.

mletenay, 2021-08-29:
@piluxh Does this problem still occurs in the latest code ?
(Since I never observer anything similar on my GW10K-ET)
@mareklab No, I'm intensively working on some code adjustments so in some near future I'll be able to create a push request to main HA source. I guess a few weeks ...

andrzej-dunowski, 2021-10-01:
I installed the component last week.  I have a GW4K-DT inverter.  The next day I saw that there was no data in SEMS.  I turned off the component and some data appeared.  Maybe the component downloads data too often, a setting that sets it to fetch data from time to time would be handy.

fizcris, 2021-10-01:
Hi,

If of any help:
For the EH/ET series if you set up a really high refresh rate (0.5-1 Hz) the PVMaster app stops working but SEMSportal never seemed to be affected that way.

piluxh, 2021-10-19:
@mletenay I think this is still happening. I switched away from using this integration due to this issue and went to a different integration which gets the data directly from the SEMS portal. At the moment I cannot reach the SEMS portal so I switched back. The yellow LED on my inverter is flashing 4 times indicating that it has lost connection to the internet (which I guess means that it cannot reach the portal as my internet is fine). I've lowered the scanning frequency now to see if that will help.

dd871, 2021-11-02:
Adding to my earlier comment, it seems values in the configuration file are being ignored if the GUI integration is used. I have now also set the scan interval in the GUI to 180 seconds and with this value both (integration and SEMS) looks to be working fine. Smaller values, e.g. 120s, are strangely still impacting SEMS, but an update every 3 minutes is good enough for me.

mletenay, 2021-11-11:
Since it works with 180s and there's actually nothing we can do on the integration part, I'm closing the ticket.

closed 2021-11-11 by mletenay