mletenay/home-assistant-goodwe-inverter #37: House_comsumption less than backup

scrhall, 2021-07-27:
Hi I have GW5048D-ES
The consumption of the house at night with low load, appears less than the consumption of the backup, should not be greater or equal? House _consumption includes backup doesn't it? 

[image]

mletenay, 2021-08-29:
At the moment the house_consumption is calculared as ppv1 + ppv2 + pbattery1 - pgrid.
Assuming the pgrid is the power at meter (CT clamp).
If the pgrid is actually power on inverter output, than it may not represent correct calculation and probably also pback_up should substracted.
I have no access to ES inverter so can't verify it, can you somehow validate this theory ?
Or is there reported sensor represented the power at meter ?

scrhall, 2021-09-03:
I can confirm that house consumption not take account all the W consumed. If I turn on a fan connected to backup, for example, I only see the W reflected in the backup sensor and not in the house consumption sensor.

scrhall, 2021-09-03:
[image]
This is the data, if I turn off all breakers of the non-critical part of my installation
[image]
If I turn on for example a ceiling fan (in the backup output), only the backup sensor goes up
[image]
I add another capture of the data, none of them match

scrhall, 2021-09-03:
[image]
Two additional screenshots, the first, is with the normal morning load. You can also see the info showing pvmaster
[image]
This last screenshot is completely disconnected from both the backup and the ac in

Where does that consumption of 80w come from?, is it the consumption of the investor himself?  Even so, neither in pvmaster nor in ha are the data coherent

ziporah, 2021-09-03:
self consumption without backup should be <13W for your inverter without backup, they don't specify the impact of enabling backup mode

mletenay, 2021-09-12:
@scrhall I've just fixed the house_consumption (will get into v0.9.10), the backup load was indeed missing there.
However I'm now wondering about the total_power and plant_power sensors. The former comes directly from inverter, the latter is calculated as pload + pback_up.
I expect they should report roughly the same numbers.
Can you observe it in real life if it is true ?

scrhall, 2021-09-12:
[image]
[image]

scrhall, 2021-09-12:
[image]
[image]

CEF-PHORAM, 2021-09-12:
The ES-5048 self-consumption with the output of the BackUp loads active is between **60 to 75** total watts. I have measured it with external instruments.

mletenay, 2021-09-12:
House consumption fixed in v0.9.0

closed 2021-09-12 by mletenay

scrhall, 2021-09-12:
The solution is not correct, it is not simply adding backup, the backup is already included in a certain way. Now the consumption appears as if adding twice the backup. I had to install the previous version again

mletenay, 2021-09-13:
I'm not reopening this ticket, since the discussion moved to new one #66.
But I need help with defining proper formula.

jebeke65, 2022-05-12:
The GW5000S-BP house consumption is much too high, hence not correct. When I check the value during the night, I’m getting around 1500W which is certainly not what is consumed in the house