mletenay/home-assistant-goodwe-inverter #41: Stops working after update

kolossboss, 2021-08-18:
Got this message in the log. „Inventer is not supported“. But runs for month.

`Logger: homeassistant.components.sensor
Source: custom_components/goodwe/goodwe_inverter.py:665
Integration: Sensor (documentation, issues) 
First occurred: 20:54:16 (1 occurrences) 
Last logged: 20:54:16

Error while setting up goodwe platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/goodwe/sensor.py", line 78, in async_setup_platform
    inverter = await discover(config[CONF_IP_ADDRESS], config[CONF_PORT], config[CONF_NETWORK_TIMEOUT], config[CONF_NETWORK_RETRIES])
  File "/config/custom_components/goodwe/goodwe_inverter.py", line 665, in discover
    raise NameError("No compatible inverter found")
NameError: No compatible inverter found
`

maurizioandreotti60, 2021-08-19:
I have the same issue after upgrade.
2021-08-19 09:13:21 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up goodwe platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/goodwe/sensor.py", line 78, in async_setup_platform
    inverter = await discover(config[CONF_IP_ADDRESS], config[CONF_PORT], config[CONF_NETWORK_TIMEOUT], config[CONF_NETWORK_RETRIES])
  File "/config/custom_components/goodwe/goodwe_inverter.py", line 665, in discover
    raise NameError("No compatible inverter found")
NameError: No compatible inverter found

rjltp, 2021-08-19:
Hi, 
The same issue here:
2021-08-19 09:49:20 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up goodwe platform for sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/goodwe/sensor.py", line 78, in async_setup_platform
inverter = await discover(config[CONF_IP_ADDRESS], config[CONF_PORT], config[CONF_NETWORK_TIMEOUT], config[CONF_NETWORK_RETRIES])
File "/config/custom_components/goodwe/goodwe_inverter.py", line 665, in discover
raise NameError("No compatible inverter found")
NameError: No compatible inverter found

mletenay, 2021-08-19:
We added EH support, something must went wrong ;-(
What inverter type doe you have ?

kolossboss, 2021-08-19:
> We added EH support, something must went wrong ;-(
> What inverter type doe you have ?

I have: GW3648-EM

mletenay, 2021-08-19:
Ok, I found potential problem there.
Can you update and verify ?

rjltp, 2021-08-19:
> Ok, I found potential problem there.
> Can you update and verify ?

Hi, I have GW5048-EM. I've updated and all looks good for me now. Thanks

kolossboss, 2021-08-19:
> Ok, I found potential problem there.
> Can you update and verify ?

Yes works again. Thanks a lot!

mletenay, 2021-08-19:
Sorry for the troubles, I missed it during review.

closed 2021-08-19 by mletenay