mletenay/home-assistant-goodwe-inverter #42: Might be an idea to copy the Inverter Test Script down with the Integration ?

craigcurtin-dev, 2021-08-20:
Many people may have an issue with getting the test script into a HassOs setup - might be an idea to include that in the repository ?

Craig

mletenay, 2021-08-22:
Not that bad idea.
At the moment we are actually extracting all the goodwe communication code to standalone library (https://github.com/marcelblijleven/goodwe). Once we finish with that, we may slightly redesign the HomeAssistant related code and include some test scripts there ...

mletenay, 2021-09-06:
We started work on moving this plugin to become native part of HA and the communication code and auto-discovery was improved in recent days, so this task is getting obsolete/irrelevant, so I'm closing the ticket.

closed 2021-09-06 by mletenay