mletenay/home-assistant-goodwe-inverter #44: GoodWe SBP 5000 support

craigcurtin-dev, 2021-08-22:
A little more progress here

PV Master app when it starts sends out a UDP broadast on 48899 for these inverters.

The inverters return their IP Address, their Mac Address and (presumably) a model number - in this case HF_E20

If i then run your test scripts with the port set to 8899 and the appropriate MAC address there is some more progress

python3 inverter_test.py
2021-08-23 03:46:44,429 __init__(59) - DEBUG: Using selector: EpollSelector
2021-08-23 03:46:44,429 discover(647) - DEBUG: Probing inverter at 172.16.100.60:8899
2021-08-23 03:46:44,430 connection_made(327) - DEBUG: Send: 'aa55c07f0102000241'
2021-08-23 03:46:44,768 datagram_received(338) - DEBUG: Received: 'aa557fc001824d30383038434757'
2021-08-23 03:46:44,768 datagram_received(343) - DEBUG: Invalid response length: 14
2021-08-23 03:46:44,768 connection_made(327) - DEBUG: Send: 'aa55c07f0102000241'
2021-08-23 03:46:45,179 datagram_received(338) - DEBUG: Received: 'aa557fc001824d303830384347'
2021-08-23 03:46:45,180 datagram_received(343) - DEBUG: Invalid response length: 13
2021-08-23 03:46:45,180 connection_made(327) - DEBUG: Send: 'aa55c07f0102000241'
2021-08-23 03:46:45,665 datagram_received(338) - DEBUG: Received: 'aa557fc001824d3038303843475735303030532d42502331300000000000000000000000000039353030304250533230'
2021-08-23 03:46:45,665 datagram_received(343) - DEBUG: Invalid response length: 48
2021-08-23 03:46:45,665 connection_made(327) - DEBUG: Send: 'aa55c07f0102000241'
2021-08-23 03:46:46,079 datagram_received(338) - DEBUG: Received: 'aa557fc001824d3038303843475735303030532d42502331300000000000000000000000000039353030'
2021-08-23 03:46:46,080 datagram_received(343) - DEBUG: Invalid response length: 42
2021-08-23 03:46:46,080 connection_made(327) - DEBUG: Send: 'aa55c07f0102000241'
2021-08-23 03:46:46,293 datagram_received(338) - DEBUG: Received: 'aa557fc001824d3038303843475735303030532d42502331300000000000000000000000000039353030304250533230425730303137333630303431302d30343030312d3038473431302d30323033342d31320b1072'
2021-08-23 03:46:46,294 discover(671) - DEBUG: Probing EH inverter at 172.16.100.60:8899
2021-08-23 03:46:46,294 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-23 03:46:46,445 datagram_received(338) - DEBUG: Received: 'aa55f703420001135600000039003500300030003000420050'
2021-08-23 03:46:46,445 datagram_received(343) - DEBUG: Invalid response length: 25
2021-08-23 03:46:46,445 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-23 03:46:46,910 datagram_received(338) - DEBUG: Received: 'aa55f70342000113560000003900350030003000300042005000530047005700350030003000080008006500000000ffffffffffffffffffffffffffffff'
2021-08-23 03:46:46,910 datagram_received(343) - DEBUG: Invalid response length: 62
2021-08-23 03:46:46,910 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-23 03:46:47,365 datagram_received(338) - DEBUG: Received: 'aa55f7'
2021-08-23 03:46:47,365 datagram_received(343) - DEBUG: Invalid response length: 3
2021-08-23 03:46:47,365 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-23 03:46:47,770 datagram_received(338) - DEBUG: Received: 'aa55f70342000113560000003900350030003000300042005000530047005700350030003000080008006500000000ffffffffffffffffffffff'
2021-08-23 03:46:47,770 datagram_received(343) - DEBUG: Invalid response length: 58
2021-08-23 03:46:47,770 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-23 03:46:48,223 datagram_received(338) - DEBUG: Received: 'aa55f703'
2021-08-23 03:46:48,223 datagram_received(343) - DEBUG: Invalid response length: 4
2021-08-23 03:46:48,223 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-23 03:46:48,296 _timeout_heartbeat(361) - DEBUG: Re-try #6, closing socket
2021-08-23 03:46:48,297 discover(671) - DEBUG: Probing ET inverter at 172.16.100.60:8899
2021-08-23 03:46:48,297 connection_made(327) - DEBUG: Send: 'f70388b800213ac1'
2021-08-23 03:46:48,628 datagram_received(338) - DEBUG: Received: 'aa55f703420001135600000039003500300030003000420050005300470057003500'
2021-08-23 03:46:48,628 datagram_received(343) - DEBUG: Invalid response length: 34
2021-08-23 03:46:48,628 connection_made(327) - DEBUG: Send: 'f70388b800213ac1'
2021-08-23 03:46:48,754 datagram_received(338) - DEBUG: Received: 'aa55f70342000113560000003900350030003000300042005000530047005700350030003000080008006500000000ffffffffffffffffffffffffffffffffffffffffffffffff0a24'
Traceback (most recent call last):
  File "inverter_test.py", line 21, in <module>
    inverter = asyncio.run(inverter.discover(IP_ADDRESS, PORT, TIMEOUT, RETRIES))
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/pi/home-assistant-goodwe-inverter-master/custom_components/goodwe/goodwe_inverter.py", line 672, in discover
    await i.read_device_info()
  File "/home/pi/home-assistant-goodwe-inverter-master/custom_components/goodwe/goodwe_inverter.py", line 972, in read_device_info
    self.software_version = response[54:66].decode("ascii")
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)
root@ha-nodejs:/home/pi/home-assistant-goodwe-inverter-master#

I have some Wireshark packet captures also of the interaction between the Inverter and PV Master if that would help - and can do some more directed capturing tomorrow if required - let me know

Obviously we are receiving some responses from the inverters - noting that they are only AC coupled and do not have any Solar panels attached - in my case they have a DC Coupled LV (48) Lithium 13.3Kwh battery to each one.

How would you tackle decoding some of the strings that are being returned to look for patterns such as voltages etc ?

I can see when looking at the Wireshark captures that PV Master sends (i assume) a polling request 

aa55c07f0102000241 - From the PC - get response from Inverter - these are the most frequently sent - so i assume some form of screen update request that is context specific to the screen it is 

I then see aa55c07f0106000245 - sent on an occasional basis.

I am happy to help on decoding all of this to add to the project - i am a network rather than software engineer though - so need some direction in delving into the return streams.

Craig

craigcurtin-dev, 2021-08-22:
Here is the partial decode of the String response when sending the 241 request

ªUÀM0808CGW5000S-BP#1095000BPS20BW00173600410-04001-08G410-02034-12r

95000BPS20BW0017 - is the model and Serial number

mletenay, 2021-08-22:
Well, that response is most likely response to so called READ_DEVICE_INFO request and should contain some basic data about the inverter like type, serial number, firmware version etc ...
I assume it is in standard format, so the library should be able to decode it.

However, from the logs it seems you have some serious problems with network, resp. UDP reliability and/or packet truncation.
See the responses to repeated  'aa55c07f0102000241' requests 
'aa557fc001824d30383038434757'
'aa557fc001824d303830384347'
'aa557fc001824d3038303843475735303030532d42502331300000000000000000000000000039353030304250533230'
'aa557fc001824d3038303843475735303030532d42502331300000000000000000000000000039353030'
'aa557fc001824d3038303843475735303030532d42502331300000000000000000000000000039353030304250533230425730303137333630303431302d30343030312d3038473431302d30323033342d31320b1072'
5 attempts and actually all of them were truncated and never the whole payload was received.
(And the library refused to process such incomplete response).
I suggest to look if there's something to improve to prevent this, otherwise whatever we'll do will be extremely unreliable.

craigcurtin-dev, 2021-08-22:
Hmm,

Thats interesting - brand new install - brand new inverters - brand new gigabit switch that they are directly wired into and then across (with brand new CAT6e) cable to a Cisco switch that i am port mirroring on.

I will setup a PC that i can place on the same switch (so all of a 2 metre cable run to the inverters) later today and run the same tests.

WIll report back

Craig

craigcurtin-dev, 2021-08-23:
OK so 

1) New PC - Windows 10 - Python 3.8 installed
2) Plugged directly into the same 5 port Gigabit switch as the 3 inverters each of them connected with a 2 metre - brand new cat 5e cable
3) Output as per below from running the test
4) To Eliminate the Inverter/cable/landongle being the issue i tried it on a 2nd inverter - see results underneath.
5) Interestingly after running this the inverter disappears from the PV Master broadcast scan (udp port 48899) for a short period.

C:\Users\Administrator\Downloads\goodwe\home-assistant-goodwe-inverter-master>C:\Users\Administrator\AppData\Local\Programs\Python\Python38\python inverter_test.py
2021-08-23 15:45:59,786 __init__(623) - DEBUG: Using proactor: IocpProactor
2021-08-23 15:45:59,786 discover(647) - DEBUG: Probing inverter at 172.16.100.60:8899
2021-08-23 15:45:59,786 connection_made(327) - DEBUG: Send: 'aa55c07f0102000241'
2021-08-23 15:46:00,067 datagram_received(338) - DEBUG: Received: 'aa557fc001824d3038303843475735303030532d4250233130000000'
2021-08-23 15:46:00,067 datagram_received(343) - DEBUG: Invalid response length: 28
2021-08-23 15:46:00,067 connection_made(327) - DEBUG: Send: 'aa55c07f0102000241'
2021-08-23 15:46:00,302 datagram_received(338) - DEBUG: Received: 'aa557fc001824d3038303843475735303030532d42502331300000000000000000000000000039353030304250533230425730303137333630303431302d30343030312d3038473431302d30323033342d31320b1072'
2021-08-23 15:46:00,302 discover(671) - DEBUG: Probing EH inverter at 172.16.100.60:8899
2021-08-23 15:46:00,302 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-23 15:46:00,692 datagram_received(338) - DEBUG: Received: 'aa55f70342000113560000003900350030003000300042005000530047005700350030003000080008006500000000ffffffffffffffffffffffff'
2021-08-23 15:46:00,692 datagram_received(343) - DEBUG: Invalid response length: 59
2021-08-23 15:46:00,692 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-23 15:46:01,146 datagram_received(338) - DEBUG: Received: 'aa55f703420001'
2021-08-23 15:46:01,146 datagram_received(343) - DEBUG: Invalid response length: 7
2021-08-23 15:46:01,146 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-23 15:46:01,552 datagram_received(338) - DEBUG: Received: 'aa55f70342000113560000003900350030003000300042005000530047005700350030003000080008006500000000ffffffffffffffffffffffffffffffffffffff'
2021-08-23 15:46:01,552 datagram_received(343) - DEBUG: Invalid response length: 66
2021-08-23 15:46:01,552 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-23 15:46:02,021 datagram_received(338) - DEBUG: Received: 'aa55f70342000113560000003900350030003000300042005000530047005700350030003000080008006500000000ffffffffffffffffffffffffffffffffffffffffffffffff0a24'
2021-08-23 15:46:02,021 datagram_received(343) - DEBUG: Invalid response length: 73
2021-08-23 15:46:02,021 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-23 15:46:02,317 _timeout_heartbeat(361) - DEBUG: Re-try #5, closing socket
2021-08-23 15:46:02,317 discover(671) - DEBUG: Probing ET inverter at 172.16.100.60:8899
2021-08-23 15:46:02,317 connection_made(327) - DEBUG: Send: 'f70388b800213ac1'
2021-08-23 15:46:02,411 datagram_received(338) - DEBUG: Received: 'aa55f70342000113560000003900350030003000300042005000530047005700350030003000080008006500000000ffff'
2021-08-23 15:46:02,411 datagram_received(343) - DEBUG: Invalid response length: 49
2021-08-23 15:46:02,411 connection_made(327) - DEBUG: Send: 'f70388b800213ac1'
2021-08-23 15:46:02,536 datagram_received(338) - DEBUG: Received: 'aa55f7034200011356000000390035003000300030004200500053004700570035'
2021-08-23 15:46:02,536 datagram_received(343) - DEBUG: Invalid response length: 33
2021-08-23 15:46:02,536 connection_made(327) - DEBUG: Send: 'f70388b800213ac1'
2021-08-23 15:46:02,802 datagram_received(338) - DEBUG: Received: 'aa55f70342000113560000003900350030003000300042005000530047005700350030003000080008006500000000ffffffffffffffffffffffffffffffffffffffffffffffff0a24'
Traceback (most recent call last):
  File "inverter_test.py", line 21, in <module>
    inverter = asyncio.run(inverter.discover(IP_ADDRESS, PORT, TIMEOUT, RETRIES))
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:\Users\Administrator\Downloads\goodwe\home-assistant-goodwe-inverter-master\custom_components\goodwe\goodwe_inverter.py", line 672, in discover
    await i.read_device_info()
  File "C:\Users\Administrator\Downloads\goodwe\home-assistant-goodwe-inverter-master\custom_components\goodwe\goodwe_inverter.py", line 972, in read_device_info
    self.software_version = response[54:66].decode("ascii")
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)

C:\Users\Administrator\Downloads\goodwe\home-assistant-goodwe-inverter-master>

2nd Inverter

C:\Users\Administrator\Downloads\goodwe\home-assistant-goodwe-inverter-master>C:\Users\Administrator\AppData\Local\Programs\Python\Python38\python inverter_test.py
2021-08-23 15:53:28,426 __init__(623) - DEBUG: Using proactor: IocpProactor
2021-08-23 15:53:28,426 discover(647) - DEBUG: Probing inverter at 172.16.100.61:8899
[... cut ...]
  File "C:\Users\Administrator\Downloads\goodwe\home-assistant-goodwe-inverter-master\custom_components\goodwe\goodwe_inverter.py", line 972, in read_device_info
    self.software_version = response[54:66].decode("ascii")
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)

C:\Users\Administrator\Downloads\goodwe\home-assistant-goodwe-inverter-master>

Any ideas ?

Tomorrow i think i will replace one of the LAN dongles with the original WIFI dongle and see if that makes any difference ?

Craig

craigcurtin-dev, 2021-08-23:
Raining cats and dogs here today (and probably tomorrow) so probably will not get to shutdown the system and replace the dongle - will see if the weather improves

craigcurtin-dev, 2021-08-25:
OK swapped the wireless dongle in on one of the inverters and ran the inverter_test.py - i increased the timeouts to 30s to see if that made any difference - still appear to be getting the same info back now

It definitely makes a difference with what is being received back

Now remember this is an AC Coupled inverter only single phase - so no PV panels so the packets coming back may not be what you think they are (in terms of total registers etc)

C:\Users\Administrator\Downloads\goodwe\home-assistant-goodwe-inverter-master>C:\Users\Administrator\AppData\Local\Programs\Python\Python38\python inverter_test.py
2021-08-25 12:51:30,613 __init__(623) - DEBUG: Using proactor: IocpProactor
2021-08-25 12:51:30,613 discover(647) - DEBUG: Probing inverter at 172.16.100.115:8899
2021-08-25 12:51:30,613 connection_made(327) - DEBUG: Send: 'aa55c07f0102000241'
2021-08-25 12:51:30,895 datagram_received(338) - DEBUG: Received: 'aa557fc001824d3038303843475735303030532d42502331300000000000000000000000000039353030304250533231315730303439333630303431302d30343030312d3038473431302d30323033342d31320b1067'
2021-08-25 12:51:30,895 discover(671) - DEBUG: Probing EH inverter at 172.16.100.115:8899
2021-08-25 12:51:30,895 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-25 12:51:31,442 datagram_received(338) - DEBUG: Received: 'aa55f70342000113560000003900350030003000300042005000530047005700350030003000080008006500000000ffffffffffffffffffffffffffffffffffffffffffffffff0a24'
2021-08-25 12:51:31,442 datagram_received(343) - DEBUG: Invalid response length: 73
2021-08-25 12:51:31,442 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-25 12:51:50,907 _timeout_heartbeat(357) - DEBUG: Re-try #2
2021-08-25 12:51:50,907 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-25 12:51:51,426 datagram_received(338) - DEBUG: Received: 'aa55f70342000113560000003900350030003000300042005000530047005700350030003000080008006500000000ffffffffffffffffffffffffffffffffffffffffffffffff0a24'
2021-08-25 12:51:51,426 datagram_received(343) - DEBUG: Invalid response length: 73
2021-08-25 12:51:51,427 connection_made(327) - DEBUG: Send: 'f70388b800213ac10000'
2021-08-25 12:51:51,458 _timeout_heartbeat(361) - DEBUG: Re-try #4, closing socket
2021-08-25 12:51:51,458 discover(671) - DEBUG: Probing ET inverter at 172.16.100.115:8899
2021-08-25 12:51:51,458 connection_made(327) - DEBUG: Send: 'f70388b800213ac1'
2021-08-25 12:51:51,817 datagram_received(338) - DEBUG: Received: 'aa55f70342000113560000003900350030003000300042005000530047005700350030003000080008006500000000ffffffffffffffffffffffffffffffffffffffffffffffff0a24'
Traceback (most recent call last):
  File "inverter_test.py", line 21, in <module>
    inverter = asyncio.run(inverter.discover(IP_ADDRESS, PORT, TIMEOUT, RETRIES))
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:\Users\Administrator\Downloads\goodwe\home-assistant-goodwe-inverter-master\custom_components\goodwe\goodwe_inverter.py", line 672, in discover
    await i.read_device_info()
  File "C:\Users\Administrator\Downloads\goodwe\home-assistant-goodwe-inverter-master\custom_components\goodwe\goodwe_inverter.py", line 972, in read_device_info
    self.software_version = response[54:66].decode("ascii")
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)

C:\Users\Administrator\Downloads\goodwe\home-assistant-goodwe-inverter-master>

The first query string when i run it through node red is giving me this info as a rough debug

[image]

So i can see the model number and the serial number of the unit in there.

and it looks like some valid data - so it almost looks like we are asking for more data than the inverter can supply - i found a link to a doc last night that describes the aa55 protocol i will read through that some more now

What do you think ?

Craig

craigcurtin-dev, 2021-08-25:
OK i think i have it now - working through this document (not sure if you have seen this that describes the AA55 protocol ??)

http://yamasun.com.tw/upload/F_20170313191367UrC8jo.PDF

So the initial query that you send out  gets the following returned

aa557fc001824d3038303843475735303030532d42502331300000000000000000000000000039353030304250533231315730303439333630303431302d30343030312d3038473431302d30323033342d31320b1067

Which chop off the address stuff at the start is

82
4d30383038 - Should be the firmware version
43475735303030532d42 - Should be the model number
50233130000000000000000000000000 - Reserved
00393530303042505332313157303034 - Should be the serial #
39333630 - For a Solar inverter this would be the PV voltage in 0.1v
303431302d30343030312d30 - Internal Version
38 - Country Code
473431302d30323033342d31320b1067 - Not Sure on this yet

Is this how you are attempting to decode the stream and hence why it is failing ?

Craig

mletenay, 2021-08-25:
Well. that is the response for the initial 'aa55c07f0102000241' aka discovery request.
And the only really relevant info we get from it is the model name at offset 12:22, in your case - GW5000S-BP.
The actual inverter data are in subsequent aka read_running_data request.
There it seems your inverter responded to both EH and ET flavors, just the response has unexpected length.
So we just need to decode that one and figure out the offset.
See e.g. https://github.com/marcelblijleven/goodwe/blob/master/goodwe/eh.py for inspiration.
I'll try to look at it closer later tonight...

mletenay, 2021-08-28:
I suggest to try the 'aa55c07f0106000245' request.
If it responds, it will be the non-modbus protocol and ES/EM offests

craigcurtin-dev, 2021-08-28:
Yep here we go - here is the response

"aa557fc001868c0ea5001600000000020001f7000000520096006300630062000048004800030000010961001300b81383010961000105051383010200c300000000000001ee000000d30002001b0000079dfd760100300100000001000000d300080020000015081d090e0000000000000000000000000000000000000000000000000000000217000001f2040000361c084f1283"

Now to work out how to decode that 

Is there a list of the ES/EM offsets ?

Craig

craigcurtin-dev, 2021-08-29:
OK a bit more progress on understanding the whole thing now

1) Start up PV Master App - it broadcasts on 48899 to find all inverters and waits 10 seconds for replies
2) PVMaster then sends out the aa55c07f0102000241 to the inverter when you choose it from the list
3) It keeps sending this 241 packet until it is happy with the response - assume that as UDP may lose packets it is doing a CRC check - in the case of the final return packet i got aa557fc001824d3038303843475735303030532d42502331300000000000000000000000000039353030304250533230425730303137333630303431302d30343030312d3038473431302d30323033342d31320b1072 - 86 bytes in total so that seems to be the magic number in the return string - will confirm that shortly
4) Once it has this as a valid response which allows it to determine the model type it then moves to sending 
5) aa55c07f0106000245 - ES/EM query standard and receives back
6) aa557fc001868c0ecd000000000000020001f8000000520096000f0064006400004800486402000001097e0006ffa013800100000006072d1382000200dc0000000000000217000000  == 73 bytes - which as we have noted before appears valid - but no checksum

More investigation to come

craigcurtin-dev, 2021-08-29:
OK i have left this running on node-red just sendin the same query over and over again to the inverter - i will leave that for a couple of hours and see if i can see some patterns. Can you tell me how the checksum at the end of the packet is calculated  - also do you have any addition documentation for the aa55 protocol than i sent you before ?

Craig

mletenay, 2021-08-29:
I have put everything I know about the inverter protocols internal structure directly into code.
See https://github.com/marcelblijleven/goodwe/blob/master/goodwe/protocol.py and documentation of Aa55ProtocolCommand and ModbusProtocolCommand classes.

craigcurtin-dev, 2021-08-29:
OK fair enough - i will just keep stepping through it and see how i go

Craig

mletenay, 2021-08-29:
Ok, you seem to be lucky, that response to ES command is perfectly matching known offsets:
'vpv1': 0.0, 
'ipv1': 0.1, 
'ppv1': 0, 
'pv1_mode': 0, 
'pv1_mode_label': 'PV panels not connected', 
'vpv2': 0.0, 
'ipv2': 0.0, 
'ppv2': 0, 
'pv2_mode': 0, 
'pv2_mode_label': 'PV panels not connected', 
'ppv': 0, 
'vbattery1': 49.4, 
'battery_temperature': 24.6, 
'ibattery1': 9.5, 
'pbattery1': 469, 
'battery_charge_limit': 74, 
'battery_discharge_limit': 74, 
'battery_status': 0, 
'battery_soc': 60, 
'battery_soh': 98, 
'battery_mode': 2, 
'battery_mode_label': 'Discharge', 
'battery_warning': 0, 
'meter_status': 1, 
'vgrid': 228.3, 
'igrid': 2.3, 
'pgrid': 2, 
'fgrid': 49.83, 
'grid_mode': 1, 
'grid_mode_label': 
'Inverter On', 
'vload': 228.3, 
'iload': 2.6, 
'pload': 156, 
'fload': 49.83, 
'load_mode': 1, 
'load_mode_label': 'The inverter is connected to a load', 
'work_mode': 2, 
'work_mode_label': 'Normal (On-Grid)', 
'temperature': 22.9, 
'error_codes': 0, 
'e_total': 22963.2, 
'h_total': 46332, 
'e_day': 12.3, 
'e_load_day': 13.7, 
'e_load_total': 31348.0, 
'total_power': 393, 
'effective_work_mode': 1, 
'grid_in_out': 0, 
'grid_in_out_label': 'Idle', 
'pback_up': 535, 
'plant_power': 691, 
'diagnose_result': 117440576, 
'house_consumption': 467

It means the BP inverter talks exactly like ES/EM

craigcurtin-dev, 2021-08-29:
Excellent thanks Mate,

I was working through it one step at a time and was getting closer

I am just reading through all your stuff now.

Just to let you know in relation to the aa55 protocol – the address bytes and meaningless and are auto assigned – it just appears everyone has latched onto the same pair each time !!

The response from the inverter is just the addresses inverted

I think it came about because some of the original stuff for USB/RS485 said to set the inverter address to 247 i.e. f7

Craig

mletenay, 2021-08-29:
BP inverters support was added to both library and HA v0.8.2

closed 2021-08-29 by mletenay