mletenay/home-assistant-goodwe-inverter #45: GW5000L-DT Platform goodwe not ready yet: Unable to connect to the inverter at host

matda59, 2021-08-23:
Still not working for me. GW5000L-DT

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:268
Integration: Sensor (documentation, issues)
First occurred: 2:34:34 PM (1 occurrences)
Last logged: 2:34:34 PM
Platform goodwe not ready yet: Unable to connect to the inverter at host=192.168.0.165 port=8899, or your inverter is not supported yet. Failures=[InverterError("No valid response received to 'aa55c07f0102000241' request"), InverterError("No valid response received to 'f70388b800213ac10000' request"), InverterError("No valid response received to 'f70388b800213ac1' request"), InverterError("No valid response received to '7f03753100280409' request"), InverterError("No valid response received to 'aa55c07f0102000241' request")]; Retrying in background in 30 seconds

mletenay, 2021-08-29:
Does the problem still persist ?
Did you try to set the inverter_type explicitly ?

matda59, 2021-08-30:
Yes i tried that now.. now my logs are saying:

Logger: homeassistant.components.sensor
Source: custom_components/goodwe/goodwe/protocol.py:104
Integration: Sensor (documentation, issues)
First occurred: 10:12:23 AM (1 occurrences)
Last logged: 10:12:23 AM

Error while setting up goodwe platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/goodwe/sensor.py", line 85, in async_setup_platform
    inverter = await connect(
  File "/config/custom_components/goodwe/goodwe/goodwe.py", line 41, in connect
    await inverter.read_device_info()
  File "/config/custom_components/goodwe/goodwe/dt.py", line 107, in read_device_info
    response = await self._read_from_socket(self._READ_DEVICE_VERSION_INFO)
  File "/config/custom_components/goodwe/goodwe/inverter.py", line 80, in _read_from_socket
    return await command.execute(self.host, self.port, self.timeout, self.retries)
  File "/config/custom_components/goodwe/goodwe/protocol.py", line 104, in execute
    await on_response_received
custom_components.goodwe.goodwe.exceptions.MaxRetriesException

RatMondeo, 2021-08-30:
Hi,

I have exact same error as matda59.  I upgraded from a working setup (two versions back - not sure what version that was).
All was working well until I upgraded.
Rebooted. - same error
Added specific "NS" model of inverter. - same error
Current Config:
  - platform: goodwe
    ip_address: 192.168.0.168
    port: 8899
    network_timeout: 2
    network_retries: 3
    scan_interval: 30
    inverter_type: NS
    sensor_name_prefix: GoodWe

Any ideas appreciated. . . .

mletenay, 2021-08-30:
Sorry for that, it's really difficult to develop without access to all these specific inverters.
Could you please update to latest version (released just a moment ago).
And then adjust the logging in HA's configuration.yaml - add or adjust logger section:

```
logger:
  default: warning
  logs:
    custom_components.goodwe: debug
```

RatMondeo, 2021-08-30:
Thanks for the quick response.
Updated and restarted, same error as above
Removed type: NS - restarted - same error above

Let me know if you need more/other details.

Thanks Again

mletenay, 2021-08-30:
Well, I need the log files, otherwise I'm stuck.
Setup the logger as mentioned above, restart and then check the debug log records.

RatMondeo, 2021-08-30:
Yea, sorry. . . .
So that is just the "home-assistant.log" file?
How do I get it to you without everyone seeing it - as I am not sure if it is poor privacy to just attach here . . .

Any guidance appreciated.
Thanks Again

matda59, 2021-08-30:
Thanks mletenay:

Posted Home-assistant.log

2021-08-30 20:23:05 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-08-30 20:23:05 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up goodwe platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/goodwe/sensor.py", line 85, in async_setup_platform
    inverter = await connect(
  File "/config/custom_components/goodwe/goodwe/goodwe.py", line 41, in connect
    await inverter.read_device_info()
  File "/config/custom_components/goodwe/goodwe/dt.py", line 107, in read_device_info
    response = await self._read_from_socket(self._READ_DEVICE_VERSION_INFO)
  File "/config/custom_components/goodwe/goodwe/inverter.py", line 80, in _read_from_socket
    return await command.execute(self.host, self.port, self.timeout, self.retries)
  File "/config/custom_components/goodwe/goodwe/protocol.py", line 104, in execute
    await on_response_received
custom_components.goodwe.goodwe.exceptions.MaxRetriesException
2021-08-30 20:23:06 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/yarl/_url.py", line 170, in __new__
    port = val.port
  File "/usr/local/lib/python3.9/urllib/parse.py", line 180, in port
    raise ValueError("Port out of range 0-65535")
ValueError: Port out of range 0-65535

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zeroconf/__init__.py", line 183, in _async_zeroconf_hass_start
    await _async_register_hass_zc_service(hass, aio_zc, uuid)
  File "/usr/src/homeassistant/homeassistant/components/zeroconf/__init__.py", line 249, in _async_register_hass_zc_service
    params["internal_url"] = get_url(hass, allow_external=False)
  File "/usr/src/homeassistant/homeassistant/helpers/network.py", line 61, in get_url
    return _get_internal_url(
  File "/usr/src/homeassistant/homeassistant/helpers/network.py", line 138, in _get_internal_url
    internal_url = yarl.URL(hass.config.internal_url)
  File "/usr/local/lib/python3.9/site-packages/yarl/_url.py", line 172, in __new__
    raise ValueError(
ValueError: Invalid URL: port can't be converted to integer
2021-08-30 20:23:06 ERROR (stream_worker) [libav.mp3float] Header missing

supergroover, 2021-08-30:
I have also the sane error with my Goodwe 6000-DT.
Version 0.8.1 works fine, versions 0.8.2 and 0.8.3 not.
- platform: goodwe
  ip_address: 192.168.1.193
  port: 8899
  network_timeout: 2
  network_retries: 3
  scan_interval: 30
  sensor_name_prefix: GoodWe
  inverter_type: DT           # One of ET, EH, ES, EM, DT, NS, XS or None to detect inverter type automatically
  include_unknown_sensors: false

2021-08-30 11:57:31 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Connecting to DT family inverter at 192.168.1.193:8899
2021-08-30 11:57:31 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Connection made to address ('192.168.1.193', 8899)
2021-08-30 11:57:31 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81
2021-08-30 11:57:33 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-08-30 11:57:33 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81
2021-08-30 11:57:35 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-08-30 11:57:35 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81
2021-08-30 11:57:37 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #4 of 3
2021-08-30 11:57:37 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81
2021-08-30 11:57:39 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-08-30 11:57:39 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up goodwe platform for sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/goodwe/sensor.py", line 85, in async_setup_platform
inverter = await connect(
File "/config/custom_components/goodwe/goodwe/goodwe.py", line 41, in connect
await inverter.read_device_info()
File "/config/custom_components/goodwe/goodwe/dt.py", line 107, in read_device_info
response = await self._read_from_socket(self._READ_DEVICE_VERSION_INFO)
File "/config/custom_components/goodwe/goodwe/inverter.py", line 80, in _read_from_socket
return await command.execute(self.host, self.port, self.timeout, self.retries)
File "/config/custom_components/goodwe/goodwe/protocol.py", line 104, in execute
await on_response_received
custom_components.goodwe.goodwe.exceptions.MaxRetriesException

mletenay, 2021-08-30:
Thanks, I probably see where's the problem, the DT inverters have default address 7F and not F7 and it got messed up during recent code improvements.
The fix will be there today ...

mletenay, 2021-08-30:
OK, version 0.8.4 with fixed DT/D-NS/XS communication was just released.
Please update and sorry for the inconvenience.

matda59, 2021-08-30:
Hi again, still not working...

sensor:
  - platform: goodwe
    ip_address: 192.168.0.165
    port: 8899
    network_timeout: 2
    network_retries: 3
    scan_interval: 30
    inverter_type: DT
    sensor_name_prefix: GoodWe
    include_unknown_sensors: false

2021-08-30 21:57:01 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-08-30 21:57:01 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up goodwe platform for sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/goodwe/sensor.py", line 85, in async_setup_platform
inverter = await connect(
File "/config/custom_components/goodwe/goodwe/goodwe.py", line 41, in connect
await inverter.read_device_info()
File "/config/custom_components/goodwe/goodwe/dt.py", line 106, in read_device_info
response = await self._read_from_socket(self._READ_DEVICE_VERSION_INFO)
File "/config/custom_components/goodwe/goodwe/inverter.py", line 80, in _read_from_socket
return await command.execute(self.host, self.port, self.timeout, self.retries)
File "/config/custom_components/goodwe/goodwe/protocol.py", line 103, in execute
await on_response_received
custom_components.goodwe.goodwe.exceptions.MaxRetriesException
2021-08-30 21:57:01 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/yarl/_url.py", line 170, in __new__
port = val.port
File "/usr/local/lib/python3.9/urllib/parse.py", line 180, in port
raise ValueError("Port out of range 0-65535")
ValueError: Port out of range 0-65535
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/zeroconf/__init__.py", line 183, in _async_zeroconf_hass_start
await _async_register_hass_zc_service(hass, aio_zc, uuid)
File "/usr/src/homeassistant/homeassistant/components/zeroconf/__init__.py", line 249, in _async_register_hass_zc_service
params["internal_url"] = get_url(hass, allow_external=False)
File "/usr/src/homeassistant/homeassistant/helpers/network.py", line 61, in get_url
return _get_internal_url(
File "/usr/src/homeassistant/homeassistant/helpers/network.py", line 138, in _get_internal_url
internal_url = yarl.URL(hass.config.internal_url)
File "/usr/local/lib/python3.9/site-packages/yarl/_url.py", line 172, in __new__
raise ValueError(
ValueError: Invalid URL: port can't be converted to integer
2021-08-30 21:57:02 ERROR (stream_worker) [libav.mp3float] Header missing

mletenay, 2021-08-30:
I need the debug log output, especially those
_DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81_
lines and their logged responses (if any).

supergroover, 2021-08-30:
> OK, version 0.8.4 with fixed DT/D-NS/XS communication was just released.
> Please update and sorry for the inconvenience.

It works again.
Thanks voor your fast respons and great work!

matda59, 2021-08-30:
Hi Mletenay,
here is the output from home-assistant.log:

2021-08-31 08:44:51 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Connecting to DT family inverter at 192.168.0.165:8899
2021-08-31 08:44:51 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Connection made to address ('192.168.0.165', 8899)
2021-08-31 08:44:51 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f03753100280409
2021-08-31 08:44:53 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-08-31 08:44:53 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f03753100280409
2021-08-31 08:44:54 ERROR (MainThread) [homeassistant.components.automation] Blueprint Low battery level detection & notification for all battery sensors generated invalid automation with inputs OrderedDict([('day', 7), ('actions', ["message: 'Low battery warning for: {{sensors}}'"]), ('threshold', 90)]): expected dictionary @ data['action'][0]['default'][0]. Got None
2021-08-31 08:44:55 WARNING (MainThread) [homeassistant.components.webhook] Received message for unregistered webhook 740667d35625c8b1531feba566a42dd071a542606ae47c4890404fd035f65d46 from 192.168.0.1
2021-08-31 08:44:55 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-08-31 08:44:55 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f03753100280409
2021-08-31 08:44:57 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #4 of 3
2021-08-31 08:44:57 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f03753100280409
2021-08-31 08:44:59 WARNING (MainThread) [homeassistant.config_entries] Config entry 'broadcom IR receiver' for broadlink integration not ready yet: [Errno -4000] Network timeout; Retrying in background
2021-08-31 08:44:59 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-08-31 08:44:59 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up goodwe platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/goodwe/sensor.py", line 85, in async_setup_platform
    inverter = await connect(
  File "/config/custom_components/goodwe/goodwe/goodwe.py", line 41, in connect
    await inverter.read_device_info()
  File "/config/custom_components/goodwe/goodwe/dt.py", line 106, in read_device_info
    response = await self._read_from_socket(self._READ_DEVICE_VERSION_INFO)
  File "/config/custom_components/goodwe/goodwe/inverter.py", line 80, in _read_from_socket
    return await command.execute(self.host, self.port, self.timeout, self.retries)
  File "/config/custom_components/goodwe/goodwe/protocol.py", line 103, in execute
    await on_response_received
custom_components.goodwe.goodwe.exceptions.MaxRetriesException
2021-08-31 08:45:01 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/yarl/_url.py", line 170, in __new__
    port = val.port
  File "/usr/local/lib/python3.9/urllib/parse.py", line 180, in port
    raise ValueError("Port out of range 0-65535")
ValueError: Port out of range 0-65535

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zeroconf/__init__.py", line 183, in _async_zeroconf_hass_start
    await _async_register_hass_zc_service(hass, aio_zc, uuid)
  File "/usr/src/homeassistant/homeassistant/components/zeroconf/__init__.py", line 249, in _async_register_hass_zc_service
    params["internal_url"] = get_url(hass, allow_external=False)
  File "/usr/src/homeassistant/homeassistant/helpers/network.py", line 61, in get_url
    return _get_internal_url(
  File "/usr/src/homeassistant/homeassistant/helpers/network.py", line 138, in _get_internal_url
    internal_url = yarl.URL(hass.config.internal_url)
  File "/usr/local/lib/python3.9/site-packages/yarl/_url.py", line 172, in __new__
    raise ValueError(
ValueError: Invalid URL: port can't be converted to integer

mletenay, 2021-08-31:
It seems you don't receive any response even to initial "_READ_DEVICE_VERSION_INFO" request.
So either L-DT is different from DT or just your inverter is configured differently ?
Are you able to connect to the inverter via the mobile app ?

RatMondeo, 2021-08-31:
My GW5000D-NS is still giving the same error.
Which mobile app should we test with, Good we SEMS Portal?

Thanks Again

mletenay, 2021-09-01:
PvMaster or SolarGo. 
(Sems app load data from the cloud and does not connect to inverter directly.)

RatMondeo, 2021-09-01:
So I was able to connect via the SolarGo app on my GW5000D-NS.  Anything I should test/check in the app?
Thanks

mletenay, 2021-09-02:
The best would be if you can sniff the packet traffic.
If not, than somewhere (Advanced settings ?) should be communication address of the inverter, usually someting like 247 or 127.
That's what we need to check/confirm.

RatMondeo, 2021-09-02:
I think it is 247. Found it under RS485 Setting.

Thanks Again

mletenay, 2021-09-02:
So it seems (some) DT inverters use 127 and some D-NS use 247.
Coincidentally I just yesterday adjusted the code to have the address configurable.
When I'll release new version, there will be optional parameter comm_address, you'll need to set that to 247 then.

mletenay, 2021-09-02:
I've just release v0.8.6 where the comm_address parameter is exposed.
Try to set inverter_type to DT and comm_address to 247 to see if helps.

matda59, 2021-09-02:
Thanks mletenay -

Todays error:

sensor:
  - platform: goodwe
    ip_address: 192.168.0.165
    port: 8899
    network_timeout: 2
    network_retries: 3
    scan_interval: 30
    inverter_type: DT
    comm_address: 247
    sensor_name_prefix: GoodWe
    include_unknown_sensors: false

2021-09-03 09:03:02 WARNING (SyncWorker_0) [homeassistant.util.yaml.loader] YAML file /config/configuration.yaml contains duplicate key "logger". Check lines 109 and 118
2021-09-03 09:03:02 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-03 09:03:02 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration sonoff which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-03 09:03:02 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration goodwe which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-03 09:03:03 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Connecting to DT family inverter at 192.168.0.165:8899
2021-09-03 09:03:03 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.165', 8899)
2021-09-03 09:03:05 ERROR (MainThread) [homeassistant.components.automation] Blueprint Low battery level detection & notification for all battery sensors generated invalid automation with inputs OrderedDict([('day', 7), ('actions', ["message: 'Low battery warning for: {{sensors}}'"]), ('threshold', 90)]): expected dictionary @ data['action'][0]['default'][0]. Got None
2021-09-03 09:03:05 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-03 09:03:05 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.165', 8899)
2021-09-03 09:03:05 WARNING (MainThread) [homeassistant.components.webhook] Received message for unregistered webhook 991501efdb0f407b356f98bb7acb086ce43161ec5bcb3f04eb9da6d44160f127 from 192.168.0.1
2021-09-03 09:03:07 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-03 09:03:07 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.165', 8899)
2021-09-03 09:03:09 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-03 09:03:09 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.165', 8899)
2021-09-03 09:03:09 WARNING (MainThread) [homeassistant.config_entries] Config entry 'broadcom IR receiver' for broadlink integration not ready yet: [Errno -4000] Network timeout; Retrying in background
2021-09-03 09:03:11 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-09-03 09:03:11 WARNING (MainThread) [homeassistant.components.sensor] Platform goodwe not ready yet; Retrying in background in 30 seconds
2021-09-03 09:03:11 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/yarl/_url.py", line 170, in __new__
port = val.port
File "/usr/local/lib/python3.9/urllib/parse.py", line 180, in port
raise ValueError("Port out of range 0-65535")
ValueError: Port out of range 0-65535
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/zeroconf/__init__.py", line 183, in _async_zeroconf_hass_start
await _async_register_hass_zc_service(hass, aio_zc, uuid)
File "/usr/src/homeassistant/homeassistant/components/zeroconf/__init__.py", line 249, in _async_register_hass_zc_service
params["internal_url"] = get_url(hass, allow_external=False)
File "/usr/src/homeassistant/homeassistant/helpers/network.py", line 61, in get_url
return _get_internal_url(
File "/usr/src/homeassistant/homeassistant/helpers/network.py", line 138, in _get_internal_url
internal_url = yarl.URL(hass.config.internal_url)
File "/usr/local/lib/python3.9/site-packages/yarl/_url.py", line 172, in __new__
raise ValueError(
ValueError: Invalid URL: port can't be converted to integer
2021-09-03 09:03:11 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Connecting to DT family inverter at 192.168.0.165:8899
2021-09-03 09:03:11 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.165', 8899)
2021-09-03 09:03:12 ERROR (stream_worker) [libav.mp3float] Header missing
2021-09-03 09:03:12 ERROR (stream_worker) [libav.mp3float] Header missing
2021-09-03 09:03:13 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-03 09:03:13 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.165', 8899)
2021-09-03 09:03:15 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-03 09:03:15 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.165', 8899)
2021-09-03 09:03:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-03 09:03:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.165', 8899)
2021-09-03 09:03:19 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-09-03 09:03:54 ERROR (stream_worker) [homeassistant.components.stream.worker] Error demuxing stream:
2021-09-03 09:04:04 ERROR (stream_worker) [libav.mp3float] Header missing

mletenay, 2021-09-03:
Hmm, it still does not respond to initial request.
Just to make it clear - @matda59 it was actually never working yet, @RatMondeo you had it working at one point in time and now it stopped, right ?

RatMondeo, 2021-09-03:
Yes, that is the case.

mletenay, 2021-09-03:
@RatMondeo Are you able to checkout the sources and run them in python ?

RatMondeo, 2021-09-04:
Sorry, pretty sure that is beyond me .. .. .

matda59, 2021-09-04:
That is correct. It has never worked.. I can do whatever UAT you request

mletenay, 2021-09-04:
@matda59 Can you connect to inverter via the SolarGo or PvMaster mobile app ?

mletenay, 2021-09-04:
@RatMondeo I'm sorry but I can't figure out why it no longer works for you.
Can you increase the log level of the plugin and test it without setting explicit inverter_type and test it with both 247 and 127 versions of comm_address ?

RatMondeo, 2021-09-04:
Sounds like a plan.  I will check tomorrow when the sun is up......

Thanks

RatMondeo, 2021-09-05:
> @RatMondeo I'm sorry but I can't figure out why it no longer works for you.
> Can you increase the log level of the plugin and test it without setting explicit inverter_type and test it with both 247 and 127 versions of comm_address ?

This got it . . . .  It worked as soon as i removed the inverter type setting.   I continued to use the 247 inverter type setting and it works fine.

Thanks for working thought this with me.
Let me know if you need any additional info to assist with debugging this if I can help, I will . . . ..

mletenay, 2021-09-05:
Can you post the debug log ?
It would be helpful to see what inverter family the discovery method selected and why.

RatMondeo, 2021-09-05:
Hi,

Hope this helps,

Debug:
2021-09-06 06:56:19 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.0.168', 8899)
2021-09-06 06:56:19 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Received: aa557f039215090606380108c400000b660000ffffffffffffffffffffffffffffffffffffffffffff0965ffffffff0000ffffffff1385ffffffff000000000000000000000000ffffffffffffffffffffffffffffffff000effffffff000000002271000003bb00490000ffff0000ffff0000ffff0000ffffffffffffffffffff09600b65ffffffffffff0190ffffffffffff01940054e213

Serial: 55000DSN212W####

ES: test
2021-09-06 07:02:32 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Connecting to ES family inverter at 192.168.0.168:8899
2021-09-06 07:02:32 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.0.168', 8899)
2021-09-06 07:02:34 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-06 07:02:34 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.0.168', 8899)
2021-09-06 07:02:36 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-06 07:02:36 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.0.168', 8899)
2021-09-06 07:02:38 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3

mletenay, 2021-09-05:
Thanks, it helps.

mletenay, 2021-09-05:
@RatMondeo Just one more more feedback please.
Can you post the logs with and without the explicit NS inverter_type config ?
(You can strip the responses after some 10th character to not expose serial).
I still need to figure out where's the problem.

RatMondeo, 2021-09-05:
Let me know if this is what you were after . . . .

Without NS:
  - platform: goodwe
    ip_address: 192.168.0.168
    port: 8899
    comm_address: 247
    network_timeout: 2
    network_retries: 3
    scan_interval: 30
#    inverter_type: ??
    sensor_name_prefix: GoodWe

Debug:
2021-09-06 08:33:46 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.0.168', 8899)
2021-09-06 08:33:46 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Received: aa557f039215090608211c084c003a0ae0000bffffffffffffffffffffffffffffffffffffffffffff0971ffffffff003cffffffff1385ffffffff000005840001000000000000ffffffffffffffffffffffffffffffff0065ffffffff000800002279000003bd00490000ffff0000ffff0000ffff0000ffffffffffffffffffff09600e80ffffffffffff0190ffffffffffff019400540259

With NS:
  - platform: goodwe
    ip_address: 192.168.0.168
    port: 8899
    comm_address: 247
    network_timeout: 2
    network_retries: 3
    scan_interval: 30
    inverter_type: NS
    sensor_name_prefix: GoodWe

2021-09-06 08:38:20 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Connecting to NS family inverter at 192.168.0.168:8899
2021-09-06 08:38:20 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.168', 8899)
2021-09-06 08:38:22 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-06 08:38:22 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.168', 8899)
2021-09-06 08:38:24 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-06 08:38:24 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.168', 8899)
2021-09-06 08:38:26 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-06 08:38:26 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.0.168', 8899)
2021-09-06 08:38:28 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket

mletenay, 2021-09-05:
Isn't there a request f703753100281a81 when NS is explicitly specified ?

mletenay, 2021-09-05:
Oh, I finally see some difference, when NS is NOT specified as inverter, it ignores the comm_address and goes to 0x7f (which is weirdly correct). When it IS explicitly specified, the comm_address is properly taken into account (and does not work).
So SolarGo is misleading with the 247 address.
Re-test it without the comm_address parameter please, again with both NS scenarios.
(It's a pain to develop this wait without direct access to hw ....)

RatMondeo, 2021-09-05:
Hope this helps  . .

Now working in  both NS scenarios

Also, Updated to v0.8.9

 - platform: goodwe
    ip_address: 192.168.0.168
    port: 8899
    network_timeout: 2
    network_retries: 3
    scan_interval: 30
    inverter_type: NS
    sensor_name_prefix: GoodWe

2021-09-06 09:08:39 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.0.168', 8899)
2021-09-06 09:08:39 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Received: aa557f0392150906090814084600350b430011ffffffffffffffffffffffffffffffffffffffffffff0971ffffffff0040ffffffff1385ffffffff0000070f0001000000000000ffffffffffffffffffffffffffffffff0094ffffffff001000002281000003bd00490000ffff0000ffff0000ffff0000ffffffffffffffffffff09600e80ffffffffffff0190ffffffffffff01940054ef3e

  - platform: goodwe
    ip_address: 192.168.0.168
    port: 8899
    network_timeout: 2
    network_retries: 3
    scan_interval: 30
    sensor_name_prefix: GoodWe

2021-09-06 09:13:41 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.0.168', 8899)
2021-09-06 09:13:41 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Received: aa557f0392150906090d17084500410b070015ffffffffffffffffffffffffffffffffffffffffffff0964ffffffff004effffffff1387ffffffff0000075b0001000000000000ffffffffffffffffffffffffffffffff009fffffffff001100002282000003be00490000ffff0000ffff0000ffff0000ffffffffffffffffffff09600e7cffffffffffff0190ffffffffffff0194005469d5

mletenay, 2021-09-06:
@matda59 Can you update to latest version, remove all optional parameters (leave only ip_address there) and try again ?
Can you even connect to the inverter via SolarGo or PvMaster mobile app ?

matda59, 2021-09-07:
hiya,

log output after only having these lines in the config yaml

sensor:
  - platform: goodwe
    ip_address: 192.168.0.165

2021-09-07 11:03:56 WARNING (SyncWorker_0) [homeassistant.util.yaml.loader] YAML file /config/configuration.yaml contains duplicate key "logger". Check lines 101 and 110
2021-09-07 11:03:56 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration goodwe which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-07 11:03:56 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-07 11:03:56 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration sonoff which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-07 11:03:57 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Probing inverter at 192.168.0.165:8899
2021-09-07 11:03:57 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.0.165', 8899)
2021-09-07 11:03:59 ERROR (MainThread) [homeassistant.components.automation] Blueprint Low battery level detection & notification for all battery sensors generated invalid automation with inputs OrderedDict([('day', 7), ('actions', ["message: 'Low battery warning for: {{sensors}}'"]), ('threshold', 90)]): expected dictionary @ data['action'][0]['default'][0]. Got None
2021-09-07 11:03:59 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-07 11:03:59 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.0.165', 8899)
2021-09-07 11:04:01 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-07 11:04:01 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.0.165', 8899)
2021-09-07 11:04:03 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-07 11:04:03 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.0.165', 8899)
2021-09-07 11:04:04 WARNING (MainThread) [homeassistant.config_entries] Config entry 'broadcom IR receiver' for broadlink integration not ready yet: [Errno -4000] Network timeout; Retrying in background
2021-09-07 11:04:05 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-09-07 11:04:05 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Probing EH inverter at 192.168.0.165:8899
2021-09-07 11:04:05 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.0.165', 8899)
2021-09-07 11:04:07 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform goodwe is taking over 10 seconds.
2021-09-07 11:04:07 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-07 11:04:07 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.0.165', 8899)
2021-09-07 11:04:09 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-07 11:04:09 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.0.165', 8899)
2021-09-07 11:04:11 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-07 11:04:11 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.0.165', 8899)
2021-09-07 11:04:13 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-09-07 11:04:13 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Probing ET inverter at 192.168.0.165:8899
2021-09-07 11:04:13 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.0.165', 8899)
2021-09-07 11:04:15 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-07 11:04:15 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.0.165', 8899)
2021-09-07 11:04:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-07 11:04:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.0.165', 8899)
2021-09-07 11:04:19 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-07 11:04:19 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f70388b800213ac1 to ('192.168.0.165', 8899)
2021-09-07 11:04:21 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-09-07 11:04:21 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Probing DT inverter at 192.168.0.165:8899
2021-09-07 11:04:21 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f03753100280409 to ('192.168.0.165', 8899)
2021-09-07 11:04:23 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-07 11:04:23 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f03753100280409 to ('192.168.0.165', 8899)
[... cut ...]
2021-09-07 11:06:52 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-09-07 11:06:52 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Probing ES inverter at 192.168.0.165:8899
2021-09-07 11:06:52 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.0.165', 8899)
2021-09-07 11:06:54 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-07 11:06:54 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: aa55c07f0102000241 to ('192.168.0.165', 8899)

mletenay, 2021-09-07:
Ok, it does not respond to anything, or it was configured with non-standard communication address.
(And from the logs I see you're not updated to latest version yet, but that shouldn't be a cause).
Can you connect the inverter via SolarGo or PvMaster mobile app ? Can you look at advanced settings there to see to the comm_address ?

dphroon, 2021-09-08:
Hi,
I have a GW5000D-NS but cannot seem to get it the component working. Firmware on inverter recently updated, not working on previous firmware or new firmware.

I've tried using settings that @RatMondeo has used/tested in this post, but still not working

Inverter responds to broadcast on 48899, but never responds to any other messages

Any ideas?

mletenay, 2021-09-08:
@dphroon First thing to check is always the mobile app SolarGo or PvMaster.
If that works, than there's hope.

dphroon, 2021-09-08:
No luck on either, unfortunately. Finds the device, but can't connect

fizcris, 2021-09-08:
Guys, be aware that if a service is flooding the udp with requests the Inverter will stop communicating with all the the devices. 

E.g: when I rise the frequency of home assistant to 2hz I cannot communicate with the inverter via app/ other devices anymore, and the hass data transfer lowers to 1 success read per minute at most.

mletenay, 2021-09-08:
@dphroon That looks bad for you. I've seen somewhere mentioned the UDP communication is on some inverters only supported since ARM firmware 13 or so. So maybe you have latest firmware but no latest ARM (whatever it means).
Can you raise the debug level and check the log if at least the initial aa55 kind discovery request has some response ?

dphroon, 2021-09-08:
2021-09-08 17:20:17 DEBUG (MainThread) [custom_components.goodwe.goodwe] Connecting to NS family inverter at 192.168.X.XX:8899
2021-09-08 17:20:17 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.X.XX', 8899)
2021-09-08 17:20:19 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-08 17:20:19 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.X.XX', 8899)
2021-09-08 17:20:21 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-08 17:20:21 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.X.XX', 8899)
2021-09-08 17:20:23 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-08 17:20:23 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: f703753100281a81 to ('192.168.X.XX', 8899)
2021-09-08 17:20:25 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket

This is from ./inverter_scan.py

2021-09-07 17:16:58,014 __init__(59) - DEBUG: Using selector: EpollSelector
2021-09-07 17:16:58,014 search_inverters(116) - DEBUG: Searching inverters by broadcast to port 48899
2021-09-07 17:16:58,014 _send_request(36) - DEBUG: Sent: 574946494b49542d3231343032382d52454144 to None
2021-09-07 17:16:58,131 datagram_received(50) - DEBUG: Received: 3XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX2
Located inverter at IP: 192.168.X.XX, mac: XXXXXXXXXXXX, name: Solar-WiFiXXXXXXXX
Trying command: AA55C07F0102000241
No response to AA55C07F0102000241 command
Trying command: 7F03753100280409
No response to 7F03753100280409 command
Trying command: 197d0001000dff045e50303036564657f6e60d
No response to 197d0001000dff045e50303036564657f6e60d command
Trying command: 680241b132303757313234320100ae16
No response to 680241b132303757313234320100ae16 command
Trying command: 680241b132343231573730320100ae16
No response to 680241b132343231573730320100ae16 command
Identifying inverter at IP: 192.168.X.XX
2021-09-07 17:16:58,133 __init__(59) - DEBUG: Using selector: EpollSelector
2021-09-07 17:16:58,133 discover(67) - DEBUG: Probing inverter at 192.168.X.XX
2021-09-07 17:16:58,134 _send_request(36) - DEBUG: Sent: aa55c07f0102000241 to ('192.168.X.XX', 8899)
^C2021-09-07 17:17:01,992 discover(97) - DEBUG: Probing ET inverter at 192.168.X.XX
2021-09-07 17:17:01,992 _send_request(36) - DEBUG: Sent: f70388b800213ac1 to ('192.168.X.XX', 8899)
^CTraceback (most recent call last):

It appears to hang at this point (or I'm not patient enough).

I tried a couple of different config options today - both produced exactly the same log. I'll run a few tomorrow and list config with log results

mletenay, 2021-09-08:
@dphroon Keep trying, I'm glad to help as soon you get any kind of answer from inverter.
But to improve your mood in case you will not succeed, look here #65 - there may be a way via the TCP gateway.
(Once we actually implement it ;-)

mletenay, 2021-09-08:
@dphroon Read this issue https://github.com/jantenhove/GoodWeLogger/issues/28. It is about XS inverters, but should apply also to DT. At the end of the thread they mention arm firmware updates ...

dphroon, 2021-09-09:
@mletenay thanks!

Sometimes it just takes one person (in that post you linked) to phrase it correctly... **ARM-Firmware**

Previously, I had already found that download, held my breath and updated the firmware myself, but to no avail. The firmware in that download for the GW5000D-NS is too old.

I had raised a service request with GoodWe to do an OTA inverter firmware update, which they most pleasantly did, but to no avail.

It turns out to be so easy. **Raise a service request with GoodWe to do an OTA ARM-firmware update**. Just like that. 10 minutes later, and everything works!

mletenay, 2021-09-09:
Glad to hear that !
I'll adjust the README to add this hint for people with troubles like you ...
@matda59 You should probably do the same - ask for ARM firmware update.

matda59, 2021-09-10:
thanks for this, i tried to raise a service request, but the online form refuses to move forward with a "Unable to process your request" error that drops down. That's brilliant can't raise a support ticket. i sent them an email. let's see.

pappajj, 2021-09-14:
This isn't in the readme yet! I'm having the same issue, have just requested a an ota arm upgrade and firmware from goodwe, will report back