mletenay/home-assistant-goodwe-inverter #46: No longer working with DT-series after recent update

mareklab, 2021-08-24:
After a recent update via HACS, the integration stopped working with Home Assistant.

It still communicates with my inverter :

2021-08-24 13:43:37 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Connecting to DT family inverter at 192.168.123.10:8899 2021-08-24 13:43:37 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Send: '7f03753100280409' 2021-08-24 13:43:38 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Received: 'aa557f0350000000000001353830303044545332303957303134324757384b2d4454202020ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff000003f203f2000802d800310001000004e4'

But then there is an error directly in HA:

2021-08-24 13:43:48 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved Traceback (most recent call last): File "/config/custom_components/goodwe/sensor.py", line 157, in async_refresh self.entity.update_value(inverter_response) File "/config/custom_components/goodwe/sensor.py", line 196, in update_value self.async_schedule_update_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 621, in async_schedule_update_ha_state self.async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 464, in async_write_ha_state self._async_write_ha_state() File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 500, in _async_write_ha_state attr.update(self.state_attributes or {}) ValueError: dictionary update sequence element #0 has length 77; 2 is required

I guess the problem is here:

**ValueError: dictionary update sequence element #0 has length 77; 2 is required**

mletenay, 2021-08-24:
Try to add include_unknown_sensors: true to your HA config.
Meanwhile I'll check the sources for the problem, sorry for the inconvenience.

mletenay, 2021-08-24:
I can't reproduce the problem (on ET inverter),  neither can't find anything obviously wrong in the code.
What HA version do you have ?

mareklab, 2021-08-24:
It is the latest, 2021.8.8. I use it on RPi 4 with Docker.
I will do one more test: I pull the full code directly from your GH repo (last night I've installed it via HACS), add the setting you recommended, and will let you know.

mareklab, 2021-08-24:
@mletenay unfortunately it did not help. 
The component has correctly connected to my inverter:
```
2021-08-24 16:44:17 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Probing DT inverter at 192.168.123.10:8899
2021-08-24 16:44:17 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Send: '7f03753100280409'
2021-08-24 16:44:17 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Received: 'aa557f0350000000000001353830303044545332303957303134324757384b2d4454202020ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff000003f203f2000802d800310001000004e4'
2021-08-24 16:44:17 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Detected DT protocol inverter GW8K-DT, S/N:58000<removed>
2021-08-24 16:44:17 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Send: '7f0375940049d5c2'
2021-08-24 16:44:17 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Received: 'aa557f0392150818102b1b0ac3000613f40008ffffffffffffffffffffffffffffffff102910220ff0094409650930000a000a000a1390138c13880000028300010000000000000000000000000000000000000000ffff01c5ffffffffffffffffffffffffffff00200000000000000000ffffffffffffffffffffffffffffffff020018620c6000000000000002d800810000ffff0054f42f'
```
But then it started throwing these errors on regular basis:

```
2021-08-24 16:47:17 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Send: '7f0375940049d5c2'
2021-08-24 16:47:17 DEBUG (MainThread) [custom_components.goodwe.goodwe_inverter] Received: 'aa557f0392150818102e1b0aee000613a40007ffffffffffffffffffffffffffffffff1015100e0fd7093809570924000a000900091388138813880000026f00010000000000000000000000000000000000000000ffff01c5ffffffff003b000087c9000007ad00200000000000000000ffffffffffffffffffffffffffffffff0200183e0c6300000000000002d80081000000000054461f'
2021-08-24 16:47:36 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/goodwe/sensor.py", line 157, in async_refresh
    self.entity.update_value(inverter_response)
  File "/config/custom_components/goodwe/sensor.py", line 196, in update_value
    self.async_schedule_update_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 621, in async_schedule_update_ha_state
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 464, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 500, in _async_write_ha_state
    attr.update(self.state_attributes or {})
ValueError: dictionary update sequence element #0 has length 77; 2 is required

```

So I have rolled back to the previous version that worked well for me... (attached: 
[goodwe.zip](https://github.com/mletenay/home-assistant-goodwe-inverter/files/7040188/goodwe.zip)
 )

mletenay, 2021-08-24:
I've found stupid mistake in DT specific code.
Sorry for the inconvenience once again, I'm in process of moving parts of the code to new library, there are already some unit tests prepared and mistakes like this should not happen anymore.
Release v0.8.1 contains the fix, please update.

closed 2021-08-24 by mletenay

mareklab, 2021-08-24:
Thank you, I do appreciate your work 🥇 

I will test it once the sun is up again for me :)

mareklab, 2021-08-25:
It seems to work well, thanks againg for help 👍