mletenay/home-assistant-goodwe-inverter #47: ET - Mapping of Smart meter values

Delorien84, 2021-08-24:
Hello,
 I have notice that unknown values: `Unknown sensor2@10`, `Unknown sensor2@12` and `Unknown sensor2@14` are power values (in W) of Smart Meter of each phases connected at the input of the grid. Showing buying and selling power on each phases. Value `Unknown sensor2@16` seems to be sums of the phases.  Minus values means buying, plus value means selling.

My customization:
```
sensor.goodwe_xxx10:
  unit_of_measurement: W
  friendly_name: GoodWe Smart Meter L1 Power
  device_class: power
  state_class: measurement
  icon: mdi:power-plug-outline
sensor.goodwe_xxx12:
  unit_of_measurement: W
  friendly_name: GoodWe Smart Meter L2 Power
  device_class: power
  state_class: measurement
  icon: mdi:power-plug-outline
sensor.goodwe_xxx14:
  unit_of_measurement: W
  friendly_name: GoodWe Smart Meter L3 Power
  device_class: power
  state_class: measurement
  icon: mdi:power-plug-outline
sensor.goodwe_xxx16:
  unit_of_measurement: W
  friendly_name: GoodWe Smart Meter LSum Power
  device_class: power
  state_class: measurement
  icon: mdi:power-plug-outline

```

mletenay, 2021-08-24:
Thanks Peter, as a funny coincidence, similar improvement as yours was proposed today also by @ziporah
I'll include your input into sources and if you'd like, you can join us at our discord channel https://discord.gg/TaXyWXT where we try to improve the protocol and plugin in general ...

mletenay, 2021-08-29:
Meter values were added in v0.8.2, more new values will follow soon.

closed 2021-08-29 by mletenay