mletenay/home-assistant-goodwe-inverter #48: Is the BP/SBP series supported for Writing to at this stage - or just reading

craigcurtin-dev, 2021-08-29:
I have verified that the new code does indeed read from the SBP5000 inverters - great work thanks.

My next goal now is to write to them to control their operation.

Do we have any insight into how this is done - or is it back to wireshark and PV Master ?

Craig

mletenay, 2021-08-29:
There is very limited support for set_work_mode and set_ongrid_battery_dod, see
https://github.com/marcelblijleven/goodwe/blob/master/goodwe/es.py
But since the ES protocol is not properly understood, then sniffing the packets is the only option...

craigcurtin-dev, 2021-08-30:
OK i will look at that to get a feel and start digging in - do you want me to feedback my findings as i go into this issue ?

Did you ever disassemble the apk from the play store to see what the PV Master app is doing ? 

I might do that in conjunction with wireshark.

Craig

mletenay, 2021-08-30:
Yes, that was until now source of 90% of my knowledge ;-)

craigcurtin-dev, 2021-08-30:
OK i am just working through getting the APK environment setup now - will report back.

Craig

craigcurtin-dev, 2021-08-30:
OK disasemmbled the Java app and was a little overwhelmed with the sheer volume of files - spent 3 or 4 hours but did not make much progress.

So i turned back to wireshark and spent a fw hours on it and have cracked it

If you post the following to the inverter

aa55c07f02390b070d04000006000005ff7f0425

You can change the charge/discharge percentage (have not isolated yet - changing from charge to discharge mode

Usual protocol stuff - last two hex bytes are the checksum - simple addition of all the proceeding values

The key value is the 05ff7f0425 at the end of the string -that value controls the % of charge (relative to inverter capacity)

So if i set 5% on my 5KW inverters that will set the charge/discharge rate to 250w

Note the normal addressing convention at the start

aa55 = protocol definition
c0 = source addresss - make sure these are unique 
7f = destination address = Inverter address - by convention the GoodWe inverter address is 247 (Decimal) 

Still to decode the rest but time for bed now - will try and crack the charge/discharge change functionality tomorrow - i.e. flipping between the two

Craig

craigcurtin-dev, 2021-08-30:
OK more progress 

Sent packet aa55b07f02390b070d0400000500000fff7f041e

Breakdown as follows

aa55b07f02390b070d = seems to be static for this request type - assume there is some definition of modes in here
0400000500000fff7f041e

04 = Time group 
0000 = start of time group  hour/minute
0500 = end of time group hour/minute
00 = not sure yet
0f = rate % of inverter power
ff = charge/discharge ff = charge, 00 = discharge
7f = not sure yet

craigcurtin-dev, 2021-09-02:
OK some more packet captures today - trying to understand how the charge % is calculated

change charging between 6am and 5PM on slot 1 - every day
aa55b07f02390b070104060011000 00ff7f    0415 - as above but charge to 0%
aa55b07f02390b07010406001100ff ffff7f     0613 - as above but charge to 1%
aa55b07f02390b07010406001100ff fbff7f    060f - as above but charge to 5%
aa55b07f02390b07010406001100ff ecff7f   0600 - as above but charge to 20%
aa55b07f02390b07010406001100ff e2ff7f   05f6 - as above but charge to 30% 
aa55b07f02390b07010406001100ff d8ff7f   05ec - as above but charge to 40%
aa55b07f02390b07010406001100ff c4ff7f   05d8 - as above but charge to 60%

Last 4 digits are the checksum - only bit that is changing is the one bit i have highlighted

So it appears to be counting down from a nominal value of 255 when in charging mode
Will check the discharge mode but assume it would count up

00 = 0%
ff = 1%
fb = 5%
ec = 20% etc

Craig

mletenay, 2021-09-02:
It seem to be negative value of the percentage = 40% = -40 = 0xd8

craigcurtin-dev, 2021-09-02:
Hmm, Thats a strange way of looking at it - i guess they are treating charging as taking from the grid then and dishcarging as adding to the grid

I will do some more testing tomorrow and workit out

Craig

craigcurtin-dev, 2021-09-05:
Just a further update on this 

Battery_mode (as returned by the Inverter_test.py) mode 2 = discharging in eco mode, mode 3 = charging in eco mode - have not tested others yet.

mletenay, 2021-09-06:
I have extracted all the inverter communication code out of this HA plugin and moved it to standalone library here https://github.com/marcelblijleven/goodwe .
If any (udp protocol specific) problems or potential improvements are found/suggested, they should be reported over there from now on.

closed 2021-09-06 by mletenay