mletenay/home-assistant-goodwe-inverter #49: Goodwe 6000-DT,  E_total does not indicate the correct value.

supergroover, 2021-08-31:
In home assistant "sensor.goodwe_e_total" gives a value of 240,8 kWh but according to the semsportal it should be 13.348,10 kWh.
The value of "sensor.goodwe_e_day" seems good. There is no e_month sensor.
I cannot trace the value of e_total stated in Home assistant to something that is stated in the semsportal.

[image]

I've also installed the addon "goodwe Sems api to MQTT" and it gives me the following values:
```
Solar Goodwe produced 13.348,1 kWh
sems_battery_soc 200,0 %
sems_battery_soh -0,100000000000364 %
sems_grid_current 1,8 IAC
sems_grid_frequency 50,01 Hz
sems_grid_voltage 226,4 VAC
sems_inverter_status Online
sems_inverter_temperature 40,8 °C
sems_inverter_type GW6000-DT
sems_produced_this_month 704,6 kWh
sems_produced_today 4,1 kWh
sems_solar_power 1.213,0 W
sems_str_1_current 2,1 VDC
sems_str_1_voltage 322,8 VDC
sems_str_2_current 2,2 VDC
sems_str_2_voltage 327,2 VDC
```

How can i help you ?

mletenay, 2021-08-31:
These two sems/mqtt values do not sound very reliable :-D
sems_battery_soc 200,0 %
sems_battery_soh -0,100000000000364 %

I'll check the code, problem might be we read just 2 bytes instead of 4 in e_total.

supergroover, 2021-08-31:
> These two sems/mqtt values do not sound very reliable :-D
> sems_battery_soc 200,0 %
> sems_battery_soh -0,100000000000364 %

No, you are right, :-) I've no battery :-)

mletenay, 2021-08-31:
Indeed, if xx90 is included into e_total as upper two bytes, the e_total is 13349.0 kWh.
But is it realistic ? e_total/h_total = cca 1.8kW average permanent output ? 
Maybe yes, although on sample data from XS inverter e_total will get from 201kWh to 13 198 950kWh.
That's odd. Either the sample data are wrong or DT uses 4 bytes and XS only 2 ??

mletenay, 2021-08-31:
Actually it would help if you can provide the raw hex value of the inverter data response (from the debug output of the plugin, see troubleshooting in readme how to enable it). I'll then include it to unit tests to ensure future correctness of the code ...

supergroover, 2021-08-31:
You mean this code?
```
2021-08-31 12:03:04 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Connection made to address ('192.168.1.193', 8899)
2021-08-31 12:03:04 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2
2021-08-31 12:03:05 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Received: aa557f039215081f0c03020c88001f0ca90020ffffffffffffffffffffffffffffffffffffffffffff08d008f90906001b001a001b1386138613860000072b0001000000000000ffffffffffffffffffffffff0000ffff019dffffffff003c0002097e0000210300140000ffff0000ffff0000ffff0000ffffffffffffffffffff0000177c0beeffffffff00cf016302f00000000000649f03
```
e_total by Sems portal = 13350.30 kWh
e_day by Sems portal = 6,10 kWh
ppv = 1.955kW

mletenay, 2021-08-31:
Exactly, that's what will help.
I'll fix the code, but I need to make XS specific tweak back to 2 bytes, so it I'll do it later evening...

supergroover, 2021-08-31:
Glad I could help. Thank you.

mletenay, 2021-09-01:
DT's e_total was fixed in v0.8.5

closed 2021-09-01 by mletenay