mletenay/home-assistant-goodwe-inverter #50: 0xF7 (247) is the default Modbus ID of the inverter

mfaust78, 2021-08-31:
First of all, thank you for the great work!
I found a spec for the ET and might be able to add some registers to the code, but that another day.

When reading trough the documents from GoodWe, I noticed that the Modbus ID is set to 247 (0xF7) by default, but can be changed (for ET) with the PV Master app (Advanced Settings -> Communication Address). See https://community.goodwe.com/solution/Energy%20Storage%20System/Parallel-Storage-Solution for screenshots.

I think it would be good to make this ID / Communication Address configurable, such that if the ID/Communication Address was changed, the inverter still can be read (e.g. in case multiple inverters are combined).

mletenay, 2021-09-01:
Yes, I have realized that the f7 address can be in fact changed in the app just recently and I was already thinking about extracting it out as parameter and not hardcoded value. (even more when DT inverters use default value 7f and not f7).
I will adjust the code, thanks for the hint.
And I'm looking forward to new ET fields :-)
(Or could you somehow share that "found" ET spec ?)

mletenay, 2021-09-01:
comm_address parameter was extracted and can be configured now (v0.8.6)

closed 2021-09-01 by mletenay