mletenay/home-assistant-goodwe-inverter #52: Goodwe 6000DT, no connection after night sleep

supergroover, 2021-09-02:
In the evening when there is no more sun, my inverter goes into sleep mode.
The addon can no longer reach my inverter and spams my log.
In the morning when it gets light again and the inverter starts up again, the addon remains "offline". It no longer re-establishes the connection to the inverter.

Is this normal behavior or is something wrong here?

After a restart in the morning of the Home Assistant server it works again.

Onderstaand de log na een herstart in de avond.
```
2021-09-01 22:59:38 DEBUG (MainThread) [custom_components.goodwe.goodwe.goodwe] Connecting to DT family inverter at 192.168.1.193:8899
2021-09-01 22:59:38 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Connection made to address ('192.168.1.193', 8899)
2021-09-01 22:59:38 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f03753100280409
2021-09-01 22:59:40 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-01 22:59:40 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f03753100280409
2021-09-01 22:59:42 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-01 22:59:42 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f03753100280409
2021-09-01 22:59:44 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #4 of 3
2021-09-01 22:59:44 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f03753100280409
2021-09-01 22:59:44 WARNING (MainThread) [homeassistant.components.sensor] Platform rest not ready yet: Server disconnected without sending a response.; Retrying in background in 30 seconds
2021-09-01 22:59:46 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-09-01 22:59:46 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up goodwe platform for sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/goodwe/sensor.py", line 93, in async_setup_platform
inverter = await connect(
File "/config/custom_components/goodwe/goodwe/goodwe.py", line 41, in connect
await inverter.read_device_info()
File "/config/custom_components/goodwe/goodwe/dt.py", line 109, in read_device_info
response = await self._read_from_socket(self._READ_DEVICE_VERSION_INFO)
File "/config/custom_components/goodwe/goodwe/inverter.py", line 83, in _read_from_socket
return await command.execute(self.host, self.port, self.timeout, self.retries)
File "/config/custom_components/goodwe/goodwe/protocol.py", line 103, in execute
await on_response_received
custom_components.goodwe.goodwe.exceptions.MaxRetriesException
```

mletenay, 2021-09-02:
No, that should not be normal. UDP is connection-less protocol, so the temporary sleep should not cause any problems.
But in this case it seems your HA was restarted during inverter sleep and so the very first initialization/discovery request to inverter failed, since there was no (awake) inverter.
Let's observe it next night if this was just a coincidence.

supergroover, 2021-09-02:
```
Logger: custom_components.goodwe.sensor
Source: custom_components/goodwe/sensor.py:169
Integration: goodwe (documentation, issues)
First occurred: 20:56:00 (103 occurrences)
Last logged: 21:47:00

Could not retrieve data from inverter:
```

```
2021-09-02 21:43:52 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.1.193', 8899)
2021-09-02 21:43:54 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-02 21:43:54 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.1.193', 8899)
2021-09-02 21:43:56 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-02 21:43:56 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.1.193', 8899)
2021-09-02 21:43:58 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-02 21:43:58 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.1.193', 8899)
2021-09-02 21:44:00 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-09-02 21:44:00 WARNING (MainThread) [custom_components.goodwe.sensor] Could not retrieve data from inverter:
```

```
2021-09-02 21:48:22 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.1.193', 8899)
2021-09-02 21:48:24 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #1 of 3
2021-09-02 21:48:24 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.1.193', 8899)
2021-09-02 21:48:26 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #2 of 3
2021-09-02 21:48:26 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.1.193', 8899)
2021-09-02 21:48:28 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Retry #3 of 3
2021-09-02 21:48:28 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Sent: 7f0375940049d5c2 to ('192.168.1.193', 8899)
2021-09-02 21:48:30 DEBUG (MainThread) [custom_components.goodwe.goodwe.protocol] Max number of retries (3) reached, closing socket
2021-09-02 21:48:30 WARNING (MainThread) [custom_components.goodwe.sensor] Could not retrieve data from inverter:
```

supergroover, 2021-09-03:
Hi,

The addon boots up fine in the morning so that part is good.  Only my log is full of error messages during this offline period.

mletenay, 2021-09-03:
OK, so it works +/- as expected. just decrease log sensitivity fro the plugin and you should be fine.

closed 2021-09-03 by mletenay

mletenay, 2021-09-04:
Just to let you know, I've found a way how to improve the code so it will no longer spam the log file after inverter sleep.
Will get there in next release ...

supergroover, 2021-09-04:
Great, thank you.