mletenay/home-assistant-goodwe-inverter #53: Set Export Limit on ES

leonw1090, 2021-09-02:
Hi,
Firstly thank you for the amazing work.

I am having problems calling the new service to change the export limit.

This is what I am sending:

service: goodwe.set_grid_export_limit
data:
  entity_id: sensor.goodwe_inverter
  grid_export_limit: 1000

The error log is :

Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/goodwe/goodwe/inverter.py:146 
Integration: Home Assistant WebSocket API (documentation, issues) 
First occurred: September 1, 2021, 10:04:09 PM (6 occurrences) 
Last logged: 9:35:27 AM

[140182692479136] Error handling message: Unknown error
[140182799048656] Error handling message: Unknown error
[140182755634576] Error handling message: Unknown error
[140182692398752] Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 25, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 525, in handle_execute_script
    await script_obj.async_run(msg.get("variables"), context=context)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1219, in async_run
    await asyncio.shield(run.async_run())
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 353, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 371, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 571, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1491, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1526, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 665, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 658, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 856, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 695, in _handle_entity_call
    await result
  File "/config/custom_components/goodwe/sensor.py", line 204, in set_grid_export_limit
    await self.inverter.set_grid_export_limit(grid_export_limit)
  File "/config/custom_components/goodwe/goodwe/inverter.py", line 146, in set_grid_export_limit
    raise NotImplementedError()
NotImplementedError

I don not have a coding background but I will help in any way possible.

mletenay, 2021-09-02:
Unfortunately, the set_export_limit is currently working only on ET/EH inverters.
But @craigcurtin-dev is within issue #48 trying to sniff out the "write" commands, so when/if he succeeds, I'll add that to ES/EM/Bp too.

leonw1090, 2021-09-02:
Many thanks for the work already invested in the integration
I am happy to test with my ES if/when you need me

Regards

mletenay, 2021-09-02:
OK, I've just quickly hacked something.
Try to update to v0.8.6 and check if the set_export_limit works for your inverter now ...

leonw1090, 2021-09-02:
Hi,

Thank you for the time but no luck
New logs after the upgrade.

This error originated from a custom integration.

Logger: homeassistant.helpers.script.websocket_api_script
Source: custom_components/goodwe/goodwe/protocol.py:138 
Integration: goodwe (documentation, issues) 
First occurred: 2:21:53 PM (1 occurrences) 
Last logged: 2:21:53 PM

websocket_api script: Error executing script. Unexpected error for call_service at pos 1: non-hexadecimal number found in fromhex() arg at position 17
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 371, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 571, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1491, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1526, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 665, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 658, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 856, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 695, in _handle_entity_call
    await result
  File "/config/custom_components/goodwe/sensor.py", line 207, in set_grid_export_limit
    await self.inverter.set_grid_export_limit(grid_export_limit)
  File "/config/custom_components/goodwe/goodwe/es.py", line 153, in set_grid_export_limit
    Aa55ProtocolCommand("033502" + "{:02x}".format(export_limit), "03b5")
  File "/config/custom_components/goodwe/goodwe/protocol.py", line 138, in __init__
    + self._checksum(bytes.fromhex("AA55C07F" + payload)).hex()
ValueError: non-hexadecimal number found in fromhex() arg at position 17

And also 

This error originated from a custom integration.

Logger: homeassistant.helpers.script.websocket_api_script
Source: custom_components/goodwe/goodwe/protocol.py:138 
Integration: goodwe (documentation, issues) 
First occurred: 2:21:53 PM (1 occurrences) 
Last logged: 2:21:53 PM

websocket_api script: Error executing script. Unexpected error for call_service at pos 1: non-hexadecimal number found in fromhex() arg at position 17
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 371, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 571, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1491, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1526, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 665, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 658, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 856, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 695, in _handle_entity_call
    await result
  File "/config/custom_components/goodwe/sensor.py", line 207, in set_grid_export_limit
    await self.inverter.set_grid_export_limit(grid_export_limit)
  File "/config/custom_components/goodwe/goodwe/es.py", line 153, in set_grid_export_limit
    Aa55ProtocolCommand("033502" + "{:02x}".format(export_limit), "03b5")
  File "/config/custom_components/goodwe/goodwe/protocol.py", line 138, in __init__
    + self._checksum(bytes.fromhex("AA55C07F" + payload)).hex()
ValueError: non-hexadecimal number found in fromhex() arg at position 17

mletenay, 2021-09-02:
Hmm, it's not inverter command failure, it's the integration code.
I need to check it...

mletenay, 2021-09-03:
I believe I have fixed it in latest release (HA 2021.9.1 is required).

closed 2021-09-03 by mletenay

leonw1090, 2021-09-04:
Thank you, I can confirm the issue is resolved.

Looking forward for all the future updates 😊 Your work is absolutely appreciated.

Keep well.