mletenay/home-assistant-goodwe-inverter #57: Modbus EMS documentation

mfaust78, 2021-09-03:
Had to search for the source, but here the link to the Modbus EMS document I've found:
https://www.solaranzeige.de/phpBB3/download/file.php?id=3047&sid=616d92dcb44764a30592a5b44a05def1
You've got allmost all registers, but I think there are a few left.
It contains a table for the Diag Status as well as Warning / Error flags. Would be cool if that is added.

mletenay, 2021-09-04:
Well, this is great. Having it a year ago would save quite a lot of effort :-/
Could be helpful if you can review the exiting code vs. the spec.
It's always better to have second pair of eyes ...

mfaust78, 2021-09-04:
Found it only a few days ago.
I'll review the existing code vs. the spec over the next weeks. How should I report something if I notice a difference?

mletenay, 2021-09-04:
Just report it here.

mletenay, 2021-09-06:
I have extracted all the inverter communication code out of this HA plugin and moved it to standalone library here: https://github.com/marcelblijleven/goodwe .
If any problems or potential improvements are found, they should be reported over there now.

closed 2021-09-06 by mletenay