mletenay/home-assistant-goodwe-inverter #60: Today's PV Generation not matches Sems Portal

kolossboss, 2021-09-05:
Hi,
The sensor `sensor.goodwe_e_day` did not match the PC Energy Sensor from Sems Portal.

- This integration shows 9 kwh
- Sems integration 14,2 kwh (correct one)

A few days ago this was working correct.

Inverter: GW3648-EM

mletenay, 2021-09-05:
That's weird. I have checked the code there were no changes related to e_day sensor recently.
And it is also covered by unit tests for some time already.
Can you observe it for several days, I can't find a problem there.

kolossboss, 2021-09-05:
Yes very weird.
The connection is also very unreliable, but I dont think thats the reason of the gap.
There is also a small gab in "Total Generation" and the graph looks also different. (with this integration the graph still rises even though it's night ... it's confusing)

[image]
[image]
[image]
[image]

kolossboss, 2021-09-06:
The update to v0.8.9 fixes the unreliable connection issue. Thanks a lot!

But the differences between this integration and SEMS portal still exists.

mletenay, 2021-09-06:
Is the discrepancy between this and SEMS still there ?
Couldn't it be we or them just use wrong sensor name for the values ?
Isn't there some other sensor which +/- matches the SEMS values ?

kolossboss, 2021-09-07:
Yes the discrepancy is sill there.

Its possible that anything is wrong, but I'm not sure what. The graph looks also very weird....
There is also a discrepancy in the "Total Generation Sensor" (last two screens)

**From this Integration:**
[image]
[image]

**From "Goodwe Sems API" Integration:**
[image]
[image]

**From "Goodwe SEMS API to MQTT sensor" Integration:**
[image]
[image]

**Total Generation Sensor:**
[image]
[image]

GIJOE-SA, 2021-09-18:
Hi,
Not sure if I understand precisely what you mean with PC Energy Sensor but maybe this something to investigate and see if your graphs make more sense.
With the ES Inverter, sensor.goodwe_e_day combines PV consumption with battery consumption. In other words, SEMS Portal show only  PV generation where sensor.goodwe_e_day shows PV consumption + Battery consumption. For the ES inverter sensor.goodwe_ppv match up to what I see in the SEMS Portal.

kolossboss, 2021-09-20:
Hi,
its a typo, I mean PV Energy Sensor.
Ok, its possible that seensor.goodwe_eday combines in any way the battery. But I have a 7kwh Battery. I think that there must be a much larger delta between this 2 sensors. But its only 50kwh and 2800kwh in total.

Sensor.goodwe.ppv only shows W and nicht kwh. But the sensor is also working correct for me. I used the sensor in the new Energy Management portal.

Jayksopp, 2022-01-09:
I have the same invertor (GW3648-EM) and I see similar behavior. The sensor sensor.today_s_pv_generation does not match the actual pv generation. If I sum the PV power over time I get a value close to the SEMS portal one. This sensor.today_s_pv_generation is something different.

Achjon, 2022-01-31:
I have the same issue, is there some final solution to the problem ?

nazeg, 2022-02-12:
Is there any fix for this? I have the same problem. Today's Load is also wrong. Plugin shows: 6,7kwh, SEMS shows: 6,76(self-usage)+2,64(bought from grid)= 9,4kw
[image]

d-uhligde, 2023-03-12:
We have the same situation with Goodwe GW3048: there is a big difference between SEMS and Modbus values. - We haven't connected a battery.
Anybody has found a solution?
Thank you

orestesmas, 2023-03-17:
Same issue here. Inverter GW3648D-ES installed last week. The "sensor.total_pv_generation" and the "sensor.today_s_pv_generation" exported by the inverter through this integration show solar production at night, where the instantaneous power produced by the panels is obviously zero.

This is weird. Even if the reason of this behaviour is that those sensors show the combined solar+battery consumption, despite the sensors' names suggesting otherwise, the produced data is pretty useless, in my opinion.

The only quick fix I see is to define some helper (virtual) sensors that integrate the different instantaneous powers to calculate the energy, and some counters to add up these values daily/monthly/yearly

d-uhligde, 2023-03-17:
@orestesmas 
It can hardly be related to the battery consumption, since I don't have a battery connected at the moment.

GIJOE-SA, 2023-03-22:
Hi, I think this must be something very specific to your inverters. I have a GW5048D-ES and after more than 2 years my values are identical for the sensor and what I see on the mobile app. 
[image]

GIJOE-SA, 2023-03-22:
> Same issue here. Inverter GW3648D-ES installed last week. The "sensor.total_pv_generation" and the "sensor.today_s_pv_generation" values exported by the inverter through this integration show solar production **even at night**, where the instantaneous power produced by the panels is obviously zero.
> 
> This is weird. Even if the reason of this behaviour is that those sensors show the combined solar+battery consumption, despite the sensors' names suggesting otherwise, the produced data is pretty useless, in my opinion.
> 
> The only quick fix I see is to define some helper (virtual) sensors that integrate the different instantaneous powers to calculate the energies, and some counters to add up these values in a daily/monthly/yearly basis.

GIJOE-SA, 2023-03-22:
Hi, this maybe something related to your model inverter, on my GW5048D-ES both these sensors in Home Assistant match the mobile app data 100%. 
[image]

GIJOE-SA, 2023-03-22:
Did some digging, on the first screen there is "Generation Today" which seems higher than the value of "Generation Today" when you select the inverter itself. On this stage it seems that that inverter value "Today's PV Generation sensor" is lacking behind a couple of hours. Maybe the reason why I did not use "Today's PV Generation" sensor in my Energy dashboard. I have created the following in my config file which I use for daily and monthly (Energy dashboard) 

`# Sensor for Riemann sum of PV gen (W -> Wh)
- platform: integration
  source: sensor.pv_power
  name: pv_generation_sum
  round: 1
  unit_prefix: k
  method: left`

and

`utility_meter:
  pv_gen_daily:
    source: sensor.pv_generation_sum
    cycle: daily
  pv_gen_monthly:
    source: sensor.pv_generation_sum
    cycle: monthly`

Hope it helps.

midadami, 2023-04-21:
Are you guys running batteries?
Energy generation

> Same issue here. Inverter GW3648D-ES installed last week. The "sensor.total_pv_generation" and the "sensor.today_s_pv_generation" values exported by the inverter through this integration show solar [...]

Thats how the logic works for GoodWe with battery. Any solar that goes into the battery doesn't count. Any energy that comes out of battery is added to the generation counter. Battery is defined as a solar storage, there is no actual battery storage.
Its a valid implementation if looked on it own. If looked when added to a energy system its not correct. Only alternative is to measure the energy with the actual power instead of using energy generation given by GoodWe.

closed 2023-12-21 by mletenay