mletenay/home-assistant-goodwe-inverter #62: Decode error on ET inverter

Delorien84, 2021-09-07:
Hi, I have update to 0.8.9 version and the plugin stop working. In the log is exception that semee sensor value cannot be decoded. Exactly problem is at  [`return ", ".join(result)`](https://github.com/mletenay/home-assistant-goodwe-inverter/blob/8ad9928bef82f0f3b7f9402a51df219a1a06df1f/custom_components/goodwe/goodwe/sensor.py#L381). Where `result` contains non `string` value. Problematic sensor is [`battery_warning`](https://github.com/mletenay/home-assistant-goodwe-inverter/blob/8ad9928bef82f0f3b7f9402a51df219a1a06df1f/custom_components/goodwe/goodwe/et.py#L146).

I have GW8K-ET inverter with one battery pack contains three battery modules.

I will past exact stack trace later.

mletenay, 2021-09-07:
Ooops, that's latest addition - decoding of the warning values.
And something went wrong since I had no warnings to test ;-/
Could you please set the level to debug and post here the inverter response string ?

mletenay, 2021-09-07:
Ok, if the error was of `TypeError: sequence item 0: expected str instance, NoneType found` kind, I don't need the log, I've found the problem.

Delorien84, 2021-09-07:
Yes, I think that is the error. I am at work at the moment and not have access to HA at home.

mletenay, 2021-09-07:
I have already made a fix for that, but until I release next version, you have to go to HACS and reinstall the plugin selecting the previous working (v0.8.8) version. Sorry for that ...

Delorien84, 2021-09-07:
Thank you for such a quick response. I have downgraded already. Waiting for next release...Thank you for your work.

mletenay, 2021-09-12:
Fixed in v0.9.0

closed 2021-09-12 by mletenay