mletenay/home-assistant-goodwe-inverter #66: ES House consumption calculation incorrect

leonw1090, 2021-09-13:
Hi 

The house consumption calculation it seem to be doubled in the new release v0.9.0
When backup load is 500w the house consumption is 1000w

I have reverted to v0.8.9 and the values are now normal

mletenay, 2021-09-13:
!@#$%, you're right. I made copy/paste mistake ...

mletenay, 2021-09-13:
I need you to provide/verify the formula, without direct access to inverter I cannot check/verify it.
Current formula is ppv1 + ppv2 + pbattery + pback_up - pgrid

mfaust78, 2021-09-13:
From the figures I see, I don't think the pback_up is correct in there.
Why do you not use "ptotal" + "backup_ptotal"? that would seam fairly accurate for my Inverter.

leonw1090, 2021-09-13:
**0.9**
vpv1: 252.6 ipv1: 2.4 ppv1: 606 pv1_mode: 2 pv1_mode_label: PV panels connected, producing power vpv2: 252.5 ipv2: 2.5 ppv2: 631 pv2_mode: 2 pv2_mode_label: PV panels connected, producing power ppv: 1237 vbattery1: 51 battery_status: 80 battery_temperature: 23.2 ibattery1: 0.3 pbattery1: 15 battery_charge_limit: 0 battery_discharge_limit: 37 battery_error: 0 battery_soc: 100 warning: 100 battery_soh: 100 battery_mode: 2 battery_mode_label: Discharge battery_warning: 0 meter_status: 1 vgrid: 228.1 igrid: 5.5 pgrid: 9 fgrid: 49.89 grid_mode: 1 grid_mode_label: Inverter On vload: 228.1 iload: 5.6 pload: 105 fload: 49.89 load_mode: 1 load_mode_label: The inverter is connected to a load work_mode: 2 work_mode_label: Normal (On-Grid) temperature: 36.2 error_codes: 0 e_total: 4726.5 h_total: 7574 e_day: 2.6 e_load_day: 3.7 e_load_total: 4448.6 total_power: 1176 effective_work_mode: 1 effective_relay_control: 32 grid_in_out: 0 grid_in_out_label: Idle pback_up: 1237 plant_power: 1342 meter_power_factor: 0.001 xx85: 0 xx87: 182 diagnose_result: Discharge Driver On, Self-use load light, Export power limit set, Real power limit set e_total_exp: 538.5 e_total_imp: 21876227.2 vgrid_uo: 0 igrid_uo: 0 vgrid_wo: 0 igrid_wo: 0 e_bat_charge_total: 0 e_bat_discharge_total: 0 house_consumption: 2480 model: GW5048D-ES serial_number: xxxxxxxxx software_version: 410-04025-20 unit_of_measurement: W friendly_name: PV Inve

**0.8.9**
vpv1: 252.3 ipv1: 2.4 ppv1: 606 pv1_mode: 2 pv1_mode_label: PV panels connected, producing power vpv2: 252.3 ipv2: 2.5 ppv2: 631 pv2_mode: 2 pv2_mode_label: PV panels connected, producing power ppv: 1237 vbattery1: 51 battery_status: 80 battery_temperature: 23.2 ibattery1: 0.3 pbattery1: 15 battery_charge_limit: 0 battery_discharge_limit: 37 battery_error: 0 battery_soc: 100 warning: 100 battery_soh: 100 battery_mode: 2 battery_mode_label: Discharge battery_warning: 0 meter_status: 1 vgrid: 227.8 igrid: 5.4 pgrid: 7 fgrid: 49.8 grid_mode: 1 grid_mode_label: Inverter On vload: 227.8 iload: 5.5 pload: 103 fload: 49.8 load_mode: 1 load_mode_label: The inverter is connected to a load work_mode: 2 work_mode_label: Normal (On-Grid) temperature: 36.2 error_codes: 0 e_total: 4726.6 h_total: 7574 e_day: 2.7 e_load_day: 3.7 e_load_total: 4448.6 total_power: 1166 effective_work_mode: 1 effective_relay_control: 32 grid_in_out: 0 grid_in_out_label: Idle pback_up: 1230 plant_power: 1333 meter_power_factor: 0.01 xx85: 0 xx87: 166 diagnose_result: 83888192 e_total_exp: 538.5 e_total_imp: 21876253.4 vgrid_uo: 0 igrid_uo: 0 vgrid_wo: 0 igrid_wo: 0 e_bat_charge_total: 0 e_bat_discharge_total: 0 house_consumption: 1245 model: GW5048D-ES serial_number:xxxxxxxxx software_version: 410-04025-20 unit_of_measurement: W friendly_name: PV Inverter icon: mdi:solar-power

All i can see is that it might be calculating backup + Powerplant and that is inflating the vaule

mletenay, 2021-09-13:
@mfaust78 backup_ptotal is not present in ES inverters.
@leonw1090 So what is the formula you propose ?

leonw1090, 2021-09-13:
Honestly I do not know

The formula that was used in 0.8.9 was fine in my case in never noticed a discrepancy more than 30w

mletenay, 2021-09-13:
According to #37 previous formula was not counting backup load at all.

scrhall, 2021-09-13:
Maybe, I express myself wrong, English is not my native language. 
In the isdue that I opened, I indicated that the backup did not correspond to the house consumption.
Turning on certain types of devices were not reflected, in house consumption, but others were. 
I think the last calculation is correct and the problem is in the inverter backup sensor

mletenay, 2021-09-13:
From my point of view the current formula seems to be correct (if the assumptions on sensors are correct)
ppv1 + ppv2 + (signed value) pbattery is energy inverter is receiving.
From panels and from battery, if the battery is being charged the energy is reduced.
pback_up is energy consumed by backup side of the inverter, so it is consumed part.
pgrid should be the amount of energy at the CT meter, i.e energy leaving or coming into house.
And amount of energy produced minus (consumed on backup plus consumed on grid side between inverter and meter) should be total house consumption.
Is there a mistake in my reasoning ?
Or are the pback_up and pgrid sensors representing something different that we think ?

leonw1090, 2021-09-13:
Does it not make sense that house consumption only measures what you house needs right now in total.
So another way of looking at it is pload + pback_up = house_consumption ( in my opinion charging and discharging the battery does not count towards home consumption)

running these sensors give me a correct reading.

So if i am correct the problem in 0.9 is that plant power is calculated ( pload + pback_up) and it seems that the code is adding plant power + bakup + pload

That is why am seeing double the wattage than expected.

leonw1090, 2021-09-13:
@mletenay if you want to chat on Discord it might help. I can livestream my screen and show you what the problem is.
just let me know a time.

CEF-PHORAM, 2021-09-13:
Hi there. For those who don't mind using the Windows operating system, they can download this software from the site:
https://powerforum.co.za/files/file/60-goodwe_es-live-udp-data-by-cef-ver/
It is made only for ES series
Version 2.70 does not require an activation key and I have corroborated the Energy Balance Calculations with external instruments, both the Backup Output (essential loads) of the 5048-ES and the NON-Essential loads, simultaneously, on the same PC with the EASTRON 220 and 230 ModBus meters.
By moving the cursor over the labels you will see descriptive texts (tool tips)
I do not use the name PLANT POWER, which they use around here, you will see in the photograph that it is very easy to understand.
On the other hand, they will be able to observe the Energy Balance in real time every 1 second approximately.
Screens copy here.
[image]
[image]

mletenay, 2021-09-13:
@leonw1090 I see what's wrong with the numbers although I could not figure out what's wrong.
Anyway, I believe that what you suggest is easiest and most accurate formula:
house_consumption = pload + pback_up - pgrid.
Am i right ?

leonw1090, 2021-09-13:
@mletenay 
When I ran my test this morning I did not need to do anything with pgrid since pload was pure ongrid load and pbackup was backup load.

So from my testing 
house_consumption = pload + pback_up 
gave me the correct household consumption

I don't believe that pgrid is necessary in the equation sins export and import form the grid does not impact the house consumption as grid load is represented by pload.

We must also note that the goodwe is only accurate up to 50w so there will always be a small difference but along as pv master and the HA says the same thing we know it is working fine.

mletenay, 2021-09-13:
I believe you have to consider pgrid since you can consume more than you produce from PV and battery.
pload is only power inverter provides, if consume more that that, you import from grid and you have to include into house consumption.
Same applies if you have export enabled, you have to subtract the excess/export power you do not consume.
Right ?

leonw1090, 2021-09-13:
I follow your thought process. I did not think of that.

if we assume the way it was calculated in version 0.8.9 is correct ( it was before the update to 0.9.0 ) is there not a way we can find what sensor is adding the extra unwanted power. Because before the update I was 100% happy with the reading it was accurate to a certain margin for error. Then we do not need to change anything. the formula used in 0.8.9 was in my opinion perfect ( ran it for 2 weeks)

dsmal150, 2021-09-13:
Maybe only formula  ppv1 + ppv2 + pbattery - pgrid. Otherwise, if is present  pback_up is added once more. And final value is incorrect  larger by  pback_up. In addition, pback_up is higher than the real.

scrhall, 2021-09-13:
I think the inverter himself does not correctly calculate the pbackup
So I think it's better to go back to the previous calculation, even if it doesn't make sense regarding pbackup.

leonw1090, 2021-09-14:
Thank you.

update 0.9.1 is working and the values are correct

closed 2021-09-14 by leonw1090