mletenay/home-assistant-goodwe-inverter #69: Inverter exchange in HA

chemuki205, 2021-09-15:
Hello there

First of all congratulate you on this wonderful integration, I have 2 EM inverters and it allows me to have them controlled from HA.

I have detected some problems and data inconsistencies, I will open another question for inconsistencies, in this I wanted to record possible improvements for the integration.

1.- I have 2 instances of HA, I use one of them as a test bed in which I make modifications that, when I see correct, I transfer to the instance in production, and I have detected a small problem when identifying the inverters in HA .

In my case I have 2 EMs, both identified in _'sensor_name_prefix:'_ as _"Inversor 1 (88)"_ and _"Inversor 2 (99)"_ respectively, each one with its corresponding IP, however the inverters appear interchanged in each of the HA instances.

I know that is silly, but I have doubts if this could cause that in a restart of HA or integration the inverters would be exchanged, thereby causing the historical data to also be exchanged, I wonder if it would be possible to identify with _'name:'_ each of the inverters with their respective IP to avoid this.

I attach images of each HA instance.
[image]

2.- In the latest version (0.9.1) I have seen that the value of _'sensor.goodwe_diagnose_result:'_ has been modified so that it now shows a descriptive text of the diagnosis, it is perfect, but I wonder if a sensor could also be maintained _'sensor.goodwe_diagnose_code:'_ with the code provided by the inverter, it would be more practical to be able to use it in templates or triggers in HA.

(Message translated by google translate from Spanish)

mletenay, 2021-09-15:
Hi, I will expose the raw diagnose_result code value, good idea.
Regarding the inverter exchange, well, it's really weird since internally the inverter entities in HA are identified by the inverter's serial number. I'll try to look at the code, but at this moment, I don't know what's wrong and why it behaves this way.

chemuki205, 2021-09-16:
I have not explained myself correctly, the exchange of inverters has not occurred in the HA instance in production, in it they are identified in the same way as it was done in the first installation of the component, I meant that, in the second HA instance, the test one, with the same * .yaml configuration, are identified in reverse, so I was worried that they could be exchanged in a HA restart or reinstallation.

If as dice they are identified by the serial number, it should not be a problem since they cannot be exchanged once added to HA.

In my case it affects me for the reason that the yaml code that I make in the HA test instance does not serve me as it is for the other, since when they are exchanged I must modify it to integrate it into the production instance, for that reason I commented on the option to add the "_name_" parameter in the configuration so that HA can identify them as indicated and not randomly.

mletenay, 2021-09-16:
I think your problem may disappear anyway, since we are going to switch from yaml configuration to the UI flow configuration in next release.

mletenay, 2021-10-17:
Can you check if your problem is present also in latest version (with UI configuration) ?

chemuki205, 2021-10-20:
They work correctly, they were not inverted when you set them up.