mletenay/home-assistant-goodwe-inverter #70: Sensors values not adding up (Potentially sensor.goodwe_house_consumption)

GIJOE-SA, 2021-09-15:
Good day,
As per all the comments so far, a big thank you for all the work and effort to get this integration off the ground and working so well.

Since the last two or three version upgrades, I have noticed that the sensor values on my power distribution card are way out.  Although they never added up 100% previously, they were very close, and could be used to reliably automate my geyser, pool pump etc.

To explain this better, previously when adding sensor.goodwe_ppv, sensor.goodwe_pgrid and sensor.goodwe_pbattery1 together you would have had a value close to senor.goodwe_house_consumption (Current Consumption). 

After the last couple of versions this has changed to huge gaps for long periods of time.  As an example below, I am consuming 653 Wats more than possible.
 [image]
Another Example
[image]

Your assistance and advice will be greatly appreciated.

Regards

tinuva, 2021-09-16:
Out of interest sake, what card is that?

GIJOE-SA, 2021-09-16:
It is the Power Distribution Card, you can install it through HACS.

[https://github.com/JonahKr/power-distribution-card](url)

mletenay, 2021-09-16:
Release v0.9.0 did have wrong formula from house_consumption calculation, but it was fixed in v0.9.1.
Is the problem still present there ?

GIJOE-SA, 2021-09-16:
I think so, I am currently running 0.9.1

mletenay, 2021-09-16:
Hmm, prior to 0.9.0 the formula for house_consumption was `ppv1 + ppv2 + pbattery - pgrid`.
Since 0.9.1 I have changed it to `pload + pback_up - pgrid` since it was reported the previous formula was somehow ignoring consumption of backup line.
If you say it's still not correct, than one of these sensor do not provide correct values.
I'm not sure what to do, whether to revert to 0.8.x behavior, since here #66 it seemed 0.9.1 does provide +/- proper values.

GIJOE-SA, 2021-09-16:
If you are referring to the Backup Port VS On Grid port of the inverter I can definitely state with out a doubt that 0.8.x was not ignoring the backup line. Most of my house is on the backup line with only the big appliances such as stove and geyser on the On Grid line. 
However, let me read #66 and see if I am missing something, maybe revert to 0.8.x to confirm what I see. 
Thank you for your patience and assistance so far, will revert back with what I find as soon as possible,

tinuva, 2021-09-16:
I am on 0.8.9 still I think and can confirm this card works correctly.
[image]

mletenay, 2021-09-16:
I agree that 0.8.x looks the most reliable.
But there's at least one sensor which is reporting wrong data (or is improperly identified).
Since you obviously can't feed 885 to pback with only 760 and no import.

GIJOE-SA, 2021-09-16:
I have done a couple of more tests. 

**Test 1 - Only my router switched on**
In this test 0.8.x wins hands down
[image]

**Test 2 - Add +- 4KW to On Grid Leg only**
In this test 0.8.x still seems the best but 0.9.1 is very close to 0.8.x
[image]

**Test 3 - Add +-1.6KW to Backup Leg only**
All of them are very close however there is +- 200w more than I expected on all of them, could just be that my appliance pulls a bit more than specified.
[image]

Although it is almost impossible to compare the near real time data with the SEMS Portal is also seems more aligned with the 0.8.x calc.

GIJOE-SA, 2021-09-17:
It does not seem that  0.9.1 is using the calculations as I have done it above for sensor.goodwe_house_consumption (Represented by H-Con) as it almost doubles my consumption. The 6KW number is correct value, am I missing something? 
[image]

tinuva, 2021-09-17:
> I agree that 0.8.x looks the most reliable.
> But there's at least one sensor which is reporting wrong data (or is improperly identified).
> Since you obviously can't feed 885 to pback with only 760 coming from battery and no import.

So this is something else I noticed.

Using version 0.8.9 and earlier, the values always matched up, with the exception that sometimes it looked as if pback is actually updated out of sync and I think this is the inverter doing it. If I just accepted that pback is accurate and in sync most of the time then the values always aligned up.

I created a template sensor to see how much non-essential load I have and it uses `home consumption - pback` and sometimes the non-essentials side just shows a load that makes no sense at all. Eg. **all** non-essential items turned off on the db board, there should be 0w load on non-essentials, but with pback reading sometimes lower or higher than it should, when its lower then it looks like there is load on non-essentials when it is impossible.

Personally, my vote is also to go back to 0.8.9 version calculation for now until you think you can figure this out. Otherwise I think I will just be forced to stay on version 0.8.9 myself. Knowing, that pback doesn't read accurately.

tinuva, 2021-09-17:
Here is 2 consecutive readings showing how it jumps around.

[image]

[image]

mletenay, 2021-09-17:
I have reverted the code to the 0.8.x behavior, it should get to next release (over the weekend I hope).
From all the observations it really seems the pback is not reliable sensor and should be treated as such.

GIJOE-SA, 2021-09-17:
> I have reverted the code to the 0.8.x behavior, it should get to next release (over the weekend I hope).
> From all the observations it really seems the pback is not reliable sensor and should be treated as such.

Thank you very much for you effort and your patience. Much appreciated.

tinuva, 2021-09-17:
Also agree, many thanks for your time and trying to assist everyone with this addon!

mletenay, 2021-09-17:
You're welcome ;-)
The main problem is that those 3 families of inverters behave quite differently and I have only 1 inverter at home.
So anything I develop for the other two, I unfortunately have no way how to verify/test it directly.
And then it happens that attempt to improve something may actually make it worse ...

GIJOE-SA, 2021-09-17:
Happy to spin up a dev HA and test things for you on the ES if you want. Let me know.

mletenay, 2021-10-16:
Closing the issue, since v0.9.2 returned back to 0.8.x version of house consumption formulas which seem to be the best ones.

closed 2021-10-16 by mletenay