mletenay/home-assistant-goodwe-inverter #71: Error in total imported energy

chemuki205, 2021-09-16:
I think there is some error in the imported energy data, I do not know if it is calculated or it is a data that is supplied directly from the inverter, but of course the value is not correct, in my case I obtain the values ​​of the image, one of each inverter and not far from that amount has been imported

[image]

mletenay, 2021-09-16:
Which inverter ? ES/EM ?

chemuki205, 2021-09-16:
EM

GIJOE-SA, 2021-09-17:
> Which inverter ? ES/EM ?

Experiencing the same with the ES
[image]

mletenay, 2021-09-19:
Well, those sensor do not seem to provide values we were hoping for.
I have added them recently when I found one description of (RS-485 nod UDP) ES/EM communication protocol and it suggested that there should be such values as e_total at those offsets.
In theory it is possible we just decode them improperly (wrong encoding or offset), but more probably the inverter does not provide those values (via UDP) at all and we just decode something completely different.

TimelessNL, 2021-09-22:
Hi,

_First I want to say great work so far!
I was working on a [GoodWeLogger](https://github.com/jantenhove/GoodWeLogger) setup but then I came across this project.
Since I already had the WiFI dongle I figured this project based on UDP was worth a shot._

I've connected HA to my GW5K-DT and everything seems to work thus far.
Fortunately my `e_day` and `e_total` do work _kinda_, but they report -0.1kWh regularly:
[image]

So my question is, is this a known issue yet?
I've enabled `debug` logging, And during a -0.1kWh moment there are no errors logged.