mletenay/home-assistant-goodwe-inverter #74: Sting values energy dashboard solar production

thrawnarn, 2021-09-26:
I’m getting waaaay to high numbers in the energy dashboard. When looking at the sensor values all seems fine. Any idea what could be wrong? The only off thing I’ve seen is that e_total and e_day sometimes is at -0.1. (I’m running a DT-inverter)

[image]
[image]
[image]
[image]

TimelessNL, 2021-09-26:
I have the same issue but reported it on the python library ([#4](https://github.com/marcelblijleven/goodwe/issues/4)) since it is most likely the library at fault?
It seems that most total counters share this error.

The solar production is probably off since `e_total` is continuously going from +something to -0.1 causing a high delta between the values.

TimelessNL, 2021-09-28:
@thrawnarn thanks to @mletenay the issue is resolved in version [0.2.6](https://github.com/marcelblijleven/goodwe/releases/tag/v0.2.6) of the library because of [this](https://github.com/marcelblijleven/goodwe/commit/1f4f8fe4b2c1ab4a757483cff14bdbd2e410ca9a) commit

However I had to manually delete the energy statistics from the sqlite database.
This can be done like so:

1. Stop HA
2. Backup current database
`cp -a home-assistant_v2.db home-assistant_v2.db.backup`
3. Open the database:
`sqlite3 home-assistant_v2.db`
4. Find the appropriate `metadata_id` for the solar production:
`select id, statistic_id from statistics_meta where statistic_id like '%goodwe_e%';`

_example:_
```
1|sensor.goodwe_e_day
2|sensor.goodwe_e_total
```
5. Delete all entries that belong to these sensors:
`delete from statistics where metadata_id = 1 or metadata_id = 2;`
Replace _[1]_ and _[2]_ with your metadata_id's.
6. Start HA
7. If all went well, the sensors should be reset and you can delete the DB backup:
`rm home-assistant_v2.db.backup`

I've also deleted the _-0.1_ entries from `e_total` and `e_day` in the  `states` table, but I am not sure if this was necessary.
But it cleaned up the history cards so that's nice.

thrawnarn, 2021-10-02:
I’ve been running 0.2.6 for a few days now and it works flawlessly. Thanks!

closed 2021-10-02 by thrawnarn