mletenay/home-assistant-goodwe-inverter #79: MS Series is functional

niko123456000, 2021-10-14:
Just a minor documentation update (if you like).

The Goodwe MS Series (mine specifically is a GW8500-MS) (Australia) worked out of the box with the integration.

Logzy72, 2021-10-14:
I also experienced the MS series working out of the box, specifically GW-5000-MS in my instance.
This inverter has 3x MPPT trackers as opposed to most inverters that only have 2.
We only see the data from 2 using this intergration.
PV1 Voltage
PV1 Current
PV1 Power
PV2 Voltage
PV2 Current
PV2 Power

What would be nice is if we could also have these values for PV3.
The "PV Power" which is the total of all 3 PV's is correct.

I also dont see and consumption data but wonder if this is due to the Goodwe Homekit meter and the fact that it connects directly to wifi and the SEMS portal and does not interface with the meter through the RS485. 

I believe that if I was just using the Goodwe smart meter without the Home kit part then I would see consumption data as the meter would then be connected to the inverter via the RS485. 

I'm wondering if this integration could be copied and modified to connect to the goodwe homekit via ip address as this does with the inverter. 
We could then have data from the inverter and data from the goodwe homekit coming into HA and see the consumption. 
Aparantly the RS485 only gets connected on homekit when feeding into the grid needs to be controlled/limited as it has its own wifi connection to send consumption data to the SEMS portal.
As opposed to when just using the goodwe smart meter it connects to the inverter via the RS485 so the inverter then know the consumption data.

mletenay, 2021-10-14:
Good to hear that.
Can you check which protocol (one of ET/ES/DT) the inverter speaks ?
I assume it's the DT since those inverters are non-hybrid, but would be nice to have it confirmed.
The DT protocol in theory supports up to 6 PV strings, but only 2 were implemented. I will extend it to 3 so you can see all values.

Logzy72, 2021-10-14:
Here is the log as requested for the MS series.
Regards,
Leigh

2021-10-15 09:02:35 DEBUG (MainThread) [goodwe] Connecting to DT family
inverter at 192.168.1.242
2021-10-15 09:02:35 DEBUG (MainThread) [goodwe.protocol] Sent:
7f03753100280409 to ('192.168.1.242', 8899)
2021-10-15 09:02:35 DEBUG (MainThread) [goodwe.protocol] Received:
aa557f03500002ffffffff35353030304d535532313857303032304757353030302d4d53204f444d20436c69656e7420534e000000000000000000000000000000000000000000000c000c0010ffff010f000100009ac2
2021-10-15 09:02:35 DEBUG (MainThread) [goodwe] Connected to inverter
GW5000-MS, S/N:55000MSUxxxWxxxx
2021-10-15 09:02:35 DEBUG (MainThread) [goodwe.protocol] Sent:
7f0375940049d5c2 to ('192.168.1.242', 8899)
2021-10-15 09:02:36 DEBUG (MainThread) [goodwe.protocol] Received:
aa557f0392150a0f09030c0c7c000205c8000305980004ffffffffffffffffffffffffffffffffffff0961ffffffff0009ffffffff1386ffffffff000001270001000000000000ffffffffffffffffffffffffffffffff006bffffffff0004000000440000000700490000ffff0000ffff0000ffff0000ffffffffffffffffffff09500f63ffffffffffff01e1ffffffffffff0103002a4038
2021-10-15 09:02:36 DEBUG (MainThread) [custom_components.goodwe] Finished
fetching GoodWe data in 0.527 seconds (success: True)
2021-10-15 09:02:41 DEBUG (MainThread) [goodwe.protocol] Sent:
7f0375940049d5c2 to ('192.168.1.242', 8899)
2021-10-15 09:02:41 DEBUG (MainThread) [goodwe.protocol] Received:
aa557f0392150a0f0903120cb800020511000405570004ffffffffffffffffffffffffffffffffffff0961ffffffff0009ffffffff1386ffffffff000001250001000000000000ffffffffffffffffffffffffffffffff0065ffffffff0004000000440000000700490000ffff0000ffff0000ffff0000ffffffffffffffffffff09500f61ffffffffffff01e1ffffffffffff0103002a5b19
2021-10-15 09:02:41 DEBUG (MainThread) [custom_components.goodwe] Finished
fetching GoodWe data in 0.962 seconds (success: True)
2021-10-15 09:02:46 DEBUG (MainThread) [goodwe.protocol] Sent:
7f0375940049d5c2 to ('192.168.1.242', 8899)
2021-10-15 09:02:46 DEBUG (MainThread) [goodwe.protocol] Received:
aa557f0392150a0f0903160cf40002058b000305580004ffffffffffffffffffffffffffffffffffff0962ffffffff0009ffffffff1386ffffffff000001250001000000000000ffffffffffffffffffffffffffffffff0060ffffffff0004000000440000000700490000ffff0000ffff0000ffff0000ffffffffffffffffffff09500f68ffffffffffff01e1ffffffffffff0103002afb4d
2021-10-15 09:02:46 DEBUG (MainThread) [custom_components.goodwe] Finished
fetching GoodWe data in 0.228 seconds (success: True)
2021-10-15 09:02:51 DEBUG (MainThread) [goodwe.protocol] Sent:
7f0375940049d5c2 to ('192.168.1.242', 8899)
2021-10-15 09:02:51 DEBUG (MainThread) [goodwe.protocol] Received:
aa557f0392150a0f09031c0cf4000305890004060c0003ffffffffffffffffffffffffffffffffffff0962ffffffff0009ffffffff1388ffffffff000001250001000000000000ffffffffffffffffffffffffffffffff0067ffffffff0004000000440000000700490000ffff0000ffff0000ffff0000ffffffffffffffffffff09500f6bffffffffffff01e1ffffffffffff0103002a5173
2021-10-15 09:02:51 DEBUG (MainThread) [custom_components.goodwe] Finished
fetching GoodWe data in 0.204 seconds (success: True)
2021-10-15 09:02:56 DEBUG (MainThread) [goodwe.protocol] Sent:
7f0375940049d5c2 to ('192.168.1.242', 8899)
2021-10-15 09:02:56 DEBUG (MainThread) [goodwe.protocol] Received:
aa557f0392150a0f0903210da00002054b000405940003ffffffffffffffffffffffffffffffffffff095effffffff0009ffffffff1385ffffffff000001260001000000000000ffffffffffffffffffffffffffffffff0069ffffffff0004000000440000000700490000ffff0000ffff0000ffff0000ffffffffffffffffffff09500f60ffffffffffff01e1ffffffffffff0103002a55d5
2021-10-15 09:02:56 DEBUG (MainThread) [custom_components.goodwe] Finished
fetching GoodWe data in 0.199 seconds (success: True)
2021-10-15 09:03:01 DEBUG (MainThread) [goodwe.protocol] Sent:
7f0375940049d5c2 to ('192.168.1.242', 8899)
2021-10-15 09:03:01 DEBUG (MainThread) [goodwe.protocol] Received:
aa557f0392150a0f0903260d71000205a5000405410004ffffffffffffffffffffffffffffffffffff095bffffffff0009ffffffff1384ffffffff000001270001000000000000ffffffffffffffffffffffffffffffff0066ffffffff0004000000440000000700490000ffff0000ffff0000ffff0000ffffffffffffffffffff09500f65ffffffffffff01e1ffffffffffff0103002a0ad1
2021-10-15 09:03:01 DEBUG (MainThread) [custom_components.goodwe] Finished
fetching GoodWe data in 0.200 seconds (success: True)
2021-10-15 09:03:06 DEBUG (MainThread) [goodwe.protocol] Sent:
7f0375940049d5c2 to ('192.168.1.242', 8899)
2021-10-15 09:03:06 DEBUG (MainThread) [goodwe.protocol] Received:
aa557f0392150a0f09032b0d44000205c9000405560004ffffffffffffffffffffffffffffffffffff0962ffffffff0009ffffffff1385ffffffff000001280001000000000000ffffffffffffffffffffffffffffffff0067ffffffff0004000000440000000700490000ffff0000ffff0000ffff0000ffffffffffffffffffff09500f6bffffffffffff01e1ffffffffffff0103002ae979
2021-10-15 09:03:06 DEBUG (MainThread) [custom_components.goodwe] Finished
fetching GoodWe data in 0.202 seconds (success: True)
2021-10-15 09:03:11 DEBUG (MainThread) [goodwe.protocol] Sent:
7f0375940049d5c2 to ('192.168.1.242', 8899)
2021-10-15 09:03:11 DEBUG (MainThread) [goodwe.protocol] Received:
aa557f0392150a0f0903300d580002058b0004055a0004ffffffffffffffffffffffffffffffffffff095cffffffff0009ffffffff1383ffffffff000001290001000000000000ffffffffffffffffffffffffffffffff0069ffffffff0004000000440000000700490000ffff0000ffff0000ffff0000ffffffffffffffffffff09500f64ffffffffffff01e1ffffffffffff0103002a8b0a
2021-10-15 09:03:11 DEBUG (MainThread) [custom_components.goodwe] Finished
fetching GoodWe data in 0.209 seconds (success: True)
[... cut ...]
2021-10-15 09:05:11 DEBUG (MainThread) [custom_components.goodwe] Finished
fetching GoodWe data in 0.210 seconds (success: True)
2021-10-15 09:05:15 ERROR (MainThread)
[homeassistant.components.hassio.handler] Client error on
/homeassistant/restart request Server disconnected

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

mletenay, 2021-10-17:
Proper support for MS inverter was added in v0.9.3.
There are now sensors also for the 3rd PV string and the plugin will automatically hide L2/L3 sensors since MS are single phase inverters.
Enjoy.

closed 2021-10-17 by mletenay