mletenay/home-assistant-goodwe-inverter #87: Home Assistant and Eenergy dashboard

JoLambinon, 2021-11-17:
### Discussed in https://github.com/mletenay/home-assistant-goodwe-inverter/discussions/55

<div type='discussions-op-text'>

<sup>Originally posted by **sotoniak** September  3, 2021</sup>
Hi, before I write my question.  This is great project. Respect! I was very surprised how many sensors GoodWe GW10K-ET has. Honestly, I don't understand most values :-)

And now my question. I fight with setting in Energy Home Assistant Dashboard. I have Inverter, battery and I don't know which values can I set. In select box with sensors many sensors miss. Please can anybody help me? 

https://www.home-assistant.io/blog/2021/08/04/home-energy-management/</div>

JoLambinon, 2021-11-17:
Hi, may I also just stay that this is a fantastic project !

Can someone please explain or point me to the right place,  I have an ES inverter and the integration does not provide sensors for batt charge and discharge for use in the HA Energy dashboard.  Can these values be made available ?

GIJOE-SA, 2021-11-21:
I have used the following in my configuration.yaml. Remove ' from code

'- platform: template
  sensors:
    # Template sensor for values - energy from batteries (active_power > 0)
    battery_discharge:
      friendly_name: "Power From Battery"
      unit_of_measurement: 'W'
      value_template: >-
        {% if states('sensor.goodwe_pbattery1')|float > 0 %}
          {{ states('sensor.goodwe_pbattery1')|float }}
        {% else %}
          {{ 0 }}
        {% endif %}
    # Template sensor for values - Charge of batteries (active_power > 0)
    battery_charge:
      friendly_name: "Charge Bateries"
      unit_of_measurement: 'W'
      value_template: >-
        {% if states('sensor.goodwe_pbattery1')|float < 0 %}
          {{ states('sensor.goodwe_pbattery1')|float * -1 }}
        {% else %}
          {{ 0 }}
        {% endif %}'
'# Sensor for Riemann sum of battery discharge (W -> Wh)
'- platform: integration
  source: sensor.battery_discharge
  name: battery_discharge_sum
  unit_prefix: k
  round: 1
  method: left"
'# Sensor for Riemann sum of battery charge (W -> Wh)
'- platform: integration
  source: sensor.battery_charge
  name: battery_charge_sum
  unit_prefix: k
  round: 1
  method: left
utility_meter:
  battery_discharge_daily:
    source: sensor.battery_discharge_sum
    cycle: daily
  battery_discharge_monthly:
    source: sensor.battery_discharge_sum
    cycle: monthly
  battery_charge_daily:
    source: sensor.battery_charge_sum
    cycle: daily
  battery_charge_monthly:
    source: sensor.battery_charge_sum
    cycle: monthly"

JoLambinon, 2021-11-22:
Thank you again.

I am completely out of my depth, but I will give it a try.

Sincerely

Jo

Jo Lambinon
083 609 1286
***@***.******@***.***>

closed 2021-12-22 by mletenay