mletenay/home-assistant-goodwe-inverter #88: Works a few seconds and then becomes Unavailable

thrawnarn, 2021-11-23:
I just got an update of my ARM of the support. Now it only works for a few seconds then it becomes Unavailable. I'm guessing it only only works when it's getting new data, and then it sets it as Unavailable.
My ARM version is: v13
They also upgraded my DSP-board.

This is when it works:
[image]

a few seconds later:
[image]

This is the history of the inverter entity:
[image]

mletenay, 2021-12-12:
This is very likely caused by the unreliability of the network, i.e. not all requests were responded or responded in time.
Try to set the integration log level to debug and see if there are failed requests.
Anyway, there's not much shat can be done on the integration page, only to improve your network stability.

closed 2021-12-12 by mletenay

TPochobradsky, 2022-01-11:
I have a similar problem. What HW components do you use? I have HA on Synology VMM and Microtic router.

kolossboss, 2022-01-28:
I have the same problem for month now. I‘m not sure whats the problem.
I forced my installer to replace the wifi modul, with the wired Ethernet modul. But same problem.
I have a very reliable unifi system with zero other issues.
I played with the Integration settings like scan interval and Network timeout. Its better, but still sometimes not available.
[image]

thrawnarn, 2022-01-28:
Yeah, I still have the same issue. My network is superstable with unifi products. And I am using ethernet all the way. I used the wireless module before I used the Lan module and I had zero problems with that, so its not a network problem.

kolossboss, 2022-01-28:
Maybe its a issue with the unifi devices?

TPochobradsky, 2022-01-28:
I have tried to connect HA with Goodwe through another router and cables without significant changes. :-( 
There are two potential problematics points. The wired Ethernet modul or Instalation HA in VMM on Synology NAS.
Any idea what to tri change?

kolossboss, 2022-01-28:
I tried the wireless and wired modul. No changes. 
My HA running with HassOS on an Intel NUC and its connected to the same Unifi Switch

TPochobradsky, 2022-01-29:
My HA running on Synology DS218+ (DSM 7.0.1) -> VMM 2.5.5 -> Home Assistant OS 7.2.
Switch is Mikrotik hAP ac^2.

kolossboss, 2022-02-04:
I tried the official integration, but its much much worse...
you cant change things like scan interval and network timeout.
Its automatically polling every 10 second. 
But changing this to 30s does not work for me.
The automation in the documentation says, that I need the update `sensor.ppv` But `sensor.ppv` it not availble and `sensor.goodwe_ppv` also did nothing when updating via automation.

thrawnarn, 2022-03-24:
Ok, I've been looking more into this, because I know that my network isn't bad. And now I think I know whats happend.

Just my backstory again; I had wifi and the integration work flawlessly, but my sems portal didn't work. Because of this I got a LAN module from my supplier. But the LAN module needed a new ARM-firmware to work with the integration. I emailed goodwe who installed the new ARM-firmware, but they also updated my DSP version from v9 to v12. And after this the integration isn't working very good.

My theory is that the DSP-version is answering UDP-calls differently. I think it only answers when something has changed. I wrote a supersimple software to check this manually. It sends the udp request packet with a timeout of 100ms and then retries after 1 second. I'll attach the log for it here. As seen it only sends bytes only
[goodwe.log](https://github.com/mletenay/home-assistant-goodwe-inverter/files/8340579/goodwe.log)
 to the changed value, and when it responds varies heavily.

The log is pretty huge, it's almost 24 minutes (1000 iterations), but I wanted to run it for a longer timespan to show how consistent it is.