mletenay/home-assistant-goodwe-inverter #89: No data to SEMS portal when using this integration

vjversatile, 2021-11-30:
Hello, I have noticed that the data logger does not send data to the SEMS portal when using this local integration. The integration itself works well and pull data consistently into home assistant but renders the Sems portal app useless. If i disable this integration, SEMS portal starts updating after a few minutes. Once i enable back, SEMS portal works for some time but shows offline after.

Is this expected behaviour?

mletenay, 2021-12-12:
No, this is not expected behavior and the integration and SEMS portal should co-exist without any issues.
But it may not always be the case, there were rare reports users having problem with SEMS like you.
Reducing the HA integration polling frequency (e.g. to 30s) seemed to help for some users.
The cause of this is unknown, in theory it may have something to do with local network setup/performance

closed 2021-12-12 by mletenay

vjversatile, 2021-12-14:
Thanks, I did try reducing polling frequency a bit previously but thought it wouldn't sort the problem. But after your message i tried frequencies over a minute and found that SEMS starts getting data if i set the frequency of this integration to around 3 mins. Thank you for pointing out. 3 mins freq should be ok in general for me.

lyricnz, 2022-11-17:
For me, this is a vital update to the HA-core integration. With the default (10 second) polling rate, my upload to semsportal is completely broken, which makes it hard to get support when I am chasing other firmware issues with them.

mletenay, 2022-11-18:
Unfortunately, there is no known explanation why it happens in some environments, while it properly works in most others.
Once there is explanation/solution I'm eager to adjust the code, but at the moment there is nothing to do.