mletenay/home-assistant-goodwe-inverter #90: battery percentage?

bp-ouhaha, 2021-12-01:
I'm using this integration in Home Assistant with great pleasure, for a GoodWe ET invertor with a BYD HVS battery.

It's working very well, also in the new Energy dashboard.

I'm missing only 1 thing: **the battery percentage**!

Thanks in advance, and if I can provide data/help, let me know, I have programming skills.

mfaust78, 2021-12-01:
On the default Dashboard the Battery percentage does not show up in my Home Assisant, but it is present.
Either in in History, a edited Dashboard or the Developer Tools you can look for "sensor.goodwe_battery_soc" or simply search for SOC (State of Charge).

bp-ouhaha, 2021-12-01:
you are totally right!

I missed it somehow.

Thanks :)

closed 2021-12-01 by bp-ouhaha