mletenay/home-assistant-goodwe-inverter #93: Data output when inverter is waiting or offline

djoenez, 2021-12-08:
Hello, Thanks alot for this code! It works perfect.

I had a question, now when my inverter does not have enough V/A it will just go to waiting mode, but the Goodwe_PPV will not output any number anymore when it is polled. Could this be catched by the code? (maybe a if time before/after then output 0 if system is offline?) or output zeros when no data can be pulled?

djoenez, 2021-12-09:
I narrowed it down some more, is it possible to (in sensor.py) add an if statement that if inverter is not in "working" mode, PPV actual will be set to 0?

mletenay, 2021-12-12:
I'm not sure I understand what you wish to achieve.
I assume the sensors report themselves as Unavailable when your inverter goes to sleep mode (and does not respond to udp packets anymore).

djoenez, 2021-12-13:
Thanks for responding. 
The issue i'm trying to solve is that my graphs mess up when my inverters become unavailable. I could use the fill zero option (in grafana). But then also when i do not get any data from my inverters, i get large dips (Which are not actually there).

So What i would like to have is an option that if the inverters is not in working mode that the script itself just pushes a 0 value. so that during the night there is 0 output, instead of "null" output.

tuomaz, 2021-12-17:
I have the same request! The possibility to report a zero value will give less headache and prettier looking graphs.

darek-margas, 2022-01-01:
For integration it would be nonsense as it really is unavailable. Just make sensor in configuration.yaml and make this sensor read value and in 'unavailable' present zero or whatever you need.

djoenez, 2022-01-05:
Hello Darek-Margas,

I do see that the implementation would be nonsense from a technical perspective. From a functional perspective atleast i see the use. But:

You describe a code piece in configuration.yaml that would read the sensor value and have an if statement that if it is unavailable  the data should be 0. Could you help me with such a code piece?

darek-margas, 2022-01-05:
> You describe a code piece in configuration.yaml that would read the sensor value and have an if statement that if it is unavailable the data should be 0. Could you help me with such a code piece?

There is number of ways. One example from my config:

`      power_consumption:
        value_template: "{% if is_state_attr('sensor.pvoutput', 'power_consumption', 'NaN') %}0{% else %}{{ state_attr('sensor.pvoutput', 'power_consumption') }}{% endif %}"
        friendly_name: "Using"
        unit_of_measurement: "W"
        device_class: power
`
When you replace NaN by anuthing you get and make matching sensors it will do.
Another method which I used to do exactly opposite is like that:

`      pv_temperature:
        unit_of_measurement: '°C'
        friendly_name: "Inverter temperature"
        device_class: temperature
        value_template: >-
          {% set mytemp =  state_attr('sensor.inverter_goodwe_miranda', 'tempperature') | float(0) %}
          {%- if mytemp  > 0 -%}
          {{ mytemp }}
          {%- else -%}
          unavailable
          {%- endif -%}
`
just write it in opposite way - else 0 and matching sensors.

Or if you want to keep range and outside make 0:

`      pv_voltage:
        unit_of_measurement: 'V'
        friendly_name: "Inverter AC Voltage"
        device_class: voltage
        value_template: >-
          {% set myvac1 =  state_attr('sensor.inverter_goodwe_miranda', 'vac1') | float(0) %}
          {%- if myvac1  >= 180 and myvac1 <= 280 -%}
          {{ myvac1 }}
          {%- else -%}
          unavailable
          {%- endif -%}
`

And so on...

djoenez, 2022-01-09:
Hello Darek Margas,

Thanks for replying. I'll implement it once I have some time.

tuomaz, 2022-01-20:
If there is a way to make the sensor "Total PV Generation" keep its value even if the inverter goes offline it would be nice. That value is actually not nosense, total pv generation is still the same even if the sun is currently not shining. I guess many people want to show that value in a dashboard (myself included). I haven't looked into HA component development and maybe it is not possible to fix this in an easy way. I can see some potential pitfalls if for example the HA instance is restarted when the inverter is offline...

mletenay, 2022-02-18:
https://github.com/home-assistant/core/issues/66596

closed 2023-02-22 by mletenay