mletenay/home-assistant-goodwe-inverter #94: Invertor not listening to UDP port 8899

jonisnet, 2021-12-09:
Last week i let Goodwe upgrade my goodwe GW3600D-NS invertor to version Firmware: 19.19.13.

The integration did succesfully connect to the convertor but there are no readings.
After some googling i found out that the UDP port that the invertor is listening to is 48899 and that 8899 isnt open.

How can i change the listening port for my integration.

Kees de Bruin

mletenay, 2021-12-09:
You can't.
And the port 48899 is actually not the port of the inverter, but only of the web frontend, usually just used with broadcast requests to identify inverters on local network.
All the Goodwe inverter work only on port 8899, it's hard-coded even in their mobile applications.

closed 2021-12-09 by mletenay

jonisnet, 2021-12-09:
Okay than i don't understand why the invertor doesnt react to any udp message i send to it on port 8899.
Maybe you can help me sort it out?

jonisnet, 2021-12-09:
Btw i am not using the wifi but Ethernet Cable only

jonisnet, 2021-12-09:
Nevermind, its working now

Thanx for the quick reply.

JannemanDev, 2024-07-12:
If you want to locate your GoodWe inverter using Linux terminal:
```console
sudo apt install socat
echo -n "WIFIKIT-214028-READ" | socat - UDP-DATAGRAM:x.y.z.255:48899,broadcast
```

where you need to change `x.y.z.255` to your network for example `192.168.10.255`

It will respond with for example:
`192.168.10.118,xxxxxxxxxxxx,Solar-WiFiXXXXX`

which are `IP,MAC,SSID` (where `SSID` includes the device serial prefixed with `Solar-WiFi`)