mletenay/home-assistant-goodwe-inverter #95: New MS Series inverter not working (no open ports)

aksvenk, 2021-12-22:
Hello,

I had a solar PV system installed a couple of weeks ago and I can't seem to connect to it no matter what.

It looks like port 8899 is closed and the `inverter_test.py` understandably gives a `[Errno 61] Connection refused`.

Is there anything I am missing or am I just out of luck :(

```
> sudo nmap -sU -p 8899 192.168.107.118 
Starting Nmap 7.91 ( https://nmap.org ) at 2021-12-22 12:24 AEDT
Nmap scan report for 192.168.107.118
Host is up (0.10s latency).

PORT     STATE  SERVICE
8899/udp closed ospf-lite

Nmap done: 1 IP address (1 host up) scanned in 0.24 seconds
```

Model: GW8500-MS
FW Version: v1.13.17

aksvenk, 2021-12-22:
I see that @niko123456000 has had some luck in [this thread ](https://github.com/mletenay/home-assistant-goodwe-inverter/issues/79#issue-1025821844) with the same inverter. Was this on a different firmware version?

niko123456000, 2021-12-22:
>$ sudo nmap -sU -p 8899 192.168.68.130
Starting Nmap 7.80 ( https://nmap.org ) at 2021-12-22 13:06 AEDT
Nmap scan report for 192-168-68-130.tpgi.com.au (192.168.68.130)
Host is up (0.089s latency).

PORT     STATE         SERVICE
8899/udp open|filtered ospf-lite

Nmap done: 1 IP address (1 host up) scanned in 3.39 seconds

Not sure what my FW is or how to find that out.

aksvenk, 2021-12-22:
False alarm. This was a firewall rules problem on my end. It's all sorted now.

closed 2021-12-22 by aksvenk