mletenay/home-assistant-goodwe-inverter #103: read datetime error on GW3600D-NS

starkillerOG, 2022-01-07:
I get the following error on a Goodwe GW3600D-NS inverter:

```
Error while setting up goodwe platform for button

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/goodwe/button.py", line 31, in async_setup_entry
    await inverter.read_setting("time")
  File "/usr/local/lib/python3.9/site-packages/goodwe/dt.py", line 166, in read_setting
    return setting.read_value(buffer)
  File "/usr/local/lib/python3.9/site-packages/goodwe/sensor.py", line 177, in read_value
    return read_datetime(data)
  File "/usr/local/lib/python3.9/site-packages/goodwe/sensor.py", line 363, in read_datetime
    return datetime(year=year, month=month, day=day, hour=hour, minute=minute, second=second)
ValueError: month must be in 1..12
```

starkillerOG, 2022-01-08:
I did some debugging:
The `raw_data` of the `read_setting` function on line 159 of `dt.py` gives back the following raw data from the time setting:
`b'\xaaU\x7f\x03\x06\xff\xff\xff\xff\xff\xff\tX'`

I got this by using the following python script: 
```
import asyncio
import goodwe

async def test_goodwe():
    ip_address = '192.168.1.IP'

    inverter = await goodwe.connect(ip_address)
    await inverter.read_setting("time")

asyncio.run(test_goodwe())
```

inside the `sensor.py` file in the `read_datetime` function this translates into year=2255, month=255, day=255,hour=255,minute=255,second=255

So this does not look correct.

@mletenay where did you get the modebus address 40313 as listed here:
`Timestamp("time", 40313, "Inverter time"),`

mletenay, 2022-01-09:
It seems several DT/D-NS registers are not there where I expected them :-(
I quickly changed the exception handler, please update to master version until I release 0.9.8
(there were some other weird reports that need to be investigated)

starkillerOG, 2022-01-10:
@mletenay thanks for improving the exception handler!
Where did you get the register adresses from?
Do you have some other manual?

On internet I found these two, but those seem to be using much lower values for the register adresses.
https://github.com/MiG-41/Modbus-GoodWe-DT
https://downloads.vodnici.net/uploads/wpforo/attachments/483/4349-1GoodweModbusPROTOCOLHybrid-ENV1-3EMseries.pdf

mletenay, 2022-01-10:
No, unfortunately there are no public documents available (or at least accessible to me).
Those offset were deducted from the SolarGo application, but either they are not applicable to all inverters speaking the DT family protocol, or they are completely wrong :-(
See discussion here -https://github.com/mletenay/home-assistant-goodwe-inverter/discussions/100#discussioncomment-1902625

mletenay, 2022-01-10:
Time settings error on D-NS inverters was fixed in 0.9.8

closed 2022-01-10 by mletenay