mletenay/home-assistant-goodwe-inverter #104: Version 0.9.7

danygi79, 2022-01-09:
After usate at versioni 0.9.7 integration stop completly to work. I downgraded to 0.9.6 and work correcly

ersiensi, 2022-01-09:
Same here. I cant see Goodwe in Settings - Integrations.
Thx for great job.

maurizioandreotti60, 2022-01-09:
> After usate at versioni 0.9.7 integration stop completly to work. I downgraded to 0.9.6 and work correcly

I have the same issue. Trying to downgrade to prevous version.

maurizioandreotti60, 2022-01-09:
I have the same issue. Trying to downgrade to trhe previous version,

maurizioandreotti60, 2022-01-09:
The new version is not the culprit of the issue, it is the installation from Hacs that caused the issue.
After the upgrade of the new version I noticed this folder tree:
custom_components\godwee\goodwe\
This is not correct because all the files must be in goodwe folder under custom_component, that is custom_components\godwee\
I moved all the files to custom_components\godwee\ folder and the integration was correctly loaded.
So the issue is not version 0.9.7 but the way Hacs install the integration.

mletenay, 2022-01-09:
This is weird, it works without any issues, here, also other people reported upgrade to 0.9.7 without any issues ?
What inverter ? What HA version ? Are there any errors in the log files ?

Jayksopp, 2022-01-09:
I had a similar problem w 0.9.7. Then downgraded to 0.9.6 
which worked. Then installed 0.9.7 again and it also worked. Yesterday when installing node red Hacs also created two nested folders: custom components/node red/node red

janchlebek, 2022-01-09:
HACS 1.19.0 and then 1.19.1 and 1.19.2 came out pretty much at the same time - might be related?

mletenay, 2022-01-09:
I would suspect some HACS issue, since there were not changes to integration between 0.9.6 and 0.9.7, only the underlying goodwe library was updated.

kolossboss, 2022-01-10:
Its a HACS issue. I have same problem with a different integration.
There is no need to install 0.9.6.
Just reinstall 0.9.7. with HACS and restart HA