mletenay/home-assistant-goodwe-inverter #111: Incorrect grid export limit unit and range on MS Series

aksvenk, 2022-02-10:
Hello,

I have a Goodwe GW8500-MS. The grid export limit on this unit is set in percentage(%) of the rated power.
eg. Setting the export limit to 60% will cap the export at 5.1kW (60% of max AC power rating of 8.5kW).

I have tested the slider of the export limit on [v0.9.8.1](https://github.com/mletenay/home-assistant-goodwe-inverter/tree/v0.9.8.1) of the plugin and I can successfully set 100 and 0 as the grid export limit value. It also gets the right value from the inverter.

Any chance the range of the export limit can be adjusted based on model or by querying the capability of the system?

Thanks

mletenay, 2022-02-10:
Can you please paste screenshot from the SolarGo where it is begin set ?
I will look into it and try to make it device specific.

aksvenk, 2022-02-11:
Here you go:

[image]
[image]
[image]

aksvenk, 2022-02-11:
I see that it has been removed in this commit [in the latest version of the package](https://github.com/marcelblijleven/goodwe/commit/d52ec474fb27cb03640bec9880452a005ab15833#diff-a2e008d17666b1b97260064666ddc877c52ceb4bdfd3f11d7a24989c129077ceL111).

[image]

I was scratching my head as to why it wasn't working in the official HA integration 😄 

Btw the time sensor above never worked for me. It's probably a different register.

aksvenk, 2022-02-11:
Btw, looking at the packet trace from SolarGo, I have been able to receive the inverter time from register 30100(0x7594).

```python
Timestamp("time", 30100, "Inverter time"),

Integer("shadow_scan", 40326, "Shadow Scan", "", Kind.PV),
Integer("grid_export", 40327, "Grid Export Enabled", "", Kind.GRID),
Integer("grid_export_limit", 40336, "Grid Export Limit", "W", Kind.GRID),
```

gives the following values

```
2022-02-11 16:14:01,573 __init__(59) - DEBUG: Using selector: KqueueSelector
2022-02-11 16:14:01,573 _ensure_lock(100) - DEBUG: Creating lock instance for current event loop.
2022-02-11 16:14:01,574 _send_request(66) - DEBUG: Sending: 7f03759400035435
2022-02-11 16:14:01,628 datagram_received(49) - DEBUG: Received: aa557f030616020b100d35b487
2022-02-11 16:14:01,628 _send_request(66) - DEBUG: Sending: 7f039d8600014051
2022-02-11 16:14:01,715 datagram_received(49) - DEBUG: Received: aa557f03020000904e
2022-02-11 16:14:01,715 _send_request(66) - DEBUG: Sending: 7f039d8700011191
2022-02-11 16:14:01,803 datagram_received(49) - DEBUG: Received: aa557f03020001518e
2022-02-11 16:14:01,804 _send_request(66) - DEBUG: Sending: 7f039d900001a195
2022-02-11 16:14:01,893 datagram_received(49) - DEBUG: Received: aa557f0302003c905f
time: 		 Inverter time = 2022-02-11 16:13:53
shadow_scan: 		 Shadow Scan = 0
grid_export: 		 Grid Export Enabled = 1
grid_export_limit: 		 Grid Export Limit = 60 W
```

mletenay, 2022-02-13:
I switched the grid_export_limit to % for DT inverters.
Could you please test the master version of HACS component before I release that ?

aksvenk, 2022-02-14:
Do you have instructions on how to go about testing that? Now that goodwe integration is part of HA 2022.2?

aksvenk, 2022-02-14:
I get this on HA 2021.12

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 313, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 81, in async_setup_entry
    return await component.async_setup_entry(entry)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 147, in async_setup_entry
    platform = await async_prepare_setup_platform(
  File "/usr/src/homeassistant/homeassistant/setup.py", line 309, in async_prepare_setup_platform
    platform = integration.get_platform(domain)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 530, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 535, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_components/goodwe/number.py", line 50, in <module>
    GoodweNumberEntityDescription(
TypeError: __init__() got an unexpected keyword argument 'step'
```

mletenay, 2022-02-14:
HACS version now needs HA 2022.2.
There's nothing special since the integration got into HA, custom components from HACS always take precedence.
So just, upgrade HA to 2022.2.x and then re-download HACS integration to master version.

aksvenk, 2022-02-14:
Thanks for that. This is the error on 2022.2 with master checkout of the plugin:

```
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/goodwe/number.py", line 116, in async_setup_entry
if inverter.family == "DT" and description.key == "grid_export_limit":
AttributeError: 'DT' object has no attribute 'family'
```

mletenay, 2022-02-14:
OK, that was my mistake.
Should be fixed now, please try master again.

aksvenk, 2022-02-15:
I downloaded the latest master and it looks good now. 

**Entity**
The units, step and max value seem to be OK.

[image]

**History**
I have modified the value of grid export limit on HA and verified the change on SolarGo 👍 .
I still see `W` as the unit on the graph. Is it due to the fact that an older entity existed with the same id but different units?

[image]

mletenay, 2022-02-15:
Yes, the W unit is very likely caused by history records of the entity with same name.
I think HA also logs that mismatch as warning during start.

aksvenk, 2022-02-16:
I can confirm that was a relic from when the entity was previously configured. Deleting the entity from the HA db and restarting fixed the unit to `%` I think this is good to go. 

Could you please look at bringing back get/set of the `inverter time` as well as per https://github.com/mletenay/home-assistant-goodwe-inverter/issues/111#issuecomment-1035892702

Thanks again for being so responsive.

mletenay, 2022-02-18:
Well, the inverter time at register 30100 is the read-only register (and is already being provided as sensor as part of the get_runtime_data call).
It cannot be used for the write operations.
I expected the read-write register at 40313, but that one is obviously wrong, it did not work for anyone.

aksvenk, 2022-02-19:
I can confirm `40313` works on my inverter. The date and time was `2022-02-19 11:50:49`

Here is the complete packet that was sent out by SolarGo.

```
7f109d790003061602130b32318150
```

Part of Data Package | Description | Value
-- | -- | --
7F | Slave address | 0x7F (127)
10 | Function code | 0x10 (16) - Write Multiple Registers
9D 79 | Starting address | Physical: 0x9D79 (40313)Logical: 0x9D7A (40314)
00 03 | Quantity | 0x0003 (3)
06 | Byte count | 0x06 (6)
16 02 13 0B 32 31 | Registers value | 0x16 (22) 0x02 (12) 0x13 (19) 0x0B (11) 0x32 (50) 0x31 (49)
81 50 | CRC | 0x8150 (33104)

Let me know if you need more data with regards to this.

aksvenk, 2022-03-01:
Do you want me to run additional tests @mletenay ?

carlodie, 2023-02-20:
Hi guys, I found the same issue in my **Goodwe GW5000D-NS**. I mean the grid_export_limit (_**number.grid_export_limit**_) was set in **%** but after verify really is **Watt**, for example: 

> 100% = 100W, 50% = 50W and so on.

> [image]

I´ll try to change the attributes setting the Power = 5000 (according my Inverter model) and adjusting the unit of measurament to Watt  but I receive a error from HA, understand must to me adjusted by the owner of the integration :)

> [image]

In this way (Maybe must to be a new one post/discussion) trying to control the exported energy to the grid (want to be ~0) and considering this model of Inverter can´t control the zero exportation I made some adjustments manually in SolarGo in order to "control" the energy exportation adjusting the Power Limit Value:

> [image]

I have the Shelly Power Meter integrated in my HA, then I can see the power generated by the solar panels and the power of the grid (Negative means inyection to the grid).
Now if you can see I made the manual adjustments in order to control the generated power by panels, then I believe can control automatically, any thouths? 

> [image]

Other example:

> [image]

mletenay, 2023-02-22:
@carlodie This is really weird and D-NS specific ??
Since all inverters have the export limit in Watts and there is explicit different code for DT based inverters which use %.
So it seems like D-NS talks the "DT" protocol, but in this aspect behaves like the other inverter families ???
Aaaargh, Goodwe !

carlodie, 2023-02-23:
> @carlodie This is really weird and D-NS specific ??
> Since all inverters have the export limit in Watts and there is explicit different code for DT based inverters which use %.
> So it seems like D-NS [...]

Hi @mletenay I don't know about the other series or families, but in my D-NS (LCD screen) and SolarGo APP the set is in "Watt" but in Home Assistant integration figures/apper in **"%"**  but really the setup is in Watt.
As I mention before, for example in Home Assistant I set 10% but when I check in SolarGo the value is 10W, no 10% and so on.

This must to be adjusted in the code of Home Assistant Integration right?

corty8, 2023-09-02:
Hi All

Can I ask what the outcome of this was? I have the same 8500-ms inverter and trying to use the export limit but it is showing in watts not percent and when I try and set it I get an error