mletenay/home-assistant-goodwe-inverter #115: Inverter Operation modes

kolossboss, 2022-03-02:
Hi, 
can anyone tell me, what exactly the modes in "Inverter Operation Modes" does? I cant find any information. 

I need a mode, were the Battery is not discharging.

Thanks for the help.

tinuva, 2022-03-03:
Try backup mode? It only uses the battery if there is not enough sun and no grid.

kolossboss, 2022-03-03:
Hi,

thanks a lot, I will try.
I also testet Backup mode, but I recognized that, as you also wrote, the battery is charged from grid.
But in my case I will let the battery charge to 100% from PV first. So its not really an issue.

I'm not sure, what settings you mean, to manage the DoD through home assistant? Can you please explain that in more detail?

Thanks a lot

mletenay, 2023-02-05:
See PvMaster user manual for mode explanations

closed 2023-02-05 by mletenay